dresden-elektronik/deconz-rest-plugin #980: DeConz Network seems to be down - Help

PrimaryInc, 2018-11-26:
Hi,

Yesterday I noticed that my DeConz network seemed to be down.
In DeConz App, no "green" linje show anymore and I do not have contact with my light.
FW  262F0500
DeConz version 2.05.39 and 2.05.48, non works.

Ideas?

PrimaryInc, 2018-11-26:
Hi @manup,

Mail is sendt with backup.
I will have to try power cycle later this afternoon.
Only thing I've tried is to restart decenz and reboot host.

manup, 2018-11-26:
Thanks I've checked the database file, the configuration looks fine all parameters are stable.

Yes please try the following order:

1. Power cycle a light
2. If it doesn't help power cycle the RaspBee via *deCONZ > Edit > Reboot Device*

If none works we might check debug output:
 `deCONZ --http-port=80 --dbg-info=1 --dbg-error=1 --dbg-aps=1 --dbg-zdp=1`

PrimaryInc, 2018-11-26:
Hi @manup 

My gateway is connected thru VMware to a windows VM.
I did try to disconnect and connect the USB device a bit earlier, did not seem to do the trick.
And via DeConz it does not seem that gateway will reboot.

Am I correct that there is no log file written? Could not find one :)

PrimaryInc, 2018-11-26:
Hi @manup,

I've hard rebooted the device unplugging it physically. Rebootet VM, stil no luck.

manup, 2018-11-26:
This is very strange, can you please run deCONZ from terminal as follows to enable logging:

`deCONZ --http-port=80 --dbg-info=1 --dbg-error=1 --dbg-aps=1 --dbg-zdp=1`

manup, 2018-11-26:
Ah sorry didn't notice this is a Windows installation, here the logging is not shown in the console. To my knowledge the only way to see it by this helper tool from MS.

https://docs.microsoft.com/en-us/sysinternals/downloads/debugview

deCONZ needs to be started as above with logging enabled though.

PrimaryInc, 2018-11-26:
Did another test trying to include a new bulb, that did not work. 
So it leads me to suspect conbee for being broken.

PrimaryInc, 2018-11-26:
@manup 
USB Dongle is in same com port as it always been. 
Is the configuration saved in the usb dongle or is deconz software?
I can see it is possible to backup and restore the dongle is self.

Is there anyway you can thing of that can verify usb dongle is ok?

I was thinking of moving it to a new windows install, install fresh version of deconz to verify.
But want to make sure I don't loose anything if do it.

manup, 2018-11-27:
Checked the log I don't know what's going on here it looks all ok from the configuration.

>Is the configuration saved in the usb dongle or is deconz software?

The configuration is saved on the device, however since the database also holds a record  deCONZ could correct things like channel if needed. But that's not the case here.

>I can see it is possible to backup and restore the dongle is self.

>Could it be a FW issue, is it possible to down grade FW?

It is possible to downgrade but I don't think this will change something here, I really wondering where your network went.

We can try two thinks now:

1. If possible move one of the bulbs near the ConBee and use *Touchlink scan for devices* in the old WebApp 
 there we could see if if the network has moved to another channel or changed settings like PANID.

 *Phoscon App > Help > Old WebApp (2016) > Menu > Settings*

2. Use ZSHARK ZigBee sniffer to look where the network is, that's a bit more complicated.

>Is there anyway you can think of that can verify usb dongle is ok?
Guess that the electronics can brake.
All the test I´ve tried has failed.

Best way to test this is to add a new device, reset one light.

>I was thinking of moving it to a new windows install, install fresh version of deconz to verify.
But want to make sure I don't loose anything if do it.

With the backup the installation can be moved to another machine, might be worth a try albeit it would be cool to verify where the network is via the Scan method mentioned above.

PrimaryInc, 2018-11-27:
Hi @manup,

I've tried to move the conbee usb stick to another computer and installed .49 og deconz,  no luck including a new light bulb there.

So I did you step 1 and guess that the light is reached. Channel 25 is the default channel that everything is configured on.
[image]

That means the conbee is working as is should.

I installed z-shark, but get a error message saying wrong firmware.

[image]

I have 2 extra Ikea GU10 light bulbs and I´ve rest and tried to include one. But no luck, It will no add to network.

ebaauw, 2018-11-27:
See #796.

PrimaryInc, 2018-11-27:
Hi @ebaauw,

Not sure what your addressing? 
My ConBee firmware is latest.
Seems to be z-shark hat is not agreeing with something.

Version 2.05.49 / 11/27/2018
Firmware 262F0500

manup, 2018-11-27:
ZSHARK uses a different firmware as deCONZ. You can switch between them, the settings are preserved on the device.

Looking at the above I wonder even more why the network is not reachable to the gateway, channel and panid are the same.

manup, 2018-11-27:
Can you please post a screenshot of the deCONZ Network Settings?

Do you have a usb extension cable?

PrimaryInc, 2018-11-27:
@manup 

I've installed zshark firmware, need to install Wireshark also.
Yeah, you and me both :)

Network settings:
[image]

I did have, I´ve moved conbee directly on to why server to eliminate possible points of failure.

ebaauw, 2018-11-27:
> Seems to be z-shark hat is not agreeing with something.

There's a bug in the ZShark firmware update.

manup, 2018-11-27:
>There's a bug in the ZShark firmware update.

Should be fixed in latest version.

[image]

Here is the problem (I guess)

Please do the following:

* Security Mode: High - No Master but TC Link Key
* Save
* Done
* Leave
* Join

PrimaryInc, 2018-11-27:
@manup, I did the cabbage but no change.
Restarted deconz, and verified that setting stil was active.
Rebooted a few bulbs (30ish)

[image]

manup, 2018-11-27:
Ah sorry double checked with the database file.
Please use this *Network Key* from your backup file:

0x0000f4f4986cd0a4906c4040b4d8f498

* Save
* Done
* Leave
* Join

restart should not be needed

PrimaryInc, 2018-11-27:
@manup, so you are the star of the evening :)

[image]

Seems to start ut again. 

So to the big question, how and why did this happen??

manup, 2018-11-27:
Cool :)
Haven't seen that one before but I'll see to add some code to handle this automatically.

PrimaryInc, 2018-11-27:
@manup fantastic :) Hopefully everything will stay only from now on.

PrimaryInc, 2018-11-28:
@manup ,
Network has been up all night, thank you very much for your help!!

I do have one question for you, I can start a new thread if that is the best way.
I have 1 bulb that constantly loos connection (light 44), it is locate in a lamp outdoor. 
Inside the lamp there is to bulbs one pointing up the other down, so a bit strange it is the same bulb loosing connection. It is also very close to other bulbs.

Is there a way I can troubleshoot this single bulb? 
Rearranged the lights in DeConz and gave them names to easier see what they are used for.

[image]

manup, 2018-11-28:
You can move the related node a bit more into free space and enable the `LQI` button in deCONZ after a while (takes a while on your network size) you will see some numbers at the lines which represent the Link Quality Indicator from which you can see how good connection of the bulb is to neighbor nodes.

One thing to note is that ZigBee and radios in general don't like metal, so if the light is enclosed in a metal case this could be an issue.

PrimaryInc, 2018-11-29:
@manup,

Ok, I´ll have a go at it.
What I notice is that this bulb is close to oter bulbs inside the same type of metall casing, that don't have issues. So I guess it also could be that this bulb is a bit of, if you know what I mean. 
As you can see there is a lot of bulbs close to one another, the bulb that has issues is the 
one points down on the "house" on the terrace.

[image]

closed 2018-12-06 by PrimaryInc