dresden-elektronik/deconz-rest-plugin #981: light bulb and sensors losses connection to deconz

andenmad, 2018-11-27:
Hi 

i have really often light bulbs there losses connection to Deconz most times it solved then i power off the bulb and turn it on again. but have 3 times the last week exprienst that i need to delete the bulb/sensor in deconz and pair i again. 

deconz running 2.05.48 i have all kind off ikea bulbs one ikea plug and some xiaomi sensors. 

can som one help troble shoot this issue ore will it help to reinstall my deconz ?

manup, 2018-11-27:
Ikea lights seem to have trouble with recent versions: by looking at the sniffer I assume this might be caused by requesting ACKs for ZCL default response after attribute reportings are received. Maybe it's too much to handle for the light firmware. 

Ill revert this change back in 2.05.49 lets see if it behaves better..

Related issues: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/914 https://github.com/dresden-elektronik/deconz-rest-plugin/issues/980

andenmad, 2018-11-27:
@PrimaryInc   FW Version: 262F0500

@manup will this solve both bulb losses connection and the problem with need to re pair.

manup, 2018-11-27:
Only a test will tell, can you please provide more details on the light types and their firmware version?

PrimaryInc, 2018-11-27:
@manup When will you be able to have .49 ready?
Note I did try to downgrade without any luck
@andenmad Ok, same as I have.

andenmad, 2018-11-27:
[image]

[image]

andenmad, 2018-11-27:
here is my network settings should i set aps acks to enable ? 

[image]

manup, 2018-11-27:
Settings look fine, the change is directly in the code see the commit above.
Version is compiling and will be available shortly

manup, 2018-11-27:
Nope 0x262F0500 still holds

manup, 2018-11-27:
2.05.49 is out now for Raspbian and Windows (Ubuntu follows later today).

https://www.dresden-elektronik.de/deconz/

PrimaryInc, 2018-11-27:
@manup I've upgraded with same status, no lights connecting. I will try to restart light when I get home. If it does not work I´ll debug again.

andenmad, 2018-11-27:
Thanks i updatet and will get back when i se the result

andenmad, 2018-11-29:
hi 
i just lost one bulb today but today i neede to restart deconz and not the light bulb? 

but after restart off deconz it worked again.

rich710se, 2018-12-02:
I also lose Ikea lights now and then, supposed randomly. Happened with one of my bulbs some days ago, I was on 2.05.48 then. Have reseted the bulp many times, removed and addedfrom Phoscson tried everything. The bulp shows up again in deconz is yellow for some time and then get red after a while. Doesnt seem to communicate. It has worked perfectly for about a year before. Its "sisterbulb" "Hallvägg02" works perfect. Any hints what to do?
[image]

zladukas, 2018-12-02:
Yesterday I've lost one IKEA bulb because Philips Hue hub accidently fall onto raspi gateway. 

Check You all 2,4GHz sources, wifi, zigbee, etc nearby, it could be serious reason for loosing Ikea lights ;-)

rich710se, 2018-12-10:
I still have the issue with my Tradfri-bulb Hallvägg01, Upgraded to 2.05.50 today but didn't change the problem, The bulb show up in deconz after reset, but don't seem to communicate.

djwlindenaar, 2018-12-31:
Hi,
I just upgraded to 2.5.52 with 0x262F0500. I'm now also seeing some lights getting lost. I can get them back online, but I don't exactly know what does it. At least it takes resetting the light several times with open network and maybe some F5 or 0 pressing. Possibly restarting deConz is part of it.

I've now seen 3 lights take off. I notice that they stop responding and it shows on the desktop by having zero links to the item.

Anything I can try to solve or help debugging?

Best regards
Daniel

manup, 2018-12-31:
Can you please create and send a backup (or just the zll.db) file to mpi@dresden-elektronik.de for further analysis.

I'm currently investigating in the backup (restore) and also upgrade problems. So far it looks that somehow the network parameters change, in a temporary invalid state. The "good" working parameters are overwritten. The database keeps all parameter sets to be able to restore the working state, but this is currently not done automatically:
* Prevent recording and using a invalid configuration (the problem likely happened in your case)
* Fix backup restore if invalid parameters are also in there 
* Add a page to view and select previous configuration and apply it

djwlindenaar, 2018-12-31:
Just sent it. Note that the network is basically functional. The lamps that stopped working did so after some time. And not all at the same time either

djwlindenaar, 2019-01-01:
Happy new year!

Yeah, I've got a few of those. Never an issue with those, though. So wondering whether it's indeed related..?
The IKEA lights losing connectivity do not need to route through a xiaomi plug. They can mesh through other routers, several and  including other IKEA lights...

Anyway looking forward to trying a new version.

rich710se, 2019-01-05:
Today with 2.05.54 My "dead" Tradfri bulb that I haven't been able to connect since 2.05.52 or earlier, just kicked in.. Much more meshing overall in my network it looks like.. Happy! Thank you for your hard work with this! :)

manup, 2019-01-06:
Yes hopefully Ikeas will have a better time in the ZigBee network now, requests were reduced a lot. :)

closed 2019-05-13 by stale[bot]