dresden-elektronik/deconz-rest-plugin #983: Xiaomi square button WXKG11LM not showing event 1000 or 1003

smofrad, 2018-11-27:
I am running Phoscon version 2.05.39 and i cant get my xioami button to show the correct events.
When i Do a sinlge press a get event id 1002. Double gives me 1004 and triple gives me 1005. According to https://github.com/dresden-elektronik/deconz-rest-plugin/issues/349#issuecomment-359294830
I should be able to get 1000 and 1003 events. 

Do I need to configure something to get the events 1000 and 1003 according to above link?

This is how it looks in the Home assistant log file

2018-11-27 20:35:27 DEBUG (MainThread) [pydeconz.deconzdevice] Smart Switch: update buttonevent with 1002
2018-11-27 20:35:27 DEBUG (MainThread) [pydeconz.deconzdevice] Smart Switch: update lastupdated with 2018-11-27T19:35:27
2018-11-27 20:35:27 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event deconz_event[R]: id=smart_switch, event=1002>
2018-11-27 20:35:27 DEBUG (MainThread) [pydeconz.websocket] Websocket data: {"e":"changed","id":"8","r":"sensors","state":{"buttonevent":1002,"lastupdated":"2018-11-27T19:35:27"},"t":"event"}
2018-11-27 20:35:27 DEBUG (MainThread) [homeassistant.components.websocket_api.http.connection.1847239024] Sending {'id': 2, 'type': 'event', 'event': {'event_type': 'deconz_event', 'data': {'id': 'smart_switch', 'event': 1002}, 'origin': 'REMOTE', 'time_fired': datetime.datetime(2018, 11, 27, 19, 35, 27, 231334, tzinfo=<UTC>), 'context': {'id': 'ae72945bef2c487f8b6c87b7aa16b383', 'user_id': None}}}
2018-11-27 20:35:27 DEBUG (MainThread) [homeassistant.components.websocket_api.http.connection.1847275376] Sending {'id': 2, 'type': 'event', 'event': {'event_type': 'deconz_event', 'data': {'id': 'smart_switch', 'event': 1002}, 'origin': 'REMOTE', 'time_fired': datetime.datetime(2018, 11, 27, 19, 35, 27, 231334, tzinfo=<UTC>), 'context': {'id': 'ae72945bef2c487f8b6c87b7aa16b383', 'user_id': None}}}

And this is how it looks in the rest-api

     "ep": 1,
        "etag": "4998b264331dd1b3ec90fdb518d4f418",
        "manufacturername": "LUMI",
        "mode": 1,
        "modelid": "lumi.sensor_switch.aq2",
        "name": "Smart Switch",
        "state": {
            "buttonevent": 1002,
            "lastupdated": "2018-11-27T19:35:27"
        },
        "type": "ZHASwitch",
        "uniqueid": "00:15:8d:00:01:b8:6b:bb-01-0006"
    },

saxn-paule, 2018-11-27:
A long press should give you 1003. See my thread regarding this button.
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/917

smofrad, 2018-11-28:
I've loooked through the documantaiton regarding supported decvices and it seems that the issue is that my button is gen1. I will close this issue.

closed 2018-11-28 by smofrad