dresden-elektronik/deconz-rest-plugin #984: Heiman HS1SA not available in REST

Andre0512, 2018-11-27:
Hello!

I bought the Heiman HS1SA Smoke Sensor. On the shop article page stood the sensor is from Heiman, however on the sensor, the package or the manual stand nowhere "Heiman", but "Photoelectronic". At least the product ID on the sensor ends with HS1SA, so I think it should be the supported sensor.

I tried to add it as sensor via the Phoscon Web App, but got no successful response. To my surprise, I found it in the deconz-gui in my cluster, but the sensor is still not available via the REST-API. I tried to reconnect it many times, but have had no success with the REST-API

deconz: 2.05.48
rest-plugin: 2.05.48

I use the Raspberry Pi 1, if this matter.

Here the screenshots from the deconz-gui:
[image]
[image]
[image]
[image]
[image]
[image]

Here the log with a grep of the relevant sensor.
```
22:56:00:180 Node: id: 31, 00:0d:6f:00:11:fc:37:67 (0x000D6F0011FC3767) scene: 2133,319018, -155,745399
22:56:17:494     * neighbor: 0x000D6F0011FC3767 (0xFE7F), LQI: 20, relation: 0x04 rxOnWHenIdle: 0
22:56:17:494     * previous child: 0x000D6F0011FC3767
22:58:49:296     * neighbor: 0x000D6F0011FC3767 (0xFE7F), LQI: 60, relation: 0x03 rxOnWHenIdle: 0
23:00:07:191 APS-DATA.indication srcAddr: 0x000d6f0011fc3767, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x0013, lqi: 71, rssi: -82
23:00:07:192 device announce 0x000D6F0011FC3767 (0xD521) mac capabilities 0x80
23:00:07:192 set fast probe address to 0x000D6F0011FC3767 (0xD521)
23:00:07:193 device announce 0x000D6F0011FC3767 (0xD521) mac capabilities 0x80
23:00:07:193 set fast probe address to 0x000D6F0011FC3767 (0xD521)
23:00:07:193 ZDP device announce: 0x000D6F0011FC3767, 0xD521, 0x80
23:00:08:122     * neighbor: 0x000D6F0011FC3767 (0xD521), LQI: 70, relation: 0x04 rxOnWHenIdle: 0
23:00:08:122     * previous child: 0x000D6F0011FC3767
23:00:08:564 [2] get active endpoints for 0x000d6f0011fc3767
23:00:10:801 send NWK_Addr_req to node  0x000D6F0011FC3767
23:00:10:803 send NWK_addr_req to 0x000D6F0011FC3767, last seen 3 s, last seen by neighbors 81 s
23:00:12:664 Power_Descriptor_req zdpSeq: 157 to 0x000D6F0011FC3767
23:00:13:571 Simple_Descr_req zdpSeq: 158 to 0x000D6F0011FC3767 endpoint 1
23:00:15:624 Simple_Descr_req zdpSeq: 159 to 0x000D6F0011FC3767 endpoint 1
23:00:21:628     * neighbor: 0x000D6F0011FC3767 (0xD521), LQI: 20, relation: 0x04 rxOnWHenIdle: 0
23:00:21:628     * previous child: 0x000D6F0011FC3767
23:00:37:076 APS-DATA.indication srcAddr: 0x000d6f0011fc3767, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x0013, lqi: 55, rssi: -84
23:00:37:077 device announce 0x000D6F0011FC3767 (0xD521) mac capabilities 0x80
23:00:37:077 set fast probe address to 0x000D6F0011FC3767 (0xD521)
23:00:37:077 device announce 0x000D6F0011FC3767 (0xD521) mac capabilities 0x80
23:00:37:077 set fast probe address to 0x000D6F0011FC3767 (0xD521)
23:00:37:078 ZDP device announce: 0x000D6F0011FC3767, 0xD521, 0x80
23:00:38:156 [2] get active endpoints for 0x000d6f0011fc3767
23:00:52:213 APS-DATA.indication srcAddr: 0x000d6f0011fc3767, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x0013, lqi: 21, rssi: -87
23:00:52:213 device announce 0x000D6F0011FC3767 (0xD521) mac capabilities 0x80
23:00:52:213 set fast probe address to 0x000D6F0011FC3767 (0xD521)
23:00:52:213 device announce 0x000D6F0011FC3767 (0xD521) mac capabilities 0x80
23:00:52:213 set fast probe address to 0x000D6F0011FC3767 (0xD521)
23:00:52:225 ZDP device announce: 0x000D6F0011FC3767, 0xD521, 0x80
23:00:53:179 [2] get active endpoints for 0x000d6f0011fc3767
23:01:04:211 APS-DATA.indication srcAddr: 0x000d6f0011fc3767, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8005, lqi: 23, rssi: -88
23:01:04:213 ZDP active ep response for 0x000d6f0011fc3767
23:01:04:339 [3.1] get IAS Zone type for 0x000d6f0011fc3767
23:01:05:139 APS-DATA.indication srcAddr: 0x000d6f0011fc3767, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0500, lqi: 15, rssi: -88
23:01:05:235 [4] get basic cluster attr 0x0004 for 0x000d6f0011fc3767
23:01:05:253 APS-DATA.indication srcAddr: 0x000d6f0011fc3767, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0500, lqi: 7, rssi: -88
23:01:05:254 APS-DATA.request id: 91, addr: 0x000D6F0011FC3767 profile: 0x0104, cluster: 0x0000, ep: 0x01/0x01 queue: 11 len: 5 (send, fast lane)
23:01:05:335 [4] get basic cluster attr 0x0004 for 0x000d6f0011fc3767
23:01:05:393 APS-DATA.indication srcAddr: 0x000d6f0011fc3767, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0000, lqi: 15, rssi: -88
23:01:05:493 [4] get basic cluster attr 0x0005 for 0x000d6f0011fc3767
23:01:06:255 APS-DATA.indication srcAddr: 0x000d6f0011fc3767, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0000, lqi: 5, rssi: -89
23:01:06:682 APS-DATA.indication srcAddr: 0x000d6f0011fc3767, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0000, lqi: 5, rssi: -89
23:01:07:106 APS-DATA.indication srcAddr: 0x000d6f0011fc3767, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0000, lqi: 15, rssi: -88
23:01:08:058 aps request id: 63 prf: 0x0000 cl: 0x0005 timeout (confirmed: 1) to 0x000D6F0011FC3767 (0xD521)
23:01:10:061 aps request id: 77 prf: 0x0000 cl: 0x0000 timeout (confirmed: 1) to 0x000D6F0011FC3767 (0xFFFD)
23:01:13:017 aps request id: 88 prf: 0x0000 cl: 0x0004 timeout (confirmed: 1) to 0x000D6F0011FC3767 (0xD521)
23:01:21:171 APS-DATA.indication srcAddr: 0x000d6f0011fc3767, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0000, lqi: 21, rssi: -87
23:01:22:205 APS-DATA.indication srcAddr: 0x000d6f0011fc3767, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0000, lqi: 7, rssi: -88
23:01:22:525 APS-DATA.indication srcAddr: 0x000d6f0011fc3767, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0000, lqi: 15, rssi: -88
23:01:23:185 APS-DATA.indication srcAddr: 0x000d6f0011fc3767, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0000, lqi: 21, rssi: -87
23:01:38:019 aps request id: 224 prf: 0x0000 cl: 0x0005 timeout (confirmed: 1) to 0x000D6F0011FC3767 (0xD521)
23:03:46:439     * neighbor: 0x000D6F0011FC3767 (0xD521), LQI: 16, relation: 0x04 rxOnWHenIdle: 0
23:03:46:439     * previous child: 0x000D6F0011FC3767
23:05:03:600     * neighbor: 0x000D6F0011FC3767 (0xD521), LQI: 70, relation: 0x04 rxOnWHenIdle: 0
23:05:03:600     * previous child: 0x000D6F0011FC3767
23:05:16:451     * neighbor: 0x000D6F0011FC3767 (0xD521), LQI: 20, relation: 0x04 rxOnWHenIdle: 0
23:05:16:451     * previous child: 0x000D6F0011FC3767
```
I hope you could help me...

Thanks
Andre

peer69, 2018-11-28:
I was able to pair 2 sensors already a while ago. I've bought two additional sensors and now i'm running into the exact same problem. Have spent hours resetting, waiting, reading cluster while keeping the sensor on, nothing helped yet.

peer69, 2018-11-28:
Great. Thank you for the commit! Hopefully this will be added to a new version soon.

Andre0512, 2018-11-28:
I compiled your pull request and after I reconnected the sensor, it worked. Thank You!

moritzthecat, 2018-11-29:
excellent - will buy some now.
thanx

peer69, 2018-12-04:
Unfortunately, this does not work for me. My sensor seems to be using another prefix which hasn't been listed yet. 
[image]

peer69, 2019-01-03:
While I never managed to get the sensors with the prefix mentioned above paired, I got another one with the EM Prefix. I can pair it, but the Alarms Cluster isn't created: 
[image]

Any idea how to fix this? Currently, alarms don't trigger anything visible in the rest api.

moritzthecat, 2019-01-05:
Hi - I would like to try out the Heiman HS1SA.
How can I make sure to buy a supported and working model?
Is there a good question to ask the seller?
Thanx in advance

peer69, 2019-01-05:
I have three different models now. One is working, which seems to be the oldest. Newer models with EM and Lumi prefix don’t work. 
For now I have given up and ordered a Xiaomi/Honeywell smoke detector to test. If this one works I will sell all my Heiman detectors as only 2 of 12 sensors work with deCONZ.

closed 2019-05-12 by stale[bot]

filipvh, 2019-07-26:
I have a HEIMAN HS1SA-E  it seems to be detected, but doesn't show in the list on the web app. But when I go look in Home Assistant it does show. I don't know if it's supposed to detect when i 'test' it.... because it doesnt :(

filipvh, 2019-07-26:
Ok tested that with a spray, didn't show up in Home Assistant as detected. New model? I'm running deConz 2.05.65 (as there is no .66 on the deb repo :( )