dresden-elektronik/deconz-rest-plugin #987: Aqara Motion and Daylight sensor reporting wrong lux level / wrong "dark" parameter

t-anjan, 2018-11-29:
I have a Xiaomi Aqara motion sensors, which has the daylight sensor built-in. It is imported into deconz using the Phoscon UI (I have a RaspBee). I have a problem with the light level which is reported by deconz. I am on deconz version 2.05.47 using the docker image here: https://github.com/marthoc/docker-deconz.

The light level reported is extremely random. Even in total darkness, I see in the Phoscon UI that the lux is shown to be above 25. When motion is triggered, the websocket message is as following:

```
2018-11-28 22:22:13,465 - root - DEBUG - {"e":"changed","id":"7","r":"sensors","state":{"dark":false,"daylight":false,"lastupdated":"2018-11-28T16:52:13","lightlevel":14314,"lux":27},"t":"event","uniqueid":"00:15:8d:00:02:79:ad:67-01-0400"}
2018-11-28 22:22:13,560 - root - DEBUG - {"e":"changed","id":"6","r":"sensors","state":{"lastupdated":"2018-11-28T16:52:13","presence":true},"t":"event","uniqueid":"00:15:8d:00:02:79:ad:67-01-0406"}
```
You can see that the motion sensor triggered the output, but the light sensor's output shows a lux of 27, and the `dark` parameter is set to `false`. This prevents any of the rules from triggering, because the Hue bridge thinks there is enough light (I use diyHue emulated Hue bridge).

Very rarely, the lux level shown comes down to less than 10 and `dark` is set to `true` (this is in total darkness). And then, the rules set up in the Hue app work. But this happens very rarely.

What could be the problem? Any suggestions as to how I can debug this would be very helpful. 

Is it possible that the daylight sensor hardware is broken? 

Thanks!

t-anjan, 2018-12-04:
On further testing, I think the problem is only with the sensor hardware. 

[image]

I placed all my sensors beside each other, in a room with uniform lighting. The above lux values in the image were reported by deconz.
You can see that 2 of the sensors have wildly different lux values. 

Is there a way in which I can tune the sensitivity of the sensor in deconz?

manup, 2018-12-04:
In my opinion the values are fine and expected, two things to note here:

* The small Xiaomi sensor is likely not very accurate compared to a professionell sensor
* More important, lux value range is not linear but logarithmic, values will have huge differences albeit position is close

You may install a app to measure this in realtime like `Lux Light Meter` for Android. Watch the Lux value while in the room and when move slightly, also watch the value next to the window with daylight. You might be surprised :)

t-anjan, 2018-12-04:
Thanks for the tip about the logarithmic scale. 

[image]

I placed all sensors right next to each other with just one source of light in front of them, about 10 feet away. I understand from what you say, that even small variations in light can lead to different lux values, but seeing 25 Lux and 83 Lux is a bit extreme, don't you think?

The problem becomes really apparent at low light levels. The sensors, which are showing 25 and 33 lux in the previous image, go down to 1 or 2 lux at low (nearly dark) light levels, which is as expected. But these other 2 sensors, which are showing 66 and 83 lux in the previous image, still continue showing 25+ lux at dark, really low light levels. This causes the `dark` parameter to always be `false`, and none of my rules get triggered.

So, seeing the above, I cannot understand how these values can be considered fine or expected. Please correct me if my understanding is wrong.

manup, 2018-12-04:
>So, seeing the above, I cannot understand how these values can be considered fine or expected. Please correct me if my understanding is wrong.

For this specific sensor they should be fine, it's a 10$ sensor for primarily motion sensing. In my opinion the light sensor is ok to tell if there is 'no light at all' or 'some light'. But for finer resolution a specific light sensor should be used.

In my experience with various consumer sensors I consider the range 20&ndash;200 Lux as very dynamic (aka it's the same for me), therefore you might use dark < 150.

In office setups we often use dark < 300 Lux.

https://en.wikipedia.org/wiki/Lux

ebaauw, 2018-12-04:
> This causes the `dark` parameter to always be false, and none of my rules get triggered.

You can set the thresholds for `dark` and `daylight` _per sensor_ through the API.  These are in `lightlevel` units rather than in lux.  I use 12000 for `config.tholddark`, meaning that below that threshold, `dark` becomes true and my lights turn on.  For `config.tholdoffset` I use values between 4000 and 12000, meaning `daylight` becomes true and my lights turn off above a `lightlevel` of 16000 to 24000.  The value depends on how the sensor is located in relation to the lights/spots: lower values for when it's in the shadow, higher values for when the sensor looks straight into a light.

I don't know the Xiaomi motion sensor, but the Hue motion sensor only notifies the gateway immediately when the `lightlevel` changes by more than 2000.  Additionally, it reports periodically every 5 minutes, so it might take deCONZ up to 5 minutes, before it reports a smaller change.  Other Xiaomi devices typically report periodically only once every hour.  So it might take quite a while for the Xiaomi motion sensor to report a smaller change in lightlevel to deCONZ.  I see the same for my Xiaomi weather sensors.

For the background on `lightlevel` See https://developers.meethue.com/develop/hue-api/supported-devices/#supportred-sensors, section 2.7:

> Light level in 10000 log10 (lux) +1 measured by sensor. Logarithm scale used because the human eye adjusts to light levels and small changes at low lux levels are more noticeable than at high lux levels.

Example | Lux | MeasuredValue
-- | -- | --
Outdoor: Overcast moonless night sky | 0.0001 | 0
Outdoor: Bright moonlight | 1 | 1
Home: Night light | 2 | 3000
Home: Dimmed light | 10 | 10000
Home: ‘Cosy’ living room | 50 | 17000
Home: ‘Normal’ non-task light | 150 | 22000
Home: Working / reading | 350 | 25500
Home: Specialized tasks, Inside daylight | 700 | 28500
Home: Maximum to avoid glare | 2000 | 33000
Outdoor: Clear daylight | > 10000 | > 40000
Outdoor: Brightest direct sunlight | 120000 | 51000

ebaauw, 2018-12-05:
> Do you mean the deconz API? Is there some API documentation for this?

Yes.  The documentation seems to be missing from 
http://dresden-elektronik.github.io/deconz-rest-doc/sensors/, but you can refer to the Hue api, section 2.7 on 
https://developers.meethue.com/develop/hue-api/supported-devices/#supportred-sensors

> What happens if the measured light is between 12000 and 20000?

Nothing.  They remain on, or remain off.  You want this “twilight zone” to prevent the lights from constantly flipping on and off when the sensor registers too much light when they’re on.  See 
https://github.com/ebaauw/homebridge-hue/issues/83

t-anjan, 2018-12-05:
Thank you for this. I thought deconz was limited to what is available through the web UI. The API has just opened up a new world for me. :-)

I have tuned the `tholddark` for a couple of sensors. I will know if it works later tonight. I wish this was available in the Phoscon UI, though. 

I will close this issue once I confirm that the `dark` parameter is being sent according to my tuned threshold numbers.