dresden-elektronik/deconz-rest-plugin #990: Ubisys C4: buttonevent=null

theFork, 2018-11-29:
After hitting all available momentary switch inputs of a Ubisys C4 device, the state/buttonevent and state/lastupdated remains unaffected:

```
{
    "config": {
        "group": null,
        "mode": "momentary",
        "on": true,
        "reachable": true
    },
    "ep": 1,
    "etag": "6a701dc849cf735effe83af945bd9c51",
    "manufacturername": "ubisys",
    "mode": 1,
    "modelid": "C4 (5504)",
    "name": "C4 (5504) 30",
    "state": {
        "buttonevent": null,
        "lastupdated": "none"
    },
    "swversion": "20170728-DE-FB0",
    "type": "ZHASwitch",
    "uniqueid": "00:1f:ee:00:00:00:2b:15-01-0006"
}
```

Generally ZHASwitch devices (tradfri remote) are working for me. How can I figure out whether this is a deconz-rest-api issue or a problem of the deconz.service?

manup, 2019-01-04:
It's been a while since I've used the C4 but as far as I remember the switch can only be configured in a simple way like 4 on/off buttons.

Configuration to support push dim on all buttons wasn't possible since the needed configuration message payload was too large to transfer.

I can schedule a test next week to see that at least the simple on/off for all buttons works.

theFork, 2019-01-07:
That would be most kind, thanks in advance.

I've just noticed that in case I assign groups using the Phoscon webinterface, the C4 generates button events:

```
HTTP/1.1 200 OK
Access-Control-Allow-Origin: *
Content-Length: 371
Content-Type: application/json; charset=utf-8
ETag: "100b9d23e05fa935f82d0a401b2030ad"

{
    "config": {
        "group": "2,3,null,null",
        "mode": "momentary",
        "on": true,
        "reachable": true
    },
    "ep": 1,
    "etag": "100b9d23e05fa935f82d0a401b2030ad",
    "manufacturername": "ubisys",
    "mode": 1,
    "modelid": "C4 (5504)",
    "name": "C4 (5504) 26",
    "state": {
        "buttonevent": 2002,
        "lastupdated": "2019-01-07T19:49:44"
    },
    "swversion": "20180921-DE-FB0",
    "type": "ZHASwitch",
    "uniqueid": "00:1f:ee:00:00:00:2a:44-01-0006"
}
```

Note, that I changed the C4 and also the firmware.

Could this behavior be intentional? I though it would be best not to use the grouping functionality of Deconz and instead trigger scenes within the layer above (homeassistant).

Can you confirm that you get no evens when there is no group assignment?

manup, 2019-01-07:
>Could this behavior be intentional?

Actually yes, the groups are used to create bindings; the switch sends commands to the group as destination. deCONZ captures these commands and generates the button events.

The groups can be kept empty.

imbaimbaimba, 2019-06-19:
@manup / @theFork 
Did you get any further with this?
I would like to try the Ubisys C4 as an in-wall Philips Hue Dimmer alternative. Would that be possible with the current integration?

imbaimbaimba, 2019-06-19:
Or maybe @ebaauw you can fill me in on whether this would work? Seems you worked on the integration.

ebaauw, 2019-06-20:
I have first hand with the Hue dimmer switch and the ubisys D1, but I’ve never laid my hands on a ubisys C4.  I only used the factory default configuration of the D1, for connecting two (stateless) pulse switches.  I could use these to turn on or off and to dim the lights in the corresponding group en lieu of the Hue dimmer.  If you want to create your own automation (using buttonevents in deCONZ rules or some other home automation system), the Hue dimmer offers way more functionality.

imbaimbaimba, 2019-06-20:
> I have first hand with the Hue dimmer switch and the ubisys D1, but I’ve never laid my hands on a ubisys C4. I only used the factory default configuration of the D1, for connecting two (stateless) [...]

Ok, thank you for the insight.

I am not sure I understand how the Hue dimmer offers way more functionality though? If the Ubisys devices offer single click, double click, hold, release for it's four buttons, that would be the same button events to use as triggers in deCONZ or HA? Can you explain how the functionality differs?

Also, I am curious about how the Ubisys devices will show up(I understand you can only talk for the D1). Can I use the switch editor inside the web UI to set up the buttons? I would assume that you would get something similar to this: [picture](https://user-images.githubusercontent.com/383386/51938361-2a0a3e80-240d-11e9-96d9-2b51719d3e87.png) when using a D1, so the buttons could be used as dimmers?

ebaauw, 2019-06-20:
The ubisys devices don’t offer any buttonevent.  They send commands to the associated group, that deCONZ (or rather the REST API plugin) reverse engineers into buttonevents.  This is limited by the commands the switch sends.  The Hue dimmer switch reports buttonevent directly to the gateway.  It’s the only device I know that sends an x001 event every second while holding.  This enables you to write a rule that dims the light while holding, where this is not possible for other switches (the command they send to the group to dim the lights cannot be sent from the API).

I don’t use Phoscon and I replaced my D1 with an iCasa dimmer (it was rebooting spontaneously - I suspect a hardware issue), so I wouldn’t know what the switch editor makes of it.

imbaimbaimba, 2019-06-20:
> The ubisys devices don’t offer any buttonevent. They send commands to the associated group, that deCONZ (or rather the REST API plugin) reverse engineers into buttonevents. This is limited by the [...]

Hmm, I'm not sure I understand what the consequence of this difference in functionality actually means for my use-case. Also, I pretty much only ever used the web interface to do anything with my ConBee, the only think I used API directly for was the experimenting with Z-Wave switches for dimming through Home Assistant, that never worked perfectly well.

Say I got a Hue bulb that I want to be able to control with four buttons:
I put the mentioned bulb in a group called, say, "Living Room".
Say I then add a Hue Dimmer Switch and for example a Ubisys C4 to the network. If I then go to each of these to, to define what their buttons do, I would go to the switch editor I mentioned before to be able to assign the buttons. The options are, at least with the Hue Dimmers, a little limited in here, but I would be able to then choose these buttons events:
Button 1: Lighter and on
Button 2: Darker an off
Button 3: On
Button 4: Off

In the case of the Hue Dimmer Switch, if I press/hold/release any of the four buttons, it will then send different x001 button events to deCONZ.
In the case of the Ubisys C4, if I do the same, it will send a command to a deCONZ group, that deCONZ then changes into x001 button events.

All of that happens in the background though I assume, so I would not see any difference, maybe except for a small delay caused by the extra "layer" where it translates the commands to events?

> This enables you to write a rule that dims the light while holding, where this is not possible for other switches (the command they send to the group to dim the lights cannot be sent from the API).

So really what I am asking about is how to understand what you're saying here. Is the Hue Dimmer Switch literally the only way to get a button, that I can assign to dim a light up/down when holding? Or does the limitations you talk about only affect other(possibly more advanced) use-cases than mine?

What I need is just the simple press to on/off and hold to dim/brighten, so that I can smarten my original Lauritz Knudsen FUGA wall switches, to be able to control Hue lights connected to permanent power. That mission has turned out to be much much harder than I expected, as I have run against a wall on every solution I tried till now.

What I ended up doing for now is soldering wires over the buttons on Philips Hue Dimmer Switches, and connecting them to NO on a bunch of relays, and then connection 4-button wall switches with 5v to open the relays. This is however not the ideal solution, as wiring this in my house, have turned out to be extremely challenging and maybe even impossible.

ebaauw, 2019-06-20:
> Is the Hue Dimmer Switch literally the only way to get a button, that I can assign to dim a light up/down when holding?

When using rules, or a home automation system: yes.   When using groups: no.  We discussed this before, see https://github.com/ebaauw/homebridge-hue/issues/478.