dresden-elektronik/deconz-rest-plugin #991: Deconz gets confused

winterscar, 2018-11-29:
When toggling states (on/off) quickly in Deconz on a group of Hue lights, Deconz becomes "confused" about the actual state of the lights. 

When this happens, the lights will show as off when they are actually on, and Deconz will take a while to realize it's wrong. 

I've seen this issue on versions _44 and _47.

winterscar, 2018-11-29:
I've taken a look at the docker logs and I get the following: 
```
20:10:46:234 Content not completely loaded (got 0 of 0), wait 20ms
20:10:46:236 5 running tasks, wait
20:10:46:240 5 running tasks, wait
20:10:46:244 5 running tasks, wait
20:10:46:270 5 running tasks, wait
20:10:46:285    0x0017880104D25206 force poll (2)
20:10:46:342    0x0017880103894FBC force poll (2)
20:10:46:370 5 running tasks, wait
20:10:46:406    0x000B57FFFE4B7E67 force poll (2)
20:10:46:448    0x00178801039C7088 force poll (2)
20:10:46:526    0x000B57FFFE4D96FF force poll (2)
20:10:46:568    0x000B57FFFE37C053 force poll (2)
20:10:47:215 Content not completely loaded (got 0 of 0), wait 20ms
20:10:48:046 Content not completely loaded (got 0 of 0), wait 20ms
20:10:48:070 delayed group sending
20:10:48:766 Content not completely loaded (got 0 of 0), wait 20ms
20:10:48:776 Content not completely loaded (got 0 of 0), wait 20ms
20:10:48:778 5 running tasks, wait
20:10:48:781 5 running tasks, wait
20:10:48:784 Content not completely loaded (got 0 of 0), wait 20ms
20:10:48:785 5 running tasks, wait
20:10:48:789 Content not completely loaded (got 0 of 0), wait 20ms
20:10:48:790 5 running tasks, wait
20:10:48:795 5 running tasks, wait
20:10:48:866    0x0017880104D25206 force poll (2)
20:10:48:870 5 running tasks, wait
20:10:48:908    0x0017880103894FBC force poll (2)
20:10:48:970 5 running tasks, wait
20:10:48:980    0x00178801039C7088 force poll (2)
20:10:49:052    0x000B57FFFE4B7E67 force poll (2)
20:10:49:085    0x000B57FFFE4D96FF force poll (2)
20:10:49:113    0x000B57FFFE37C053 force poll (2)
20:10:49:643 Content not completely loaded (got 0 of 0), wait 20ms
20:10:49:670 delayed group sending
20:10:50:370 Content not completely loaded (got 0 of 0), wait 20ms
20:10:51:039 Content not completely loaded (got 0 of 0), wait 20ms
20:10:51:047 Content not completely loaded (got 0 of 0), wait 20ms
20:10:51:048 5 running tasks, wait
20:10:51:052 5 running tasks, wait
20:10:51:057 Content not completely loaded (got 0 of 0), wait 20ms
20:10:51:058 5 running tasks, wait
20:10:51:062 5 running tasks, wait
20:10:51:067 5 running tasks, wait
20:10:51:070 5 running tasks, wait
20:10:51:158    0x0017880103894FBC force poll (2)
20:10:51:170 5 running tasks, wait
20:10:51:265    0x00178801039C7088 force poll (2)
20:10:51:271 5 running tasks, wait
20:10:51:278    0x000B57FFFE4B7E67 force poll (2)
20:10:51:387    0x000B57FFFE4D96FF force poll (2)
20:10:51:434    0x000B57FFFE37C053 force poll (2)
20:10:51:930 Content not completely loaded (got 0 of 0), wait 20ms
20:10:51:974 delayed group sending
20:10:53:494    0x0017880104D25206 force poll (2)
20:10:57:326 ZCL attribute report 0x000B57FFFE4D96FF for cluster 0x0300, ep 0x01
20:10:57:464 ZCL attribute report 0x000B57FFFE4D96FF for cluster 0x0008, ep 0x01
20:10:58:370 dev /dev/ttyAMA0
```

manup, 2018-11-29:
There is not much we can do about it, after lights receive a group command the new light state is either queried or when supported lights will report the new state. This may take several seconds depending on the network size.

Group commands are not meant to be invoked too fast, it should cause no harm but as you noticed it will take some time till things are settled.

Out of interest, do you have a real need for fast toggling groups? :)

closed 2019-04-05 by stale[bot]