dresden-elektronik/deconz-rest-plugin #992: Getting API key without lightbulb

qm3ster, 2018-11-30:
- I installed the following deCONZ:
[deconz/raspbian/deconz-latest.deb](https://www.dresden-elektronik.de/deconz/raspbian/deconz-latest.deb)	**29-Nov-2018 23:30**	4.7M
(For my ODROID-C1 I had to make a fake empty apt package that
provides `wiringpi` so it wouldn't complain)

- started it with `deCONZ -platform minimal`

- and started doing the REST [Getting Started](http://dresden-elektronik.github.io/deconz-rest-doc/getting_started/). I see that to get an API key I need to press a UI button.
(I am successfully getting the 403)

When I go to  `http://<IP>:<80>`, I initially get a "Wireless Light" page with only a navbar, but then it quickly redirects to `http://<IP>/pwa/login.html`

If I use this PWA to connect to the deCONZ, and set a password as prompted, it puts me to
`http://<IP>/pwa/commissioning.html?1543526405000`, which doesn't allow me to continue without connecting a device first.

What's the url of the **Unlock Gateway** button? Can I hit it without completing "commissioning"?
Better yet, can I get an API key locally on the deCONZ machine, without exposing the API key generation API to the whole hostile network for a whole minute?

I just want my API key :sob:

manup, 2018-11-30:
>I just want my API key 

To jump over commissioning page open  `http://<IP>/pwa/settings-gateway.html`

We will introduce a skip button in future versions.

qm3ster, 2018-11-30:
Thanks!
I worked around it by jumping to `http://<IP>/login.html` instead :)
(which didn't cause the redirect, so I could log into old app)

Do you happen to know how deCONZ detects and uses the `/dev/ttyUSB0` device?
I am trying to pass it to a different machine over `ser2net`.

manup, 2018-11-30:
The USB devices are enumerated by Qt classes and and filtered by vendor/product id.

closed 2019-01-20 by manup