dresden-elektronik/deconz-rest-plugin #993: Device Integration: Samsung SmartThings Button

mysteryan, 2018-11-30:
Is it possible to add compatibility for SmartThings Button: 
https://www.amazon.com/Samsung-SmartThings-GP-U999SJVLEAA-One-Touch-Appliances/dp/B07F8ZFFQK/
Hoping integration would be simple since other SmartThings products have been integrated and its already recognized by Deconz GUI.  Device transmits events for button press, double press, hold and temperature.  Node shows up in GUI, but missing from rest api.
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

mysteryan, 2018-12-04:
Debug (aps,zcl,zdp) output doesn’t seem to with frep filtered result of nwk address doesn’t seem to yield anything useful.
[image]

I do notice changes in zone status when I press the button shown in no press, press, double press and hold in that order:
[image]
[image]
[image]
[image]

mysteryan, 2018-12-11:
Just realized that it’s not necessary to manually press read to see the button presses.  It’s shown instantly in GUI as ticks in the zone status.  Alarm 1 for press, alarm 2 for double press, and alarm 1+2 for press hold.
[image]
[image]
[image]

I’ve tried binding the IAS cluster to a group in hopes that I’ll get usable values in rest, but no luck.  So everything looks like its working within GUI, just can’t get it to publish the values in rest.

blueleader07, 2018-12-28:
Any chance this will be available soon?  Looks like manup has provided the code?
https://github.com/mysteryan/deconz-rest-plugin/commit/876af0f427521ead699aa6e1e197a18a333d6510

Excited to use this one.  I'm in the same boat where I see the Alarm 1 / Alarm 2 status being updated ... but can't get the device added to the REST API.

qtjupsrkb, 2019-07-02:
Is support for the Samjin button like to be added in the future?

manup, 2019-07-03:
>Is support for the Samjin button likely to be added in the future?

Yes, we have it here for testing, but there is no ETA yet.

qtjupsrkb, 2019-07-03:
Thanks for the update @manup !

That's exciting to hear :-)

keithcroshaw, 2019-07-29:
Hello, Thanks for all the hard work going on here. 
I have one of these buttons that I'm looking to integrate.
Has it made it out of the lab yet?

elmgrenen, 2019-08-01:
^Wondering the same thing as keithcroshaw

indrora, 2019-09-15:
Is there a way we can test this using, e.g., Home Assistant?

keithcroshaw, 2019-09-15:
Was support released @indora?
I'll gladly test if it was.

manup, 2019-09-15:
The SmartThings button isn't integrated yet, but will be very soon :)

mysteryan, 2019-09-16:
Button press, double press and hold seems now seems to fire the appropriate buttonevent in the latest commit.  Temperature reports also seem to be working properly now.  Thanks.

closed 2019-09-16 by mysteryan

manup, 2019-09-16:
Cool, thanks for testing.

elmgrenen, 2019-09-16:
How do I add this to Home Assistant? I use conbee2 with deconz plugin so I use phoscon app

manup, 2019-09-16:
You'll need to wait for the next version 2.05.70 (or compile the plugin).
The button is then available in the same manner as other switch resources.

ecfitzgerald, 2019-12-04:
I have 2.05.71 and I can add the button, but in Phoscon its in Switches, but if I try to add the switch to a group with the switch editor, the switch does not appear.  I see the temperature and battery in Hassio.  I suspect the not seeing the switch in the switch editor is the primary symptom, I can see my Ikea switch and use it in phoscon and hassio, but not this.

keithcroshaw, 2019-12-04:
I wonder if it shows up as an event in Home Assistant? I wish that all buttons would show up as binary sensor entities.

bisby, 2019-12-04:
It does indeed show up as an event in Home Assistant. single, double, and long presses are all different event types.

```json
{
    "event_type": "deconz_event",
    "data": {
        "id": "button",
        "unique_id": "1",
        "event": 1004
    },
    "origin": "LOCAL",
    "time_fired": "2019-12-04T03:30:23.164600+00:00",
    "context": {
        "id": "1",
        "parent_id": null,
        "user_id": null
    }
}
```

"event" was 1002 for single press, 1004 for double press, and 1001 for long press.

mysteryan, 2019-12-05:
> I have 2.05.71 and I can add the button, but in Phoscon its in Switches, but if I try to add the switch to a group with the switch editor, the switch does not appear. I see the temperature and [...]

Treat it like a sensor and create a rule to toggle an existing group according to its buttonevent.

ebaauw, 2020-04-11:
Does this device support Tamper detection and low battery in the IAS Zone Status?  The code was inconsistent that `state.tampered` and `state.lowbattery` were not exposed on pairing, but were on restore from the database.  I now don't expose them for both situations, see https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2672#issuecomment-612491012.

mattiasflodin, 2023-01-03:
This is listed in the [support page](https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Supported-Devices) but I see no indication in the issue whether it was actually verified as working. Is it supposed to work? Because I'm having no success pairing this in deconz 10.0.0.4  (FW version 261F0500).