dresden-elektronik/deconz-rest-plugin #994: Some OSRAM lights randomly rebooting(?) with firmware 0x262f0500

markuspf, 2018-11-30:
I recently updated my RaspBee's firmware to 0x262f0500. Since the update 2 of my OSRAM bulbs (one RGBW and one E14 candle) randomly flicker and then stay on. I can switch them off, but they come on at some point later.

I had this problem before but it went away. Sadly I don't know the firmware version that had this problem previously.

Any hints?

ebaauw, 2018-12-01:
Hints, or rather long shots:
- Check the firmware version of the bulbs: are they different from your other bulbs that don’t show this behaviour;
- Try turning them off specifying the transitiontime of 0.  This should cause deCONZ to send a ZigBee _Off_ command, instead of the _Move to level with on/off_ with an added default transitiontime of 0.4 seconds.  There was a bug in OSRAM firmware V1.03.07 that caused the bulb to remain on (turn back on?) when receiving a _Move to level with on/off_.
- To investigate further, you might try different ZigBee commands with different parameters from the deCONZ GUI.  Use the _Cluster info panel_ on the _On/Off_ and on the _Level control_ clusters.

I’m seeing something similar with my OSRAM Gardenlights: I turn them off, but they spontaneously turn on after a few seconds.  When I set the brightness to 1, and then turn them off, they stay off.  Because of this workaround, I never fully investigated what’s happening, which ZigBee commands are causing this.  My guess would be they don’t like the transitiontime that deCONZ adds to each command.

I don’t see how a different version of the RaspBee or ConBee firmware would impact this. As far as I’m aware, the logic what command to send to a light to turn it off hasn’t been changed in any recent version of the REST API plugin.

markuspf, 2018-12-01:
> > I recently updated my RaspBee's firmware to 0x262f0500.
> 
> Which version of deCONZ are you running?
> 

2.05.48

> > I don’t see how a different version of the RaspBee or ConBee firmware would impact this. As far as I’m aware, the logic what command to send to a light to turn it off hasn’t been changed in any recent version of the REST API plugin.
> 
> Hard to tell might be not related to On/off cluster commands, a while ago I've seen OSRAM E27 bulbs do a power cycle for unknown reasons and light goes to default on. Only a guess but there might be a bug in the light firmware which doesn't like some commands and may trigger a watchdog after some time. Could be anything from ZDP to ZCL commands. deCONZ for sure sends way more requests for maintenance to the device then the OSRAM gateway. In deCONZ core there is already code which stops querying OSRAM lights for neighbor tables after some time to relax situation but it might not be enough.

My OSRAM bulbs (6 E27 RGBW, 5 E14 "candles") all have the same firmware version (and I havent updated them in months since it requires installing the osram gateway, pairing the lights etc), but only one of the E27 and one of the E14 has the above mentioned problem

More interesting things: In deCONZ the bulb in question has a red bar next to it, and in the Phsocon app its greyed out as unreachable, but if I switch the group it is in on and off it just comes on and off with the rest of the group. Sending commands to the bulbs from deCONZ fails with "sending failed".

mysteryan, 2018-12-03:
Had same exact issues with two out of 18 osram bulbs. A TW BR30 and a RT TW Osram that turns on spontaneously and only responds to group command on firmware 2.05.39+.  Problem dissapeared after a reset and re-paired.

markuspf, 2018-12-09:
This "solution" has the obvious flaw that I have to keep resetting and reconnecting bulbs if they start misbehaving after an update.

grover, 2018-12-11:
I can confirm the same issue with my OSRAM TW50 PAR16. Come on and stay on randomly, a group of 5 came on overnight (luckily in the hallway) and stayed on. Today had another incident: 4 came on, turned them off and a couple of minutes two of them turned back on.

Curiously the REST resource for the light thinks its off (e.g. state.on is false).

deCONZ 2.05.49 and FW 262F0500

markuspf, 2018-12-14:
Just for completeness: Resetting the bulbs does indeed work; it's of course still quite irritating, but might also not be deCONZ' fault, after all Osram lights have proven (to me) very flimsy when it comes to reliability.

manup, 2018-12-25:
Does the issue still occur with 2.05.52? I have a OSRAM E27 in my setup since a few days now. No problems so far. In the past this bulb did turn on on it's own.

closed 2019-05-01 by stale[bot]