dresden-elektronik/deconz-rest-plugin #996: HEIMAN Gas Detector HS1CG M

ACrazyConcept, 2018-12-01:
I bought the HEIMAN Gas Detector HS1CG M.
I can pair it in Phoscon. And even though it doesn't show up in the sensor list in Phoscon, it does appear in deCONZ as GAS_V15.

It then works for a while, but then it loses the connection after a while and turns red. I can re-pair it and it will work again. But it's getting a bit old to do it continuously.

closed 2019-04-07 by stale[bot]

kentania, 2019-05-04:
Hi everyone.
I installed a Conbee II on Raspberry Pi 3 b +, through phoscon app I managed to connect almost everything, I have a gas sensor hs1cg-m of heiman that I can't associate in any way. In the compatibility list is heiman hs1cg present, is there a difference with hs1cg-M? Is there any particular procedure to associate it? Is anyone already using it?
Tips?

SwoopX, 2020-02-02:
This post might hold the solution: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2403#issuecomment-581164140