dresden-elektronik/deconz-rest-plugin #997: Livolo zigbee devices

khzd, 2018-12-03:
Hello all
I would like to know if the new livolo zigbee switch is planned to be supported 
Cheers

manup, 2018-12-04:
Not on the list yet, if you can provide more details as described in the Wiki it would be helpful.

https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Request-Device-Support

Smanar, 2018-12-04:
Nice device ! You have one ?
It seem it s a switch with dimmer and without neutral, glass desing, realy interesting.

khzd, 2018-12-04:
Yes I have one for testing purpose , without neutral and round back for European wall hall :)
Is it with dimmer ?

khzd, 2018-12-04:
I plan to order your zigbee dongle if you support livolo , so I can’t provide you the wiki procedure.

Smanar, 2018-12-05:
Yes, It seem at least some of them works with dimmer.
But unfortunately it's a realy new device, so it can take long time before someone having a deCONZ device buy one. For me I have just buy the legrand model so I will not buy this model now ^^

closed 2019-04-11 by stale[bot]

fantomnotabene, 2019-05-30:
@Smanar I have these switches (with one and two buttons) and want to provide details. But they works only on 26th channel. How can I switch my RaspBee to it? Web interface offers 11, 15, 20 and 25.

Smanar, 2019-05-30:
If you are looking on deconz option, you can choose the level 26.
But it is perhaps disabled because of reglementation on some countries.

Fifke, 2019-06-06:
I checked the settings (in Belgium) but was not enable to choose channel 26. I have a lot off these switches, but still not able to connect them to conbee.

Smanar, 2019-06-07:
On deconz, not Phoscon, you haven't channel 26 (with pressing F9) ?

Fifke, 2019-06-16:
I have got my conbee up and running with home assistant on a raspberry pi. In deconz i can choose channel 25 will try one of thecomming days if it finds livolo switch. I have xiaomi mi eu version sensors connected with it but cant connect to alexa wich would be great... Still got to figure that out...

fantomnotabene, 2019-06-17:
@Fifke no, it must be 26th ;( 
I tried 25th.

Fifke, 2019-06-17:
Ok will try that .

Fifke, 2019-06-17:
Nothing to find on ch 26 in deconz

Smanar, 2019-06-17:
I m the only one with that ?
[image]

Fifke, 2019-07-26:
I have channel 26 , but no switches are found...

vladbisceanu, 2019-10-13:
Hi guys,

Has any progress been done on this integration?

I've the Conbee II stick set up with Home Assistant and I want to order some Livolo Light Switches (without their gateway), but from what I rad online, I am unsure if they will work.

Thanks!

jpiquee, 2019-10-17:
Hello, i just bought a Livolo simple switch without dimmer and impossible to include into jeedom with deconz key (no channel 26). Please help this switch are really nice and work well.

Aikidyst, 2019-12-01:
Hello,
jsut today succesufully connected LIVOLO Zigbee TI0001 switch to DeCONZ.
set channel to 26 and changed ext pan id. (jsut google about livolo channel and pan id)
to change channel you need to follow deCONZ documentation (pdf): leave network, change settings save and create new network. you will need to recconect all of your devices again.
After that you will see Livolo devices at deConz. 
BUT at above will bring this device to be actually working. AS i think RESTAPI need to be updated to iclude this devices as supported.

Starbase12, 2019-12-04:
Good evening,

it would be very nice if it's possible to pair them with Deconz/ConBee II. I'm new using deconz and maybe Aikidyst or someone else can explain it in more detail how to make livolo zigbee wallswitches pairable. Thanks in advanced.

delitero, 2020-02-16:
@Aikidyst what ext pan id have you used? I'm being able to find the livolo switch in deconz but HA plugin always fail to connect the switch.

Cyrille63, 2020-03-21:
Hi,
Like delitero I find the livolo Switch in deconz but Home Assistant plugin fails to find the livolo switch
Help please, thanks in advanced

delitero, 2020-03-21:
> Hi,
> Like delitero I find the livolo Switch in deconz but Home Assistant plugin fails to find the livolo switch
> Help please, thanks in advanced

I was never been able to make it connect through deconz. However, i had no issue with CC2531 USB stick and mqtt.

Smanar, 2020-03-22:
Someone have already succed to pait a livolo switch with deconz ?
This one for exemple is not complete https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2036#issuecomment-602075993