dresden-elektronik/deconz-rest-plugin #998: Xiaomi round push button

alex0411, 2018-12-03:
At the moment I only get (1000, 1002, 1004, 1005, 1006) from the REST API. Model is Xiaomi WXKG01LM.
With zigbee2mqtt I could use this clicks: single, double, triple, quadruple, many, long, long_release click.
Right now click and long respond both with 1000.

deCONZ 2.05.49

_Originally posted by @alex0411 in https://github.com/dresden-elektronik/deconz-rest-plugin/issues/349#issuecomment-443799184_

manup, 2018-12-03:
Can you please provide the debug output while doing the above actions, we can check the code to add more actions.

    deCONZ --dbg-aps=2 --dbg-zcl=1 --dbg-info=1

manup, 2018-12-03:
Also please continue discussion in the existing issue for the switch, closing this for now.

Related WXKG01LM issue: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/917

alex0411, 2018-12-05:
Yes, this is the debug output

single click
`21:35:53:296 APS-DATA.indication srcAddr: 0x00158d0001f3ac97, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0006, lqi: 143, rssi: -73
21:35:53:296    asdu: 18070a00001000
21:35:53:296 button 1000 Normal press
21:35:53:296 ZCL attribute report 0x00158D0001F3AC97 for cluster 0x0006, ep 0x01
21:35:53:299 discard sensor state push for 11: state/lastupdated (already pushed)
21:35:53:504 APS-DATA.indication srcAddr: 0x00158d0001f3ac97, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0006, lqi: 143, rssi: -73
21:35:53:504    asdu: 18080a00001001
21:35:53:504 button 1002 Normal release
21:35:53:504 ZCL attribute report 0x00158D0001F3AC97 for cluster 0x0006, ep 0x01
21:35:53:507 discard sensor state push for 11: state/lastupdated (already pushed)`

double
`21:33:50:444 APS-DATA.indication srcAddr: 0x00158d0001f3ac97, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0006, lqi: 167, rssi: -70
21:33:50:445    asdu: 18050a00802002
21:33:50:445 button 1004 Double press
21:33:50:446 ZCL attribute report 0x00158D0001F3AC97 for cluster 0x0006, ep 0x01
21:33:50:449 discard sensor state push for 11: state/lastupdated (already pushed)`

triple
`21:35:12:115 APS-DATA.indication srcAddr: 0x00158d0001f3ac97, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0006, lqi: 119, rssi: -76
21:35:12:115    asdu: 18060a00802003
21:35:12:116 button 1005 Triple press
21:35:12:116 ZCL attribute report 0x00158D0001F3AC97 for cluster 0x0006, ep 0x01
21:35:12:119 discard sensor state push for 11: state/lastupdated (already pushed)`

quad
`21:36:20:616 APS-DATA.indication srcAddr: 0x00158d0001f3ac97, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0006, lqi: 135, rssi: -74
21:36:20:616    asdu: 18090a00802004
21:36:20:616 button 1006 Quad press
21:36:20:617 ZCL attribute report 0x00158D0001F3AC97 for cluster 0x0006, ep 0x01
21:36:20:619 discard sensor state push for 11: state/lastupdated (already pushed)`

many
`21:37:06:358 APS-DATA.indication srcAddr: 0x00158d0001f3ac97, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0006, lqi: 167, rssi: -70
21:37:06:358    asdu: 180c0a00802080
21:37:06:359 no button handler for: lumi.sensor_switch ep: 0x01 cl: 0x0006 cmd: 0x0A pl[0]: 000
21:37:06:359 ZCL attribute report 0x00158D0001F3AC97 for cluster 0x0006, ep 0x01
`

long
`21:37:48:452 APS-DATA.indication srcAddr: 0x00158d0001f3ac97, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0006, lqi: 127, rssi: -75
21:37:48:452    asdu: 180f0a00001000
21:37:48:452 button 1000 Normal press
21:37:48:453 ZCL attribute report 0x00158D0001F3AC97 for cluster 0x0006, ep 0x01
21:37:48:455 discard sensor state push for 11: state/lastupdated (already pushed)`

long_release
`21:38:34:100 APS-DATA.indication srcAddr: 0x00158d0001f3ac97, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0006, lqi: 87, rssi: -80
21:38:34:100    asdu: 18140a00001001
21:38:34:100 button 1002 Normal release
21:38:34:100 ZCL attribute report 0x00158D0001F3AC97 for cluster 0x0006, ep 0x01
21:38:34:103 discard sensor state push for 11: state/lastupdated (already pushed)
`

alex0411, 2018-12-05:
I found out how it is done with zigbee2mqtt:
https://github.com/Koenkk/zigbee-shepherd-converters/blob/master/converters/fromZigbee.js#L237

manup, 2018-12-05:
Thanks that helps, so they basically zigbee2mqtt uses a timer to determine if it is a long press, when the release event isn't received within 300 ms. In that case they fire a virtual `hold` event.

alex0411, 2018-12-05:
No problem. And for many there is also no button handler at the moment (see debug output).

I also have the strang behavior, that a single click in rare cases gets recognized as 1002.

cnrd, 2018-12-14:
Please do not implement a virtual hold for the rest API unless this can be implemented while still sending the 1000 (button down) and 1002 (button up) events, I'm using these buttons with Home Assistant and an appdaemon script where I can define custom virtual hold events: https://gist.github.com/cnrd/da1e4b970bfd3fee208a3014211736b6 this would not be possible without getting the 1000 and 1002 events.

cnrd, 2019-01-13:
Mine only seems to report 100% battery and some of them haven't had a battery change for about 2 years.

manup, 2019-01-13:
The battery level is based on a mapping of the reported voltage. The problem is that the voltage will only drop significantly shortly before the battery dies.

[image]

manup, 2019-01-20:
The switch now support all possible events as described in:
 https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Xiaomi-WXKG01LM

Further the switch is supported in the Phoscon App switch editor since 2.05.57:
https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/V2_05_57

Closing the issue for now.

closed 2019-01-20 by manup

chimpy, 2019-01-20:
You’re awesome!

alex0411, 2019-01-21:
Very nice, thanks!