dresden-elektronik/deconz-rest-plugin #999: Raspbee stuck on joining

OshriDan, 2018-12-03:
After an update of the raspberry pi 4 days ago my Raspbee transmitter is not working any more :(
It's continuing blinking red and the gui shows that it's joining but this state does not change.
I can see all my zigbee lights and sensors but my router is shown in yellow (used to be blue) and there are no (green) connection any more.
this is what i've tried:
1.reset raspbee.
2.flash new firmware.
3.install new deconz rest api
4.reset about every thing I can.
5 factory reset to bridge via deconz

current deconz version V2_05_49 and raspbee firmware version 2620f0500

please help !!!

manup, 2018-12-03:
Can you please post a screenshot of the deCONZ network settings?
Seems that they are messed up somehow.

OshriDan, 2018-12-04:
[image]

manup, 2018-12-04:
Ok don't know why it is configured as router but this seems to be the problem.

Please do the following:
* Device Type: Coordinator
* Save
* Done
* Leave
* Join

This should bring the network in working state with the red led on permanently.

OshriDan, 2018-12-04:
Hi thanks for the help!!!
I've changed to Coordinator and the raspbee became blue but I'm missing the join,leave ... buttons.
Any idea ?

OshriDan, 2018-12-04:
I use to have them but now there gone, all icons the all row actually.
I don't know how to get it back.
When I look on the website Phoscon on gateway next to firmware I see not connected.
Strange...

manup, 2018-12-04:
[image]

Should look like this, the buttons can't disappear, maybe the window is not fully visible?

OshriDan, 2018-12-04:
ok..managed to get it back but I see only "joining.." in blue and "leave" and "join" are grayed out so I cannot leave.
I think the raspbee is stuck on joining state or something.
I've tried to flash latest firmware and reset but still no luck..

manup, 2018-12-04:
Hmm weird, the Joining should stop after a few failed attempts.

Please try:

* Devicy Type: Coordinator
* Save

and on commandline:

    GCFFlasher_internal -r

It should have the same effect as Leave + Join.

OshriDan, 2018-12-04:
Thank you for all the help but still no change..
Tried both (cmd and vie gui) still stuck on joining , other suggestions ?

manup, 2018-12-04:
Could it be that after the upgrade you serial port is not configured/used as before?

You may check configuration as described here:
https://github.com/dresden-elektronik/deconz-rest-plugin#software-requirements

OshriDan, 2018-12-04:
Nop serial port is set correctly.
It's very strange because I can see all lamps (as yellow nodes) but don't have green connection between them and the gate and cannot control them as well.
I even tired blowing air in the raspbee and check if it's all connected well.
It just won't stop trying to join.

manup, 2018-12-04:
The lights are just visible since they are loaded from the database not actually due ZigBee traffic.

Something is really wrong here, it shouldn't Joining infinitely.

Do you have a second sd-card available to test if the official image works?

OshriDan, 2018-12-04:
I'll test tomorrow ... have to go to sleep ;)
Keep you updated.. again thanks a lot for the help !

OshriDan, 2018-12-05:
Well.. tried with raspbee img and same result.. gui is stuck on joining and no other nodes are shown...
Is it time to conclude that I could barry the raspbee in my backyard, is he... dead?
Any other ideas are welcome :)

OshriDan, 2018-12-06:
Update..
I've installed zshark and there I could connect to the network and received packets.
The red light also stopped blinking then.
This is my config (via get from the api) maybe it helps.
`    "config": {
        "UTC": "2018-12-06T21:33:17",
        "apiversion": "1.16.0",
        "backup": {
            "errorcode": 0,
            "status": "idle"
        },
        "bridgeid": "00212EFFFF0137EB",
        "datastoreversion": "60",
        "devicename": "RaspBee",
        "dhcp": true,
        "factorynew": false,
        "fwversion": "0x262f0500",
        "gateway": "192.168.2.1",
        "internetservices": {
            "remoteaccess": "disconnected"
        },
        "ipaddress": "192.168.2.12",
        "linkbutton": false,
        "localtime": "2018-12-06T22:33:17",
        "mac": "b8:27:eb:df:9f:05",
        "modelid": "deCONZ",
        "name": "Phoscon-GW",
        "netmask": "255.255.255.0",
        "networkopenduration": 180,
        "panid": 34353,
        "portalconnection": "disconnected",
        "portalservices": false,
        "portalstate": {
            "communication": "disconnected",
            "incoming": false,
            "outgoing": false,
            "signedon": false
        },
        "proxyaddress": "none",
        "proxyport": 0,
        "replacesbridgeid": null,
        "rfconnected": false,
        "starterkitid": "",
        "swupdate": {
            "checkforupdate": false,
            "devicetypes": {
                "bridge": false,
                "lights": [],
                "sensors": []
            },
            "notify": false,
            "text": "",
            "updatestate": 0,
            "url": ""
        },
        "swupdate2": {
            "autoinstall": {
                "on": false,
                "updatetime": ""
            },
            "bridge": {
                "lastinstall": "",
                "state": "allreadytoinstall"
            },
            "checkforupdate": false,
            "install": false,
            "lastchange": "",
            "lastinstall": "",
            "state": "allreadytoinstall"
        },
        "swversion": "2.5.49",
        "timeformat": "12h",
        "timezone": "Etc/GMT-1",
        "uuid": "c72169db-8511-43cc-9582-1c52e607cb2a",
        "websocketnotifyall": true,
        "websocketport": 443,`

please help.

manup, 2018-12-06:
Looks like the NVRAM (EEPROM) of your RaspBee is in a invalid state and for some reasons setting it from Network Settings doesn't help.

I'll add a command to the next firmware to clear NVRAM reverting it back to factory new state. Please stay tuned.

OshriDan, 2018-12-11:
Hey @manup  :)
I know you are probably very busy with other issues...but I was wondering if you have any rough estimation when does the nvram reset functionality would be available?
My all house is kind of dimmed for a 2 weeks now since I don't have raspbee working anymore :(
Really appreciate the help!!

Thanks in advance :)

manup, 2018-12-11:
Hi, I see to get it into the next version arriving this week (or at least provide small python script to trigger the reset). Understand the situation, fingers crossed.

OshriDan, 2018-12-11:
Amazing!!!
Thank you!

manup, 2018-12-12:
Small question while I'm looking in the code: After a power-cycle of the RaspBee is the device type in the network settings shown as **Router** (I assume yes).

RaspBee power-cycle:

    sudo GCFFlasher_internal -r

manup, 2018-12-12:
Here is a slightly modified firmware, please install it and afterwards try the following:
* Open deCONZ Network Settings
* Select Device Type: Coordinator
* Save
* Done
* Edit > Reboot Device

https://www.dresden-elektronik.de/rpi/deconz-firmware/deCONZ_Rpi_0x26300500.bin.GCF

OshriDan, 2018-12-13:
Yes it does return every time to Router mode.
Great ! Ill give it a shot after work and report back :)

OshriDan, 2018-12-13:
It works !!!
Everything works now and I can join the network :)
Thank you so much man you saved me :)))

closed 2018-12-13 by OshriDan