dresden-elektronik/deconz-rest-plugin #1001: Philips Dimmer not sending events?

FezVrasta, 2018-12-04:
Hi, I have just paired the Philips Hue Dimmer to my DeCONZ, I see it in the app, and I see the battery status on Home Assistant.

I first tried to assign some lights to the dimmer in the Phoscon App but it didn't seem to work (lights didn't turn on/off).

I then removed the lights from there and tried to listen for the events on Home Assistant and I can't see any `deconz_event` happening when I press the buttons.

Am I missing some step?

manup, 2018-12-04:
Which deCONZ and firmware version are you using?
There were some related fixes in recent versions.

FezVrasta, 2018-12-04:
I think I'm outdated, I'll update and report back, thanks

FezVrasta, 2018-12-05:
@manup I upgraded firmware and software but I still can't retrieve any event from the dimmer. (all my other Zigbee devices work just fine)

manup, 2018-12-05:
You likely need to delete and rejoin the dimmer so that setup and bindings can be made.

FezVrasta, 2018-12-05:
Thanks that did it!

closed 2018-12-05 by FezVrasta