dresden-elektronik/deconz-rest-plugin #1002: Scenes with Amazon Echo / Alexa

mozzbozz, 2018-12-04:
Wondering if it is possible to call the scenes created in Phoscon with Alexa somehow? Connecting the raspbee gateway with the Amazon Echo worked without a problem. I can also control the lights and groups, but the scenes aren't shown (unlike before, when I used a Philips Hue bridge).

Is that expected behaviour or did I miss to set something up? (maybe related to this issue, not sure if I understand it correctly: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/29)

Thanks!

ebaauw, 2018-12-04:
The Hue bridge API for scenes is very different from the deCONZ API.  We would need to mimic the Hue API, and, more importantly and more work, the way scenes are setup before they can be recalled from a client expecting the Hue API.

KodeCR, 2018-12-10:
Like #1007, I believe you need the hue skill for scene support which doesn’t work with deconz because of the proprietary cloud login. The native hue support only does on off and brightness I think.

An Alexa smart home skill is not too complex to do, maybe someone want to take on that project.

closed 2019-04-16 by stale[bot]

mozzbozz, 2019-04-27:
I finally had some time looking into this. Especially I researched how to implement the Alexa Smart Home API. However, it seems like you'd always need to go the way via the cloud to set the scenes, which is not really what I want to do - e.g. the original Hue Skill was very unreliable for me when I still used the Hue Bridge and I was happy to be able to locally control the Raspee with Alexa. It also reacts way faster to voice commands than going via another cloud instance (apart from voice recognition).

I came up with the idea that the deCONZ API should just expose the scenes as "lights" / devices. So, if I e.g. have a scene "TV scene", I'd say "Alexa, TV scene on" and everything would just work as expected - even though Alexa has no idea, that's a scene, not a real device.

What do you think about that idea? Is that viable? Do you see any problems with that (maybe you know more about the side effects a change like this would have)?