dresden-elektronik/deconz-rest-plugin #1004: Conbee does not communicate

alex1891, 2018-12-04:
Hey all,
I have problems with my deconz setup. I updatet the firmware of my conbee usb stick to the newest firmware (deCONZ_Rpi_0x262f0500.bin.GCF ) an I am still not able to get values from my xiaomi sensors an my ikea bulbs. The log shows no errors. I updatet deconz to the newest version, too. 

I am using deconz on my Raspberry in a docker container.

Best reguards
Alex

manup, 2018-12-04:
Can you please provide the logs including deCONZ startup?

alex1891, 2018-12-04:
Sure:
`[marthoc/deconz] Starting deCONZ...,
[marthoc/deconz] Current deCONZ version: 2.05.49,
[marthoc/deconz] Web UI port: 80,
[marthoc/deconz] Websockets port: 443,
libpng warning: iCCP: known incorrect sRGB profile,
This plugin does not support propagateSizeHints(),
This plugin does not support propagateSizeHints(),
This plugin does not support propagateSizeHints(),
21:14:12:901 HTTP Server listen on address 0.0.0.0, port: 80, root: /usr/share/deCONZ/webapp/,
21:14:12:916 CTRL. 3.16.221:14:13:001 dev /dev/ttyAMA0,
21:14:13:001 ZCLDB init file /root/.local/share/dresden-elektronik/deCONZ/zcldb.txt,
21:14:13:180 parent process /bin/sh,
21:14:13:180 gw run mode: normal,
21:14:13:180 GW sd-card image version file does not exist: /root/.local/share/dresden-elektronik/deCONZ/gw-version,
21:14:13:181 sd-card cid: 1b534d303030303010ffffffff00f337,
21:14:13:182 DB sqlite version 3.16.2,
21:14:13:184 DB PRAGMA page_count: 38,
21:14:13:184 DB PRAGMA page_size: 4096,
21:14:13:184 DB PRAGMA freelist_count: 3,
21:14:13:184 DB file size 155648 bytes, free pages 3,
21:14:13:184 DB PRAGMA user_version: 6,
21:14:13:184 DB cleanup,
21:14:13:185 DB create temporary views,
21:14:13:234 don't close database yet, keep open for 900 seconds,
22:14:13:235 started websocket server at port 443,
22:14:13:241 discovery updated announce interval to 10 minutes,
22:14:13:246 found node plugin: libde_rest_plugin.so - REST API Plugin,
22:14:13:249 found node plugin: libde_signal_plugin.so - Signal Monitor Plugin,
22:14:13:267 found node plugin: libstd_otau_plugin.so - STD OTAU Plugin,
22:14:13:297 dev /dev/ttyAMA0,
22:14:13:350 discard sensor state push for 1: state/dark (already pushed),
22:14:13:352 discard sensor state push for 1: state/status (already pushed),
22:14:13:353 discard sensor state push for 1: state/lastupdated (already pushed),
22:14:13:953 dev /dev/ttyAMA0,
22:14:13:981 dev /dev/ttyAMA0,
22:14:14:039 Device firmware version 0x262F0500,
22:14:14:057 unlocked max nodes: 200,
22:14:14:346 Device protocol version: 0x0109,
22:14:14:369 new node - ext: 0x00212effff024590, nwk: 0x0000,
22:14:14:375 Node: id: 1, 00:21:2e:ff:ff:02:45:90 (0x00212EFFFF024590) scene: 107.000000, 1.000000,
22:14:14:378 Node: id: 2, 7c:b0:3e:aa:00:ad:0a:5d (0x7CB03EAA00AD0A5D) scene: 65.000000, 13.000000,
22:14:14:387 Node: id: 3, 00:0b:57:ff:fe:8e:3a:2c (0x000B57FFFE8E3A2C) scene: 131.000000, -94.000000,
22:14:14:398 Node: id: 4, 00:15:8d:00:02:37:62:27 (0x00158D0002376227) scene: -182.000000, -34.000000,
22:14:14:407 Node: id: 5, 00:0b:57:ff:fe:73:09:20 (0x000B57FFFE730920) scene: 185.000000, -56.000000,
22:14:14:418 Node: id: 6, 00:0b:57:ff:fe:8e:3b:82 (0x000B57FFFE8E3B82) scene: -48.000000, 19.000000,
22:14:14:429 Node: id: 7, 00:0b:57:ff:fe:8b:5c:01 (0x000B57FFFE8B5C01) scene: -166.000000, 97.000000,
22:14:14:440 Node: id: 8, 00:15:8d:00:02:32:05:63 (0x00158D0002320563) scene: -70.000000, 17.000000,
22:14:14:449 Node: id: 10, 00:15:8d:00:02:37:12:d1 (0x00158D00023712D1) scene: 125.000000, -10.000000,
22:14:14:458 Node: id: 11, 00:15:8d:00:02:38:7a:ce (0x00158D0002387ACE) scene: -162.000000, -86.000000,
22:14:14:467 Node: id: 12, 00:15:8d:00:02:33:fb:83 (0x00158D000233FB83) scene: -54.000000, -134.000000,
22:14:14:478 DB save zll database items 0x00000880,
22:14:14:497 DB saved in 18 ms,
22:14:14:521 don't close database yet, keep open for 900 seconds,
22:14:14:521 LightNode 1: Fernseher added,
22:14:14:522 SensorNode 2 set node 0x7cb03eaa00ad0a5d,
22:14:14:522 DB save zll database items 0x00000800,
22:14:14:532 DB saved in 10 ms,
22:14:14:555 SensorNode 7 set node 0x000b57fffe8e3a2c,
22:14:14:555 DB save zll database items 0x00000800,
22:14:14:566 DB saved in 10 ms,
22:14:14:588 SensorNode 3 set node 0x00158d0002376227,
22:14:14:589 SensorNode 4 set node 0x00158d0002376227,
22:14:14:589 SensorNode 5 set node 0x00158d0002376227,
22:14:14:589 DB save zll database items 0x00000800,
22:14:14:600 DB saved in 10 ms,
22:14:14:621 SensorNode 6 set node 0x000b57fffe730920,
22:14:14:622 DB save zll database items 0x00000800,
22:14:14:633 DB saved in 11 ms,
22:14:14:656 don't close database yet, keep open for 900 seconds,
22:14:14:656 LightNode 2: Kugel added,
22:14:14:656 DB save zll database items 0x00000800,
22:14:14:667 DB saved in 10 ms,
22:14:14:690 don't close database yet, keep open for 900 seconds,
22:14:14:691 LightNode 3: Deckenlampe added,
22:14:14:691 DB save zll database items 0x00000800,
22:14:14:702 DB saved in 10 ms,
22:14:14:723 SensorNode 8 set node 0x00158d0002320563,`

manup, 2018-12-04:
Connection between deCONZ and device looks ok so far.

Do I understand correctly that the network isn't working at all or are "just" some devices aren't working as expected?

alex1891, 2018-12-04:
I have some lights (IKEA) and some xiaomi sensors and no of them are working right now. the app works fine, i see all my devices but i receive no data from the sensors and can´t control my lights.
I tried to pair a xiaomi window sensors -> does not work
Triied to pair a ikea bulb -> does not work

manup, 2018-12-04:
Ok, so a few things to check.

In *Phoscon App > Menu > Settings > Gateway* it should look like this:
* Firmware version shown
* Red LED on the shield on: indicating network up

[image]

Is the ConBee connected directly to the Raspberry? If possible you may try a usb extension cable.

alex1891, 2018-12-04:
Okay it looks slightly different:
[image]

Does the usb stick has and led? i can´t find one

manup, 2018-12-04:
Install the windows version on your notebook and plugin the ConBee.

https://www.dresden-elektronik.de/deconz/win/deCONZ_Setup_Win32_V2_05_49.exe

Then please make a screenshot of the deCONZ Network Settings dialog so we can check and fix the configuration.

alex1891, 2018-12-04:
Here we go:
[image]

manup, 2018-12-04:
The Network tab is the interesting one :)

alex1891, 2018-12-04:
hupsi :)
[image]

manup, 2018-12-04:
Hmm strange no channel selected..

* Select channels 11, 15, 20, 25
* Save
* Close
* Join

The network should be up then and one of the channels will be selected.

You may have to rejoin your devices (on the docker install).
Names and rules should be preserved.

alex1891, 2018-12-04:
works! thank you for the support :)

manup, 2019-01-16:
Closing the issue for now.

closed 2019-01-16 by manup