dresden-elektronik/deconz-rest-plugin #1006: Conbee not showing up as a hue bridge

maxandersen, 2018-12-05:
I got my conbee working great with approx. 40 bulbs in a mix of Philips hue and ikea. All shows up nicely in homeassistant on a hassio. 

But one of the main reasons I went for conbee was the possibility of using existing hue apps with it. 

And that just don't work. Tried multiple apps using Discovery and add by ip and the bridge is never found. 

Is there something I need to enable? Some ports to open it?

manup, 2018-12-05:
Which version of deCONZ are you using? Be ware that not all features work with every app.
Most apps require deCONZ using port 80.

Hue Essentials for Android is happy with all ports. 

While looking for bridges you need to enable **Authenticate app** in the *Phoscon App > Settings > Gateway*.

[image]

maxandersen, 2018-12-05:
Using this version:

Version 2.05.22 / 4/18/2018
Firmware 261F0500

I'm on iOS - do yo know of an app here that can use different than Port 80?
. If port 80 is required I'll need some other way than hass.io :/

manup, 2018-12-05:
You may try 2.05.49 but I'm afraid it won't help here much, discovery is mostly done by the apps.

>I'm on iOS - do yo know of an app here that can use different than Port 80?

No, haven't looked into iOS apps for a while, there might be some though.

maxandersen, 2018-12-05:
Moving to Port 80 worked!

Hue app complains https not available but otherwise basic stuff works so far. Is there a https port default to enable ?

manup, 2018-12-05:
Https is currently not supported directly by deCONZ, some people managed to use a reverse proxy for that purpose.

closed 2018-12-06 by maxandersen