dresden-elektronik/deconz-rest-plugin #1010: How to disable the power consumption reporting from osram plug

Smanar, 2018-12-05:
Hi, I know this device haven't a working power consumption reporting, but 
+ I have see it's deCONZ that do forced pool to update this value https://github.com/dresden-elektronik/deconz-rest-plugin/issues/423
 + The websocket spam useless informations for it

There is a proper way to disable it ?
Will be enought setting the sensor to off ?

manup, 2018-12-05:
I think one approach might be that we disable the creation of this sensor as well as skip loading already existing sensors from the database.

Smanar, 2018-12-05:
But it's possible ATM ? I mean without files modification ?
Because I have tried setting it to off, not working, setting it to unreachable, not available.

I can edit the code to disable the creation of this sensor, but not sure it's a good solution, if there is another plug (even a working one) who use pooling, it will be the same problem.

A good solution will be to disable spamming 0 values if the plug is set to off ?

I haven't understand "skip loading already existing sensors from the database".

manup, 2018-12-05:
>But it's possible ATM ? I mean without files modification ?

Just delete the sensor resource :)

Smanar, 2018-12-05:
Hu ? Lol, not tried yet.
But If I don't press the pairing button, deconz will never tried to re pair it ?

manup, 2018-12-05:
Should work :)

Even while sensor search is active only joining (or power cycled) devices will result in sensor creation. So it should be fairly safe.

ebaauw, 2018-12-05:
> Just delete the sensor resource :)

With the later versions, won't this cacade delete the node?

To disable the spamming while keeping the resource, you could delete the binding, or change the attribute reporting settings from the deCONZ GUI.

I think we should probably comment out the `/sensors` whitelist entries for the OSRAM plugs (at least until they support power management), so only the `/lights` resource will be created.

Smanar, 2018-12-05:
From your own post Osram plug don't support attribute reporting on Active Power ^^, It's the first thing I have tried. Haven't take a look for binding.

I have deleted the sensor, It's working ATM

manup, 2018-12-05:
> With the later versions, won't this cacade delete the node?

Damn haven't thought of this one, yes you're right.

Smanar, 2018-12-05:
Ok, so bad news for me ^^. All works fine with deleting, I have tried all I can try, and still no problems.

But as I can understand, it will be not possible later. And I m not sure unbinding is possible on this device just for electrical mesurement.

closed 2018-12-06 by Smanar