dresden-elektronik/deconz-rest-plugin #1017: lumi.sensor_motion.aq2 comes with ZHAALarm?

wvuyk, 2018-12-09:


ebaauw, 2018-12-09:
Looking at the `uniqueid`, the ZHAAlarm is created for the _IAS Zone_ cluster.  In theory, it could raise alarms for low battery, tampering, or general alarm.  I don't know the Aqara motion sensor, so I wouldn't know whether this cluster reports anything useful in practice.  When did you last pair one of these?  The sensor resource is probably created now, because the improved reading of the node descriptors now reveals the _IAS Zone_ cluster.

Did deCONZ enroll the device?  If not, try writing the RaspBee/ConBee mac address to the _IAS_CIE_Address_.  What _Zone Type_ does it report?

wvuyk, 2018-12-09:
This is an motion sensor that has been working for a year. But with 2.0.5.49 (on an ikea routered network) it reported unreachable over and over. I do not think it was the cause of IKEA, but I limited (switched off) bulbs. This made the Aqara motion sensor become unreachable.
So I did reset the sensor and paired it to my RaspBee. This is where the ZHAAlarm was added. So indeed I guess you might be right, between a year ago and now this ZHAAlarm was added. Guess I will have to wait to see its purpose for now (battery might be most usefull besides percentage?)

manup, 2018-12-09:
Might be related: https://github.com/dresden-elektronik/deconz-rest-plugin/commit/f0ba5e017172b74cc1201bbc0f73b95da1a65b5f

wvuyk, 2018-12-10:
@manup thanks, it might. Will wait for a new version and will rejoin the sensor to see what happens.

manup, 2018-12-10:
Looks like the sensor might just have a newer firmware and also includes IAS Zone cluster

[image]

Note also application version 5

[image]

wvuyk, 2018-12-10:
Not here, pressing 0, pressing read, causing motion does not fill the basic cluster. IAS zone here is also filled. This sensor came in last year I think, should be a first version?

ebaauw, 2018-12-10:
Or the cluster had been there all along, but is only discovered by deCONZ with the improved pairing logic.  I’m seeing a lot of “new” clusters for many Xiaomi devices since these improvements.

manup, 2018-12-10:
Her is a sample from my other motion sensor bought in 2017

No IAZ Zone cluster and device type Dimmer Switch

[image]

Has application version 3

[image]

manup, 2018-12-10:
Sniffer Simple Descriptor response for the 2017 sensor

[image]

wvuyk, 2018-12-10:
Good old Hitchhikers Guide! LOL Still love the humor in it.

I had to go back into histroy here, but indeed the sensors (7 total) came in April this year, so they might be newer in version. Have been trying again to read the cluster here, but it does not go any further then getting the model identifier. For some reason it does not want to reveal its secrets :-)

[image]

manup, 2018-12-10:
The Xiaomi sensors are very quite, there are only two ways to query data (ZDP, ZCL)

* Within 500 milliseconds after the hourly special report is received
* Read a single attribute in cluster info, and quickly press the button on the sensor once
* For switches it seems to be possible to 5 times press a button to trigger a parent/mac poll request

Just added some code to automatically read the datecode of Xiaomi end-devices (after hourly report) if it works we can use this to queue pending tasks like the vibration sensor configuration in this manner.

wvuyk, 2018-12-10:
owww.. and update the sensitivity of the vibration sensor? That would be a nice one. Have one here, but it does not update very easy so far.

manup, 2018-12-10:
[image]

Cool, automatic query after hourly report seems to work.

A few notable things
* After special report is received (seq 202)
* Automatic read of datecode 
* The response has APS ACK enabled (might be handy haven't seen this before for Xiaomi sensors)
* Right after the ACK the sensor polls again (Data Request)

>owww.. and update the sensitivity of the vibration sensor? That would be a nice one. Have one here, but it does not update very easy so far.

Yes should be quite easy using the pending attribute, since we now know when the sensors are also listening we have a hopefully reliable window to configure the sensor (not only during setup).

wvuyk, 2018-12-18:
Upgraded to 2.05.52, these motion sensors still add the ZHAAlarm sensor, although no attributes for this sensor are updated (lastupdated="none")... 

Should we just ignore this for now, or surpress it?

MattL0, 2019-03-07:
Adding a xiaomi motion sensors still create a zhaalarm device.

any ways to delete this attribute?

MattL0, 2019-03-22:
Please don’t forget :p

closed 2019-07-27 by stale[bot]