dresden-elektronik/deconz-rest-plugin #1018: deCONZ hangs for 5 minutes during startup; /groups/0 not available

ebaauw, 2018-12-09:
Not sure what's happening here.  Shortly after starting deCONZ v2.05.49, it freezes completely for ~5 minutes.  The GUI doesn't respond; when minimising and maximising the GUI, it doesn't even redraw.  The log is quiet during this time, but afterwards, the messages appear with a timestamp from during the freeze (note the difference between syslog and deCONZ time:
```
Dec 09 16:19:51 pi2 deCONZ[13372]: 16:19:51:307 Node: id: 58, 84:18:26:00:00:0b:ac:51 (0x84182600000BAC51) scene: -962.057463, -950.942537
Dec 09 16:19:51 pi2 deCONZ[13372]: 16:19:51:313 Node: id: 59, 00:17:88:01:02:74:ba:72 (0x001788010274BA72) scene: -1200.000000, -100.000000
Dec 09 16:24:19 pi2 deCONZ[13372]: 16:19:51:323 Node: id: 60, 00:17:88:01:10:49:55:03 (0x0017880110495503) scene: 546.000000, -544.000000
Dec 09 16:24:19 pi2 deCONZ[13372]: 16:19:51:333 Node: id: 61, 00:17:88:01:02:00:ac:7c (0x001788010200AC7C) scene: -1437.000000, -394.339920
```

After the GUI coordinates have been loaded, I see a `PROTO: CRC error` message:
```
Dec 09 16:24:19 pi2 deCONZ[13372]: 16:19:51:833 Node: id: 99, 00:15:8d:00:02:39:14:5e (0x00158D000239145E) scene: 546.233293, -373.441510
Dec 09 16:24:19 pi2 deCONZ[13372]: 16:19:51:881 Node: id: 100, 00:15:8d:00:02:4a:70:18 (0x00158D00024A7018) scene: -1346.989853, -508.468633
Dec 09 16:24:19 pi2 deCONZ[13372]: PROTO: CRC error
Dec 09 16:24:19 pi2 deCONZ[13372]: 16:24:19:032 DB save zll database items 0x000008C3
Dec 09 16:24:19 pi2 deCONZ[13372]: 16:24:19:228 DB saved in 196 ms
```

I'm also seeing complaints about the NWK update ID:
```
Dec 09 16:24:26 pi2 deCONZ[13372]: 16:24:26:894 CTRL [00:15:8d:00:02:4a:70:18] found reference in zll.db
Dec 09 16:24:26 pi2 deCONZ[13372]: 16:24:26:896 Beacon src: 0x51A2 ch: 25 updateId: 10
Dec 09 16:24:26 pi2 deCONZ[13372]: 16:24:26:896 * node has higher updateId should be 0, TODO handle
Dec 09 16:24:26 pi2 deCONZ[13372]: 16:24:26:896 Beacon src: 0x7E36 ch: 25 updateId: 10
Dec 09 16:24:26 pi2 deCONZ[13372]: 16:24:26:896 * node has higher updateId should be 0, TODO handle
Dec 09 16:24:26 pi2 deCONZ[13372]: 16:24:26:896 Beacon src: 0x4855 ch: 25 updateId: 10
```
The _Network Settings_ in the GUI shows 10.

It seems like deCONZ is running normally, but:
```
$ ph get /groups/0
ph get: error: 3 resource, /groups/0, not available
```

Attached the log:
[log.gz](https://github.com/dresden-elektronik/deconz-rest-plugin/files/2660667/log.gz)

ebaauw, 2018-12-09:
`top` shows deCONZ at 103% CPU during the freeze.

I re-installed v2.05.49 and reset and re-flashed the ConBee to 0x262f0500: no change.

The database showed a device for group 0x0000, the id and the groupname referred to my ubisys D1 dimmer.  I changed it back to what it looks like in my test network, and now `/groups/0` now works.

ebaauw, 2018-12-09:
> Interesting did the D1 put itself into that group?

Dunno.

I do see a message complaining about an invalid group:
```
Dec  9 19:03:52 pi2 deCONZ[31731]: 19:03:51:279 Get group for id error: invalid group id 230,240
```
which is for the D1
```json
{
  "config": {
    "group": "230,240",
    "mode": "momentary",
    "on": true,
    "reachable": true
  },
  "ep": 2,
  "etag": "c75c0bc8d4672d41f24d10b55d4220ec",
  "manufacturername": "ubisys",
  "mode": 1,
  "modelid": "D1 (5503)",
  "name": "Dining Room Table",
  "state": {
    "buttonevent": 1002,
    "lastupdated": "2018-12-08T19:02:17"
  },
  "swversion": "20171108-DE-FB0",
  "type": "ZHASwitch",
  "uniqueid": "00:1f:ee:00:00:00:19:63-02-0006"
}
```

ebaauw, 2018-12-09:
>  this will be fixed in the next version

As in 2.05.51?  I see no improvement in Tokyo 2.05.50.

> it shouldn't have any other effect.

The REST API is also frozen.  I think homebridge-hue handles this well, but it does issue a lot of timeout error messages...

manup, 2018-12-09:
>As in 2.05.51? I see no improvement in Tokyo 2.05.50.

Yes, should be fixed during this week.

>The REST API is also frozen. I think homebridge-hue handles this well, but it does issue a lot of timeout error messages...

Single thread is busy :)

ebaauw, 2018-12-13:
17 minutes now.
```
Dec 13 18:37:50 pi2 deCONZ[29402]: 18:37:50:441 Node: id: 62, 00:17:88:01:10:50:1c:df (0x0017880110501CDF) scene: 71.000000, -909.000000
Dec 13 18:54:58 pi2 deCONZ[29402]: 18:37:50:450 Node: id: 63, 00:17:88:01:02:00:a8:5b (0x001788010200A85B) scene: 356.000000, -992.000000

```

ebaauw, 2018-12-14:
> I wonder if there might be another issue in place here.

The time seems to increase with each restart, making me think it might be an issue with the database.  I issued a VACUUM from sqlitebrowser, but no visible improvement (and only a small gain on the file size).  The `device_descriptor` table now has over 13,000 rows, over 12,500 of which are for the constantly rebooting `lumi.ctrl_ln2.aq1`.  Has that table been indexed properly?  Maybe deCONZ should cleanup old entries - the rate at which the switch reboots is ridiculous, but over time, you might see the same for lights behind traditional wall switches.

Do you know how I can kick the switch from the network, without having physical access to the buttons?

Can I delete the rows from the `device_descriptor` table without damaging my configuration?

manup, 2018-12-14:
>The device_descriptor table now has over 13,000 rows, over 12,500 of which are for the constantly rebooting lumi.ctrl_ln2.aq1. Has that table been indexed properly?

That should definitely not happen, the table should only hold one row per descriptor (node descriptor, simple descriptor, etc).

>Maybe deCONZ should cleanup old entries - the rate at which the switch reboots is ridiculous, but over time, you might see the same for lights behind traditional wall switches.

There should be no duplicates at all since the descriptors are static, I'll add some code to clean it up and prevent duplicates from being created.

>Do you know how I can kick the switch from the network, without having physical access to the buttons?

Delete the light in Phoscon App this should send a ZDP mgmt leave request with rejoin = false. However due the recently discovered ZDP issue it might not work if the switch isn't power-cycled before.

>Can I delete the rows from the device_descriptor table without damaging my configuration?

Should work, you can also do this selectively for just the ones with the specific device id. However recommend to create a backup of the file first :)

ebaauw, 2018-12-14:
```
select device_id, endpoint, count(*) from device_descriptors group by device_id, endpoint having count(*) > 1
```
Yields:
[image]

manup, 2018-12-14:
`select * from device_descriptors where device_id = (SELECT id FROM devices WHERE mac = '00:15:8d:00:02:a5:34:9d');`

Can you please provide a screenshot from this command with the mac address of your device.

The code already seems to check for double entries I wonder how they are created in your setup.

For my lumi.ctrl_ln2.aq1 it looks fine, the device has datecode `6-25-2018`.

[image]

manup, 2018-12-14:
[image]

5,6,7 are the switch endpoints ..

ebaauw, 2018-12-14:
[image]

manup, 2018-12-14:
I think the bug is here

[image]

If `rc != SQLITE_OK`, the code still proceeds and adds a new entry.
`rc` might be SQLITE_BUSY or the database might also be locked while sqlitebrowser has unwritten changes.

ebaauw, 2018-12-14:
> Can you please provide a screenshot from this command with the mac address of your device.

That's over 12,500 rows?
[image]

ebaauw, 2018-12-14:
Delete all the duplicates.  Restart delay is down to 2 seconds:
```
Dec 14 11:47:15 pi2 deCONZ[23157]: 11:47:15:517 Node: id: 61, 00:17:88:01:02:00:ac:7c (0x001788010200AC7C) scene: -1434.942537, -394.339920
Dec 14 11:47:17 pi2 deCONZ[23157]: 11:47:15:527 Node: id: 62, 00:17:88:01:10:50:1c:df (0x0017880110501CDF) scene: 71.000000, -909.000000
```

manup, 2018-12-14:
Last two commits should fix the problem and prevent it from happening in the future.
If possible please try with the larger zll.db file, it should start more fast and to the cleanup automatically.

ebaauw, 2018-12-14:
No noticeable delay during startup, even with the full database.  The database is looking good - no more duplicates in the `device_descriptors` table.  

After the cleanup, I shutdown deCONZ and did a manual VACUUM.  This reduced the database file from 848,896 bytes to 306,176 bytes.

closed 2018-12-14 by ebaauw