dresden-elektronik/deconz-rest-plugin #1020: v2.05.50 Xiaomi and IKEA issues

ademobile, 2018-12-10:


ademobile, 2018-12-10:
Sorry - cat walked on keyboard just as I was starting issue!

After upgrading to 2.05.50 I lost all my IKEA 5 button remotes and their default functionality once re-added (could only restore on/off and 'wired' brightness control through switch editor, and no colour temperature control). All apart from one Xiaomi switches would no longer work and would not re-add. I have had to roll back to 2.05.49 to get everything working again. I will update more as I work out what is not working as it should

ademobile, 2018-12-10:
I have:

1 x lumi.remote.b186acno1 and this is discovered when being added by deCONZ (you can see the device MAC of 00158d00028cae19), however the web interface never shows it added. This one has never worked and was only purchased about 4 weeks ago.

1 x lumi.ctrl_ln1.aq1, this is discoved as a light pre 2.05.50, however stopped working after upgrade and when re-added, has the same problem as above. Its MAX is 00158d00024c1279.

1 x lumi.ctrl_ln2.aq1, this again was OKish pre 2.05.50, one switch was reported the other was a 'unknown'. Once upgraded to 2.05.50 both switches were shown, but I was not getting any events. I re-added it and it now appeared as a switch rather than two lamps. I then have no way of adding the switch to a light as these were not created!

manup, 2018-12-10:
>After upgrading to 2.05.50 I lost all my IKEA 5 button remotes and their default functionality once re-added (could only restore on/off and 'weird' brightness control through switch editor, and no colour temperature control). 

What do you mean by lost aren't they shown?

The default rules to control color temperature are no longer created on setup, since the Switch Editor is the preferred way to configure the buttons. Currently direct color temperature control is not available, but will be again soon. Please use scenes as workaround.

**Important note** to use the Switch Editor (which is used with normal groups) lights *must not* be assigned to the switch directly since this will fight with the rules the Switch Editor creates.

We're working on the UI to clarify this.

manup, 2018-12-10:
>1 x lumi.ctrl_ln1.aq1, this is discoved as a light pre 2.05.50, however stopped working after upgrade and when re-added, has the same problem as above. Its MAX is 00158d00024c1279.

> 1 x lumi.ctrl_ln2.aq1, this again was OKish pre 2.05.50, one switch was reported the other was a 'unknown'. Once upgraded to 2.05.50 both switches were shown, but I was not getting any events. I [...]

2.05.50 was testet with these two, albeit mainly switch functionality was tested, the lights should be available as on/off plugin unit - I'll check it out later.

ademobile, 2018-12-10:
> What do you mean by lost aren't they shown?
>
> The default rules to control color temperature are no longer created on setup, since the Switch Editor is the preferred way to configure the buttons. [...]

[image]

These 4 IKEA remotes were no longer shown on the UI. They were listed under 'Switches' however two of the no longer 'worked' (i.e. did not control the lights or give any events under the web socket). Once re- added they could be seen, but the only way to control lights was now through the switch editor. Changes made here did not happen until I restarted deCONZ. This was getting to much as my partner was getting angry with me that nothing worked!!

manup, 2018-12-10:
Ah ok thanks for the description, I think I know what's happening here since the switches aren't shown, will be fixed in the next version this week. Sorry for the inconvenience.

ademobile, 2018-12-10:
> Ah ok thanks for the description, I think I know what's happening here since the switches aren't shown, will be fixed in the next version this week. Sorry for the inconvenience.

No problem, I understand things happen, it's the 'Partner Acceptance Factor' that is always the issue when things don't work as they expect. I.e. a switch turns something on or off.

Any chance of knowing when the lumi.remote.b186acno1 is going to work? As this is really needed and I getting most of the grief over that one switch!

manup, 2018-12-10:
The online version on dresden-elektronik.de is fixed now so that IKEA and Philips switches are shown again on the main page, will be also part of deCONZ 2.05.51.

[http://www.dresden-elektronik.de/pwa](http://www.dresden-elektronik.de/pwa?12345)

>No problem, I understand things happen, it's the 'Partner Acceptance Factor' that is always the issue when things don't work as they expect. I.e. a switch turns something on or off.

Totally understand, this challenge comes up quite frequently :)

>Any chance of knowing when the lumi.remote.b186acno1 is going to work? As this is really needed and I getting most of the grief over that one switch!

It should already work, at least the code is in 2.05.50, but I couldn't test it since I don't have the switch yet.

https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Xiaomi-WXKG03LM-2018

ademobile, 2018-12-10:
> The online version on dresden-elektronik.de is fixed now so that IKEA and Philips switches are shown again on the main page, will be also part of deCONZ 2.05.51.

I'll keep an eye out for this update and once release, I'll try agin and report back, however the that switch is not showing up in the web interface and no events from the web socket

rafuz, 2018-12-11:
I am experiencing the same problem with 2.05.50 using [Vimar Zigbee Green](https://www.vimar.com/it/it/catalog/product/index/code/03905) wall switches. They disappear from the web interface and I don't see any event from them in the websocket. Reverting to 2.05.49 takes them back working correctly.

joch, 2018-12-12:
I’m having the same issue with my Ikea dimmers. Other switches are visible in the UI as expected, but not the round dimmers. (Same with the hosted version of Phoscon as well, but that did at least show the two switches on the front page under the groups)

[image]
[image]

rafuz, 2018-12-12:
Yes, I see them as Philips Tap and it would be great to get additional functions for them in Deconz.

[image]

Anyway the buttons do not match so I simply capture the button codes from Deconz rest interface and command lights from HomeAssistant/Node-Red.

They are also different from the last buttons from Vimar that get also the Friends of Hue logo and can be used with Hue Bridge.

manup, 2018-12-16:
Dimmers and Taps are shown again in 2.05.51, however please consider using the switch editor on normal light groups as more powerful alternative to binding lights directly to a switch.

ademobile, 2018-12-17:
I am closing this issue as it was specific to v2.05.50 and now v2.05.51 is out and I have raised new issues for that version

closed 2018-12-17 by ademobile