dresden-elektronik/deconz-rest-plugin #1021: Add Heiman HS3SA Smoke Sensor

peer69, 2018-12-10:
There seems to be a new version of Heiman Smoke Sensors (HS3SA). Has anyone tried to pair it to deCONZ yet? I've ordered one but still needs some time to arrive.

closed 2019-05-17 by stale[bot]

saxn-paule, 2021-09-13:
I also got two of the HS3SA-E smoke detectors. Pairing via Phoscon was possible, but it's always shown as not connected:
[image]

Via Home-Assistant three entities are detected:
[image]

But they don't provide a status. In case of testing the alarm function nothing happens in Phoscon/HA.

Mimiix, 2021-09-13:
@peer69 Does this device work correctly for you?

peer69, 2021-09-13:
@Mimiix : Phoscon looks the same but the GUI shows connected and API also returns values that seem to be correct. So, yes. As much as its supposed to, it is working.