dresden-elektronik/deconz-rest-plugin #1023: Busch Jaeger wall switches randomly lose connection

grover, 2018-12-11:
I've a few wall switches, which get random disconnects - ie. state.unreachable is true and the node in deCONZ is red. Sometimes these recover, sometimes I have to pull the key front out of the wall socket to restore connectivity. In one case the in-wall unit was a dimmer and the other case was a relais.

In both cases the group commands still work, but the group state change isn't reflected in the in state.any_on or state.all_on. Curiously the light resource created for the switches does seem to reflect the state of the light itself, but still reporting it as unreachable. (Not sure if the disconnect happened after the state update.)

deCONZ 2.05.49
FW 0x262F0500

grover, 2018-12-11:
Just happened to a wall switch connected to a power supply unit. Anything additional I can provide to diagnose this?

manup, 2018-12-11:
Haven't played with these for a while, I'l put a few BJ wall switches in my test network, see if I can recreate the issues.

grover, 2018-12-11:
Curiously the last switch to report this does still cause rules to fire (e.g. buttons are still reported.)

The sensor resource stays unreachable.

manup, 2018-12-11:
Is it a 1x, 2x or 4x switch? Might be a problem in updates to all resources.

grover, 2018-12-11:
The switch on the dimmer and relais units are 1x, the one on the power supply is a 2x. Two nodes (the dimmer and power supply one) are red in deCONZ. Pressing 0 in deCONZ on the dimmer/relais has restored their color to yellow.

The REST api still reports them as unreachable.

Finally, the blinking dots on those nodes are stuck at yellow for a while and then blink red a few times.

grover, 2018-12-12:
Today a Philips Hue E14 Color & White also exhibited this behavior.

grover, 2018-12-13:
Ok this is unfortunately getting annoying - can I downgrade to 2.04.90? Is the database format compatible or will I lose something?

Or would it be sufficient to downgrade the firmware on the raspbee?

manup, 2018-12-14:
Downgrade should work, however please make a backup first with current running versions to be on the safe side.

I'm heaving more time the next days to track down some issues, 2.05.51 arriving this week will bring a few fixes might help here as well.

>Or would it be sufficient to downgrade the firmware on the raspbee?

Hard to tell, depends on what is causing the issues.

manup, 2018-12-14:
Also please make sure the BJ ota files are not in the ~/otau directory, otherwise if the switch doesn't accept the update it will go in an infinity loop to update. (just had this problem)

closed 2019-04-20 by stale[bot]