dresden-elektronik/deconz-rest-plugin #1024: duplicate CLIP sensors

imammedo, 2018-12-11:
Can't pinpoint when issue started but it looks like switch editor creates CLIP-Sensor-s which are duplicates modulo sensor id and that confuses homeassistant (and me :)) which sees 3 sensors with the same uniqueid and sometimes name. So question is if it's the same sensor or different ones, they all seem to be used in different rules (I have mapped 2 remotes using switch editor and 1 of them also mapped to color loop action). Maybe HA is wrong by looking at Phoscon sensors, but non unique "uniqueid", "name" and "etag" don't look correct either.

Currently I have following duplicates:
    "45": {
        "config": {
            "on": true,
            "reachable": true
        },
        "etag": "ff295bb7e5c36de0fb210881b9c734b3",
        "manufacturername": "Phoscon",
        "modelid": "PHOSCON_FSM_FLAG",
        "name": "CLIP-Sensor TOOGLE-COLORLOOP",
        "state": {
            "flag": false,
            "lastupdated": "2018-12-11T19:22:14"
        },
        "swversion": "1.0",
        "type": "CLIPGenericFlag",
        "uniqueid": "fsm-x0bb"
    }

    "11": {
        "config": {
            "on": true,
            "reachable": true
        },
        "etag": "ff295bb7e5c36de0fb210881b9c734b3",
        "manufacturername": "Phoscon",
        "modelid": "PHOSCON_FSM_FLAG",
        "name": "CLIP-Sensor TOOGLE-ON-OFF",
        "state": {
            "flag": false,
            "lastupdated": "2018-12-11T19:22:14"
        },
        "swversion": "1.0",
        "type": "CLIPGenericFlag",
        "uniqueid": "fsm-x0bb"
    },
    "12": {
        "config": {
            "on": true,
            "reachable": true
        },
        "etag": "ff295bb7e5c36de0fb210881b9c734b3",
        "manufacturername": "Phoscon",
        "modelid": "PHOSCON_FSM_FLAG",
        "name": "CLIP-Sensor TOOGLE-ON-OFF",
        "state": {
            "flag": false,
            "lastupdated": "2018-12-11T19:22:14"
        },
        "swversion": "1.0",
        "type": "CLIPGenericFlag",
        "uniqueid": "fsm-x0bb"
    },

manup, 2018-12-11:
Good catch, I'll forward it to the dev, guess a more meaningful uniqueid could be based on the related physical switch uniqueid.

manup, 2018-12-16:
Fixed in 2.05.51

https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/V2_05_51