dresden-elektronik/deconz-rest-plugin #1025: Xiaomi vibration sensor. Need some ideas

MattL0, 2018-12-11:
Hi,

Just received my sensors. But realise that the vibration is not sensible at all. So I will scratch that function. 

Do you guys have some ideas about what  you monitor with this sensor? (Real life exemple)

Thanks

MattL0, 2018-12-12:
The way i see this device: Same as the cube with the disaventage of aving less functions, and the advantage of beeinh smaller

rich710se, 2018-12-12:
I will try them on my mailbox to get a notice when the lid is opening

wvuyk, 2018-12-12:
Have tested them on diswasher, washing machine etc, no signals. Guess the machines nowadays are quite stable. I have put one on the juwelbox of the wife to scare her, by announcing this over the speakers... :-)

wvuyk, 2018-12-12:
It is not yet available in rest api, but I know there is being worked on :-)

ebaauw, 2018-12-12:
And there's still people complaining about the Wife Acceptance Factor...

Smanar, 2018-12-12:
lol

manup, 2018-12-12:
I have two of these, one on the table one on the couch. Both are configured with high sensitivity and they trigger the same motion sensor control as my Philips Hue motion sensor which is nice since the motion sensor doesn't catch small movements.

peer69, 2018-12-21:
Not sure how to use this one in my mailbox, which was the reason I got one. Vibration event gives me false positives, tilt event gives me false negatives. Might have to play around some more with the sensitivity setting but any time trucks pass along my house the sensor is triggered, setting the sensitivity higher again results in false negatives. Might have to use a PIR or door sensor in my case.

JuriLeon, 2019-02-11:
Turn off an (offline) 3D Printer after finishing the print

Mouky66, 2019-04-12:
any news on the sensitivity?

wvuyk, 2019-04-12:
Sensititvity is available in the REST API, works perfectly now.

closed 2019-08-17 by stale[bot]

xvolte, 2020-08-08:
> Sensititvity is available in the REST API, works perfectly now.

Can you explain how to set it up ? i don't understand how to set sensitivity, nor the values i should put ...
Thanks in advance,

Smanar, 2020-08-08:
`curl -H 'Content-Type: application/json' -X PUT -d '{"sensitivity": 1}' http://192.168.1.1:80/api/az45z4a8ze/sensors/1/config`
1 more sensible 21 less.

nwhistler, 2020-08-18:
I've just changed the sensitivity to 1. And I rebooted deconz and now it's not in a pending state. But for vibration, it is still showing as false even though the level is at 14... How do I get it to show as true?

`{
  "config": {
    "battery": 100,
    "on": true,
    "pending": [],
    "reachable": true,
    "sensitivity": 1,
    "sensitivitymax": 21,
    "temperature": 2900
  },
  "ep": 1,
  "etag": "57c95ccd71f8ca76a0ee69b7b1fdc12f",
  "lastseen": "2020-08-18T05:35:39.661",
  "manufacturername": "LUMI",
  "modelid": "lumi.vibration.aq1",
  "name": "AC Sensor",
  "state": {
    "lastupdated": "2020-08-18T02:49:05.035",
    "orientation": [
      -2,
      73,
      17
    ],
    "tiltangle": 360,
    "vibration": false,
    "vibrationstrength": 14
  },
  "type": "ZHAVibration",
  "uniqueid": "00:15:8d:00:02:b6:00:61-01-0101"
}`

Smanar, 2020-08-18:
"sensitivity": 1 is the more sensible, if whith that the sensor don't detect vibration I think you can't do better ....

ebaauw, 2020-08-18:
> I've just changed the sensitivity to 1. And I rebooted deconz and now it's not in a pending state.

Which probably means the sensitivity setting hasn't been written to the sensor.  `config.pending` should be cleared by the API when it writes the sensitivity setting.  It will wait for the device to wake before doing that.  You might try and shake it to force it to wake, otherwise, it might take up to an hour before the next periodic report.