dresden-elektronik/deconz-rest-plugin #1026: Xiaomi sensor going becoming unavailable for a very specific amount of time

GatoPharaoh, 2018-12-12:
for a few weeks now I am having trouble with some of my xiaomi door/window sensors. they seem to get unavailable pretty regularly. two of the sensors just become unavailable for about 50 minutes, then return for about 20 minutes to then become unavailable again for about 50 minutes.

this happens throughout the whole day. I reset them once which made the problem go away. but after some days and a reboot of my hub, one of the sensors started doing it again. a third sensor is also acting weirdly for a few weeks now, but there seems to be no pattern when it does so.

[image]

[image]

Any ideas what's the problem here?

thanks in advance

rafuz, 2018-12-12:
I am using deconz in a docker container (from marthoc) and I am experiencing exactly this issue. When I reboot deconz some Xiaomi Mi Door/Window sensors starts to drop out on a timely basis:

[image]

The sensors having this problem are always the same and if I reset them (10 seconds press + rejoin) the issue disappear and the sensor is stable again.
Yesterday evening a did a reboot of deconz and I collected the logs until this morning. I did a reset only of a couple of the affected sensosr and I have left the others in their unstable behaviour.

[image]

I have a full deconz log from the reboot yesterday to this morning.

[deconz.zip](https://github.com/dresden-elektronik/deconz-rest-plugin/files/2671558/deconz.zip)

I am using deconz for some smart plugs, tap switches and water sensors.

rafuz, 2018-12-13:
Any idea about that?
I did a delete and rejoin of some of the problematic nodes. I will update if I find something different when I reboot Deconz

manup, 2018-12-14:
Not sure how is dropout meant? Once the sensors become available there is in fact currently no code to mark them non reachable again.

Note for all Xiaomi battery devices it's important to know that if nothing happens (like window is closed all the time) they just send a heartbeat once per hour.

manup, 2018-12-14:
>This thing happens only when I restart Deconz.

Since 2.05.50 the reachable state of sensors should be restored right after startup. Versions before did set reachable to false and waited until a sensor sends data.

>EDIT - Just an observation: the sensors goes back online exactly every hour when he sends the heartbeat and the goes back "not reachable" after 13-15 minutes. Everything can be seen in the log I attached in my first post.

Interesting, I'll check the logs I thought this isn't possible with current version looking forward to fix it in next version. I'm only having the Aqara version of the door/window contact which doesn't show this behavior.

ebaauw, 2018-12-14:
>  I thought this isn't possible with current version

@manup, maybe deCONZ is trying to poll the sensor for a non-existing cluster, like you saw for the ZHAConsumption?

rafuz, 2018-12-14:
> Since 2.05.50 the reachable state of sensors should be restored right after startup. Versions before did set reachable to false and waited until a sensor sends data.
>
>
> Interesting, I'll check the [...]

It should be ok to just set the sensor reachable at boot and to not query it for something that could bring it to "not reachable". In fact the sensor alway report the open/close state when the window moves so "it works". Only the states in Deconz seems messed up.

rafuz, 2018-12-15:
I have done a reboot of deconz right now, many sensors are not reachable, also the ones that I have deleted and re-added in phoscon. I will check later the situation and report back with logs.

rafuz, 2018-12-16:
After the reboot all re-created sensors came back online.
This evening I have deleted and re-paired all the remaining problematic sensors and I will try a reboot tomorrow in the evening to check if they begin to flap again. At this point I think that the original pairing of these sensors had some unknown issue and deleting from the REST interface and rejoining them fixed the issue.

GatoPharaoh, 2018-12-30:
today shortly after midnight all the sensors in question stopped having trouble. I did not do any manual updates so I don't know why they suddenly started working again but as of now they do report as expected.

will continue to monitor the situation

manup, 2019-01-06:
With version 2.05.54 the Xiaomi Mi branded sensors and switches should behave better and stay alive.

https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/V2_05_54

GatoPharaoh, 2019-01-08:
@manup after 24 hours I can say the following
door/window sensors started to act up after the update (I updated from 2.05.52) - I deleted and repaired them and the seem to work for now - but I'll continue to monitor it

_but one problem came with the update_
my xiaomi motion sensors now get stuck for hours reporting "movement.  they report movement when passing but sometimes the just do not go back to "no movement".
i then have to pass them again which makes them return to no movement after a minute. but they get stuck pretty regularly. 

one sensor also changed to movement by itself and got stuck there. don't know what's  going on with them.
[image]

henel321, 2019-01-22:
I had the exact same issue yesterday after doing a restore of my config on 2.05.52 and then updating to 2.05.57 (Intrestingly enough the sensors where completely invisible after the restore on 2.05.52 so I thought my setup was completely dead, after updating to 2.05.57 all my config looked good again but 2 out of 4 Xiaomi window sensors going unavailable for periods of time. Not the same periods though.)

I reset/repaired them this afternoon and since then they are reporting properly again. So feels like this is a "fringe" case when sensors get into a weird state -  but it still is around post 2.05.54...  

I'm on 2.05.57, dpkg-installed on Debian so no Docker image. 

@rafuz what version were you on when you had problems and are you having any now?

GatoPharaoh, 2019-01-30:
I'm now on 2.05.55 and reset the sensors which acted weird. After more than a week and multiple restarts they haven't gone unavailable yet. Only thing I still want to try is if this is still the case after updating to v .57.

GatoPharaoh, 2019-02-24:
issue hasn't reoccured

closed 2019-02-24 by GatoPharaoh