dresden-elektronik/deconz-rest-plugin #1029: No Issue - How to start with Restapi

Nobbbi, 2018-12-12:
Hi,

i use the Xiaomi WXKG01LM Round Switch.
Until now i used iobroker and a script to switch my ikea bulbs within deconz.

I would like to use the restapi to switch these lamps, i have problems to get startet.
I know the documentation from dresden elektronik.
What i need for a better understanding, are some examples.

What i like to do: 
one klick - lamp1 on/off
two klicks - lamp2 on/off 
three klick - all off 

Maybe some one can give me some hints, thank you !

Smanar, 2018-12-12:
I haven't see something like that in API, I think you need to make all with your script, Counting the klick on a special period and send request according to it.

manup, 2018-12-12:
>I thought that i can use the rest api to define rules, for all the things i can not realize with the phoscon web app.

That's correct, and intended. Currently our docs are very outdated, a more comprehensive version is in development. Anyway I suggest you study the Hue API rule and sensor example as they mostly work the same in the deCONZ REST-API.

The Hue Dimmer Switch and Hue Tap examples are a great entry point.

Smanar, 2018-12-12:
Have you a public link for that pls ?
I m trying to search but the only link I found need a Hue developer account on philips site.

manup, 2018-12-12:
I'm afraid you need to register to see the content, a bit cumbersome but it's free and worth it to get some docs.

Smanar, 2018-12-12:
NP, I m going to see, thx.

Nobbbi, 2018-12-13:
Thanks, i'll give it a try !

manup, 2019-01-16:
Closing the issue for now.

closed 2019-01-16 by manup