dresden-elektronik/deconz-rest-plugin #1032: 2.05.49/50 Ikea plug/adapter, unable to pair

olemr, 2018-12-13:
The Ikea Tradfri adapter (aka wall plug) have been out of stock for some time.
Yesterday it was back in stock and I bought some.

Now, I'm not able to pair it to deconz (running headless on Ubuntu, no GUI)
I have tried: (with Ikea adapter 5-10cm from ConBee)
1 - open network, power on adapter - nothing detected by PWA or old WEB app
2 - reset with pin in hole for 5s, with and without subsequent power cycle - nothing detected
3 - reset with HUE1 I/O for 10s, with and without subsequent power cycle - nothing detected
4 - reset with old WEB app Touchlink, with and without subsequent power cycle - nothing detected
Touchlink Blink dims the light on the Ikea adapter but does not work the relay.

I was originally on 2.05.50, but noticed some issues to be fixed in .51, so I downgraded to .49
No luck.

Could these re-stocked Ikea adapters be different from the previous ones?
Could deconz support for these 'lights' have been broken since september?

I'm at a loss. Any suggestions?

olemr, 2018-12-13:
Well, my face is red.
Turned out I was not pressing the pin into the reset hole firmly enough.
Lesson learned:
Adapters/plugs straight out of the box **has** to be reset locally using a pin.
Reset using a HUE1 remote or Touchlink is **not** working.

manup, 2018-12-14:
>Reset using a HUE1 remote or Touchlink is not working.

Interesting, good to now, at least for the lights it works I've had expected it's the same for the plug.

ebaauw, 2018-12-14:
Is the plug a ZLL or ZHA device?  Is it on the latest IKEA firmware?  Can you reset it using the IKEA remote?

olemr, 2018-12-14:
Don't know about ZLL or ZHA. Sorry.
REST data is:

```
  "36": {
    "etag": "9c51cf263f9f1db411c290c8512d084e",
    "hascolor": false,
    "manufacturername": "IKEA of Sweden",
    "modelid": "TRADFRI control outlet",
    "name": "Light 36",
    "state": {
      "alert": "none",
      "bri": 254,
      "on": true,
      "reachable": false
    },
    "swversion": "1.4.020",
    "type": "On/Off plug-in unit",
    "uniqueid": "d0:cf:5e:ff:fe:7d:3b:a0-01"
  },

```
Haven't tried using a Ikea remote.
They are all (10 of them) paired with ConBee now :-)

manup, 2018-12-22:
Does the outlet support dimming? If not we should remove the `state/bri` attribute.

Filialen, 2018-12-22:
No, it's on/off only.

cweitkamp, 2019-01-26:
> Does the outlet support dimming? If not we should remove the `state/bri` attribute.

@manup Did you already find the time to remove the `state/bri` property for the TRADFRI control outlet?

wvuyk, 2019-01-26:
That sure would make it easier to recognise a on/off plug...

manup, 2019-01-27:
>This should be done for any On/Off plug-in unit and any On/Off light, notably for the innr SP 120 and the Busch-Jaeger RM01, currently whitelisted in homebridge-hue to ignore state.bri. Thanks.

For now we should apply the level cluster filter to skip `state.bri` on a per device base, I don't trust the device ids anymore.

>@manup Did you already find the time to remove the state/bri property for the TRADFRI control outlet?

This will be done for 2.05.58. However maybe brightness control is possible and `state.bri` comes back later.

I've received a Ikea control outlet for testing. Did some tests yesterday.

* After connecting the firmware was updated from version 1.4.020 to 2.0.019
* After the update was finished. the outlet didn't work anymore
* Power-cycle didn't help but after a reset via reset-hole the outlet worked again with the new version
* Endpoints changed, version 1.4.020:
 [image]

* Version 2.0.019:
[image]

I was hoping for power consumption or brightness control after the update, but both aren't supported.

However, beside the brightness control doing nothing for real, the cluster is full functional and updates it values. A *Move to Level with on/off* does also turn on/off the outlet.

Maybe this feature can be enabled somehow. The control outlet has a manufacturer specific cluster 0xFC7C. I've issued a discovery attributes command to it and the cluster returned two attributes.

[image]

The meaning is still unknown, I haven't tried to write these, since I was afraid to break my only test unit :)

I've also found some new attributes in the basic cluster:

[image]

* 0x000A is the IKEA product type often printed on the device. Useful to detect if a device is EU/US version. This attributes is supported by all IKEA devices!
* 0x0008 and 0x0009 are only present on the control outlet, the meaning of them is unknown

My next idea was to connect the outlet to the IKEA gateway and do some sniffing, but unfortunately the gateway doesn't connect to the IKEA app anymore. I'm not sure how to fix this.

ebaauw, 2019-01-27:
I suspect the IKEA and innr outlets (and others) support the _Level control_ cluster for the sole purpose of turning on or off on the _Move to Level (with On/Off)_ command.  Unlike Philips Hue, the innr and IKEA switches, dimmers, and even motion sensors communicate directly with the lights, and are hardwired to transmit this command instead of _On_ or _Off_.

lbschenkel, 2019-01-27:
>     * After connecting, the firmware was updated from version 1.4.020 to 2.0.019

@manup, how did you get firmware version 2.0.019? `ikea-ota-download.py` didn't download anything for `0x1101`, and I manually checked http://fw.ota.homesmart.ikea.net/feed/version_info.json and there is nothing there either.

manup, 2019-01-27:
f = urllib.urlopen("http://fw.ota.homesmart.ikea.net/feed/version_info.json")
    #f = urllib.urlopen("http://fw.test.ota.homesmart.ikea.net/feed/version_info.json")

You can edit the ikea-ota-download.py file and use the `test` feed url.

Be aware as the name suggests this firmware might cause problems or even brick the device.

lbschenkel, 2019-01-27:
Right. I don't have the IKEA gateway, in which circumstances does it update to test firmware? Is there a setting to be in a "beta channel" or something like that?

manup, 2019-01-27:
I don't know, just saw that url somewhere on the internet and it happily works the same as the other url.

cweitkamp, 2019-01-28:
> This will be done for 2.05.58. However maybe brightness control is possible and state.bri comes back later.

Sounds good. Thank you.

manup, 2019-02-02:
With 2.05.58, the IKEA control outlet and on/off switch are supported, closing the issue for now.

closed 2019-02-02 by manup