dresden-elektronik/deconz-rest-plugin #1033: IKEA Tradfri Dimmer connected but temperature buttons don't work

iamkarlson, 2018-12-13:
Hey, I was able to connect ikea dimmer with buttons. However, not all the buttons works. Brightness and on/off buttons work, temperature (< and >) buttons don't. Are there any bits of advice about making it works? 

Firmware was updated yesterday to the latest available through Phoscon app.

strips, 2019-01-05:
I have the same issue. Trying to assign < > buttons to color cycle and scenes with color temp does not work. Cycle does nothing. Scenes can only store brightness. It always reverts to the same neutral white temp.

Product ConBee
Version 2.05.52 / 16/12/2018
Firmware 26300500

TRÅDFRI  LED bulb E27 980 lumen
Wireless dimmable/white spectrum opal white
FW 1.2.217 (updated today 2019-01-05)

Remote 
FW 1.2.214 (updated today 2019-01-05)

manup, 2019-01-05:
Please have a look in the latest version. The Ikea 5 button remote control can be configured since a while in the Phoscon App Switch Editor. The latest deCONZ version 2.05.54 adds color temperature control.

https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/V2_05_54

**Important:** When using the Phoscon App Switch Editor
* Don't assign lights directly to the remote control (the group which is controlled by the switch)
* Select a normal light group *Edit > Switch editor* and assign the remote here 

Basic documentation on the Switch Editor:

https://doc.phoscon.de/app/doc.html#group-edit-switches-en

peterlindeman, 2019-03-05:
Still doesn't work in latest release. If the button through the App Switch Editor I can only set color temp to warmer but never to colder. 
When the button is directly coupled to the light group the color temp buttons do not work at all.

For now reverting back to the Ikea bridge

MickL, 2019-06-18:
Same here for version `2.05.65 / 24.5.2019` on Hass.io with RaspBee. On/Off and dim is working, but not changing color. Is there any news on this @manup ?

Where do I find the switch editor?

manup, 2019-06-18:
How did you configure your remote?

The following should work:
[image]

> I added the lights to the Tradfri remote group. On/Off and dim is working, but not changing color. Is there any news on this @manup?

**Important:** *Don't* assign lights directly to the remote when using the Switch Editor, this would cause undefined behavior, since the remote and the gateway would send control commands to the lights.

>It adds "a little bit" of warm/cold but the Tradfri bulp has only 3 settings (cold, warm, super warm) so i would have to press multiple times to change the actual color

Keep the button pressed... the transition time is chosen quite long. I think we can tune this to be faster in future versions.

When you assign lights directly to the remote, the side buttons won't work since non IKEA lights don't understand these commands, and IKEA lights would need to be configured in a certain way to support this. As a workaround if you wan't fixed values as well as non IKEA lights working use multiple scenes which you can cycle through with the side buttons.

MickL, 2019-06-18:
Ah a holding does the trick but it needs 10 seconds to change the color one step instead of one single click. Anyway couldnt you add this behavior initially when connection so new users dont have to use the switch editor?

Despite of this workaround I really would love to see:
- Single press to change warm/cold
- Toggle 3 modes only (probably 0%, 50%, 100%)

peterlindeman, 2019-08-16:
> How did you configure your remote?
>
> The following should work:
> [image]
>
>
> **Important:** _Don't_ assign lights directly to the remote when using the Switch Editor, this would cause undefined [...]

Configured this now several times but cannot get changing the color of the IKEA color bulb working. Hopefully this can be fixed

aaa89h, 2019-08-21:
I'm sure my deconz is up to date
But oddly i don't have the key assignment feature showing up at all
Is it because I'm using deconz on home assistant?
I checked for updates but i have the latest deconz firmware

MickL, 2019-08-22:
Im on HA/hassio, too. It is just very very hidden. The UI looks good but the user experience is very bad imo.

peterlindeman, 2019-12-20:
Hmm, still doesn’t work in current version….

infernix, 2020-01-10:
Can confirm, this does not work.

G3rry71, 2020-01-14:
Confirmed, not working

Qalculus, 2020-03-28:
Changing color temperature is not working at all for me. I am using version 1.05.75 / 2020-03-08.

Caine72, 2020-03-31:
Same issue. Latest software.

I worked around the issue by creating three scenes. Cold, White, Yellow.  
Then let the buttons call Previous and Next Scene to get the carrousel. 

This works fine .. but with one drawback. Scenes seem to be all or nothing. I.e I have to change both temperature and brightness at the same time. Seem like one cannot do only one of the things in a scene.  ( please correct me if i'm wrong).
So, if i already have a "correct" brightness set , changing scene will re-set it to my pre-set scene value instead of keeping current.

michaelbroman, 2020-07-11:
I have the exakt same problem, all but the color temp is working. My current config is to use the toggle instead but that is not the desirable solution.
RaspBee fw 26580700 and latest deconz.

Thimboo, 2020-07-12:
Same problem here. 
HassOS 4.11 with ConBee II and deCONZ 5.3.6. 
Tradfri FLOALT panel WS 30x30 with Tradfri Remote

hreichert, 2020-08-10:
Same problem here.

Mimiix, 2020-08-11:
Moving this to Phoscon repo!