dresden-elektronik/deconz-rest-plugin #1034: Cannot connect my Xaomi Aquara sensors to Conbee

Gallimathias, 2018-12-13:
Hello, everybody. I have seen in a post that the ConBee is a nice alternative to all the manufacturer gateways. 

I have 3 Xiaomi Aquara Temperature / Humidity sensors (WSDCGQ11LM) but I can make them press and do not connect to my network.

Currently the ConBee is connected with my Windows computer with deConz. On my Raspberry PI I could not connect to the Phoscon app.

manup, 2018-12-14:
Do you use recent deCONZ and firmware versions?

https://github.com/dresden-elektronik/deconz-rest-plugin#precompiled-deconz-packages

If the ConBee is connected directly to the PC it might help to use a usb extension cable, we've seen multiple cases there otherwise the signal interference was too high (note to myself: put this in documentation).

Gallimathias, 2018-12-14:
Yes, I had the latest firmware.

The tip with the cable is great and works.

Unfortunately the sensor shows only the air pressure in the webapp and also unfortunately only one sensor. But I can manage that. In Deconz all 3 sensors are functional.

closed 2018-12-14 by Gallimathias