dresden-elektronik/deconz-rest-plugin #1035: state.xy broken

ebaauw, 2018-12-13:
Not sure when this was introduced, but it seems `state.xy` is broken.  When reading the _Color Control_ cluster attrobites, deCONZ issues multiple web socket notifications for `state.x` and for `state.y`, all with value 65279 (0xfeff - the maximum value for _Current X_ and _Current Y_).  After that the API returns `[1, 1]` for `xy`.

A PUT of `state.xy` sets the correct colour to the light, and updates the API state, but on the next poll, the values revert to `[1, 1]`.

ebaauw, 2018-12-13:
It's caused by the double definitions in general.xml for attributes 0x0003 (_Current X_ / _PowerOn X_) and 0x0004 (_Current Y_ / _PowerOn Y_) in the _Color Control_ cluster, introduced in https://github.com/dresden-elektronik/deconz-rest-plugin/pull/1015 for https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1014.

manup, 2018-12-14:
Thanks, good catch. Still would be good if it gets supported I'll have a look into the ZCLDB code.

ebaauw, 2018-12-17:
Fixed in v2.05.52.

ZCLDB code should be tracked in #1014, so closing this issue.

closed 2018-12-17 by ebaauw