dresden-elektronik/deconz-rest-plugin #1041: Xiaomi Battery Level

phdelodder, 2018-12-15:
I recently switch from Xiaomi gateway to a Conbee, the battery level report from Xiaomi gateway is completely different than from Conbee.  

Xiaomi gateway value was 42%, Conbee value is 92%

ebaauw, 2018-12-15:
The Xiaomi sensors don't report battery percentage, only voltage.  The deCONZ REST API plugin translates this to percentage, using the same approach as the Smartthings handlers for Xiaomi.  The Xiaomi gateway might use a different algorithm.

Note that deriving the battery percentage from the voltage is inherently unreliable.  Typically CRxxxx batteries provide full voltage for a long time, and then drop quickly as the battery is nearing depletion.

Also note that it might take up to an hour (after pairing the device or restarting deCONZ) before the battery percentage is updated.

phdelodder, 2018-12-15:
Thanks for the explanation,

closed 2018-12-15 by phdelodder