dresden-elektronik/deconz-rest-plugin #1042: deCONZ-GUI won't start

patm80, 2018-12-15:
Hello,

I'm trying to get the RaspBee Module to work. I've installed everything according to the manual PDF file but the deconz-gui service fails before initializing and won't start. The LED on the module shows red color, according to the manuel it means that it is connected to the firmware. When I look at the logs from "journalctl -xe" I have the following output:

`Dez 15 16:32:16 raspberrypi systemd[1]: deconz-init.service: Start request repeated too quickly.
Dez 15 16:32:16 raspberrypi systemd[1]: Failed to start deCONZ: ZigBee gateway -- Initialisation.
-- Subject: Unit deconz-init.service has failed
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- Unit deconz-init.service has failed.
--
-- The result is failed.
Dez 15 16:32:16 raspberrypi systemd[1]: deconz-init.service: Failed with result 'start-limit-hit'.
Dez 15 16:32:16 raspberrypi systemd[1]: deconz-gui.service: Service hold-off time over, scheduling re
Dez 15 16:32:16 raspberrypi systemd[1]: Stopped deCONZ: ZigBee gateway -- GUI/REST API.
-- Subject: Unit deconz-gui.service has finished shutting down
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- Unit deconz-gui.service has finished shutting down.
Dez 15 16:32:16 raspberrypi systemd[1]: deconz-gui.service: Start request repeated too quickly.
Dez 15 16:32:16 raspberrypi systemd[1]: Failed to start deCONZ: ZigBee gateway -- GUI/REST API.
-- Subject: Unit deconz-gui.service has failed
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- Unit deconz-gui.service has failed.
--
-- The result is failed.
Dez 15 16:32:16 raspberrypi systemd[1]: deconz-gui.service: Unit entered failed state.
Dez 15 16:32:16 raspberrypi systemd[1]: deconz-gui.service: Failed with result 'exit-code'.
Dez 15 16:32:16 raspberrypi systemd[1]: deconz-init.service: Start request repeated too quickly.
Dez 15 16:32:16 raspberrypi systemd[1]: Failed to start deCONZ: ZigBee gateway -- Initialisation.
-- Subject: Unit deconz-init.service has failed
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- Unit deconz-init.service has failed.
--
-- The result is failed.
Dez 15 16:32:16 raspberrypi systemd[1]: deconz-init.service: Failed with result 'start-limit-hit'.
Dez 15 16:32:18 raspberrypi sudo[1277]:  USER : TTY=pts/0 ; PWD=/home/USER ; USER=root ; COMMAN
Dez 15 16:32:18 raspberrypi sudo[1277]: pam_unix(sudo:session): session opened for user root by USER
Dez 15 16:32:21 raspberrypi sudo[1277]: pam_unix(sudo:session): session closed for user root
Dez 15 16:32:24 raspberrypi sudo[1292]:  USER : TTY=pts/0 ; PWD=/home/USER ; USER=root ; COMMAN
Dez 15 16:32:24 raspberrypi sudo[1292]: pam_unix(sudo:session): session opened for user root by USER
Dez 15 16:32:25 raspberrypi deCONZ-update2.sh[293]: no database file (zll.db) found
Dez 15 16:32:27 raspberrypi sudo[1292]: pam_unix(sudo:session): session closed for user root
Dez 15 16:32:32 raspberrypi sudo[1311]:  USER : TTY=pts/0 ; PWD=/home/USER ; USER=root ; COMMAN
Dez 15 16:32:32 raspberrypi sudo[1311]: pam_unix(sudo:session): session opened for user root by USER`

I'm running a normal raspbian image on my by where the module is supposed to operate. The results I found regarding this issue when searching didn't bring any success unfortunately. Did anyone have the same problem or can point me to what I'm doing wrong?

I really appreciate every suggestion and help!

Thanks in advance,
Patrick

bdickhaus, 2018-12-16:
Patrick,
It is likely a permissions issue. You need to have a user "pi" and as I understand it, it has to be user 1000. The easiest thing is to use their raspbian package, which is what I did. Have you tried starting deconz manually from the command line?
Bill

patm80, 2018-12-16:
Hey Bill,

thanks for your response! Ok, I was not aware that there had to be the Pi user still on the system. I will try that. Yeah I also tried to start it over the commandline. The service wont start and throws the error I posted in the first post.
I'll try installing it with the Pi user.

Thanks,
Patrick

patm80, 2018-12-16:
Hey,

short update regarding my issue. Installing the module with the pi user worked. I flashed raspbian again and followed the instruction in the manual with the standard pi user and now the module works! Thanks so much for your help, Bill! Really appreciate it :) 

Thank you,
Patrick

SwoopX, 2018-12-19:
Check out the wiki for running deconz as another user ;) Went all through it the hard way...

closed 2019-04-26 by stale[bot]