dresden-elektronik/deconz-rest-plugin #1043: ConBee is not recognized by Raspberry

Gallimathias, 2018-12-15:
After the ConBee works via USB cable on my Windows computer (#1034) (the PWA side doesn't recognize any sensors but at least DeConz).

Now I connected the stick (with extension cable) to the Raspberry 3.

[image]

I can't open the webapp anymore and I can't log in to pwa.

I noticed that the stick here looks like a gateway, on the Windows computer i have the logo of a USB stick.

I tried the deconz versions: latest, beta (2.5.50), dev and compiled them myself.

My distribution on Raspberry 3:
Raspbian GNU/Linux 9.6 (stretch)

Gallimathias, 2018-12-17:
So i have tried some things with no results:

__First:__ in dev Folder the device is ttyUSB0.
__Second:__ lsusb results is: Bus 001 Device 006: ID 0403:6015 Future Technology Devices International, Ltd Bridge(I2C/SPI/UART/FIFO)

User pi is in group dialout. And this user executes deconz.

/dev/ttyUSB0 permissions: crw-rw---- 1 root dialout 188, 0 Dez 17 21:34 /dev/ttyUSB0
so the dialout group actually has permissions to use the device

In deconz log i see that deconz try to communicate with /dev/ttyS0 so i use --dev=/dev/ttyUSB0 as parameter but it doesn't work.

In the log i don't see any error or exception. For log i try following: 

deCONZ -platform minimal --dbg-error=2 --dbg-info=1 --dbg-aps=1 --dbg-zcl=1 --dbg-zdp=1 --http-port=8080 --dev=/dev/ttyUSB0

manup, 2018-12-17:
>Second: lsusb results is: Bus 001 Device 006: ID 0403:6015 Future Technology Devices International, Ltd Bridge(I2C/SPI/UART/FIFO)

This looks ok.

Are there other processes running which access the serial interface?

Does the official sd-card image work?

Gallimathias, 2018-12-22:
No deconz uses the serial ports as the only one.

I can't try the official image because I don't have a second sd card currently lying around.

Gallimathias, 2018-12-22:
Is there an alternative to deconz?

Can I check in other ways whether the communication with the stick would generally work?

Gallimathias, 2018-12-31:
I solved the problem with the help of a friend:

* reinstall Deconz on windows computers
* Downgrade of Deconz on the Raspberry to 2.05.48
* Update of the firmware on the ConBee to 26300500
* All sensors were trained on the windows computer (only possible on a Windows laptop without extension cable). Open the network in the classic app and add the sensors in Phoscon.
* edit zll.db, e.g. adjust IP addresses
* Copy zll.db, config.ini and session.default to the Raspberry and overwrite existing ones
* Connecting the ConBee to the Raspberry (without extension cable)
* Now Everything works fine.

closed 2019-05-07 by stale[bot]