dresden-elektronik/deconz-rest-plugin #1044: switch not available (cannot be toggled) after deconz reboot

greencharlie100, 2018-12-16:
Using: PI, Hass.io, RaspBee, deconz...

A switch is successfully paired with deconz and manageable through HA and API commands 

Situation: 
 -> deconz reboot  (caused by system (pi) reboot - I am simulating electrical power failure)

Result:
HA cannot control switch after a reboot until each gang is physically toggled first. Then HA interface can control switch again.
API commands cannot control switch after a reboot until each gang is physically toggled first. Then HA interface can control switch again.

API Error - after an attempt to change the state of the switch:
[
    {
        "error": {
            "address": "/lights/7",
            "description": "resource, /lights/7, not available",
            "type": 3
        }
    }
]

Why can I not control the switch through the API or HA after deconz reboots? It is critical that everything should sync together after al temporary power failure (PI reboot).

Does anyone have a solution to this?

greencharlie100, 2018-12-17:
Did anyone get a solution for this?

greencharlie100, 2018-12-17:
Any help or info on this? 

It seems that deconz is not aware of its connected switches, or at least able to control a switch if deconz is restarted - until the switch sends data to deconz via a switch toggle..

manup, 2018-12-17:
Which deCONZ version are you running?

And what device type and modelid are the switches, battery operated or mains powered switches?

greencharlie100, 2018-12-17:
deCONZ 2.05.5

Wired switches - Xiaomi Aqara QBKG03LM

manup, 2018-12-17:
Guess these are not yet fully supported

https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Supported-Devices

Can you please provide the more info as described here

https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Request-Device-Support

greencharlie100, 2018-12-17:
I assumed that It was supported as it paired etc no problem. The only issue with this switch is what I describe above. 
I don't know if I will be able to fulfil the 'Request Device Support' as I don't believe I can get the required information (node details etc) in hass.io.

manup, 2018-12-17:
>I assumed that It was supported as it paired etc no problem. The only issue with this switch is what I describe above.

I guess only the light might show up and be controllable but the whole switch part with button events isn't supported.

greencharlie100, 2018-12-17:
I can send you the screen shots first thing tomorrow - what are VNC viewer username and password for dresden-elektronik OS image so I can access the PI and take the screenshots.

manup, 2018-12-17:
Cool thanks, VNC credentials:

pi
raspbeegw

greencharlie100, 2018-12-17:
You were quick...
Here is what I think you require. If there is anything else let me know and I will post it tomorrow.
[image]
[image]
[image]
[image]

manup, 2018-12-17:
Interesting so this is a end-device which tells `receiverOnWhenIdle = false`, which I think is wrong. We've had this in the past and need to add some code to workaround this.

greencharlie100, 2018-12-19:
As long as there is a workaround - I am delighted. I look forward to the next update. Thanks.

greencharlie100, 2018-12-19:
Something that I have noticed (not really a big issue) when adding the switches on different occasions:

One time they add as a switch
Another time they add as a light

Same switch model - what determines if it a switch or a light? I tried changing the type in API but was not allowed.

As I said, it is not a big issue but something that would be better if it could be consistent.

manup, 2018-12-19:
The QBKG03LM should (always) be created as one switch and two lights for endpoint 2 and 3. I've checked the code @ebaauw  added the `lumi.ctrl_neutral` as supported prefix and handling on various places.

There is one todo for the `lumi.ctrl_neutral1` (QBKG04LM) but otherwise, it seems, only the receiverOnWhenIdle issue remains.

greencharlie100, 2018-12-19:
OK, thanks.

greencharlie100, 2018-12-29:
Has this issue been addressed in an update?
"Xiaomi Aqara QBKG03LM switch not available after deconz reboot- Request Device Support #1044"

I also notice that the switches switch off by themselves after a random period of time?

greencharlie100, 2018-12-29:
"1046 I also notice that the switches 'switch off' by themselves after a random period of time?"

I do know from testing the switch:
    If the switch (light) is toggled on and I physically turn off electricity in my home for 1 second and turn it 
    back on again
The switch (light) will come of for a split second before turning itself off automatically...

The temporary power drop triggers the switch to turn off as soon as the power comes back ito the circuit?

ebaauw, 2018-12-29:
> what determines if it a switch or a light

The word "switch" is used for two different sorts of devices:
1. To control smart devices.  These are battery-powered and _send_ ZigBee commands to control smart devices (directly or through the coordinator).  These are exposed as `/sensors` resources;
2. To smartify dumb devices (very much like a smart plug).  These are mains powered and _receive_ ZigBee commands to switch on or off (and sometimes even dim) the connected (wired) device.  These are exposed as `/lights` resources.

To complicate matters, some devices (like the Xioami) combine both features in the same device and are exposed as `/lights` as well as `/sensors` resources.

> Xiaomi Aqara QBKG03LM switch not available after deconz reboot

I saw the same for my `lumi.ctrl_neutral2` switches, but after changing the REST API plugin to set `config.reachable` on receiving the Xiaomi special report, this resolves itself within a couple of minutes.  My switches have _Date Code_ `11-11-2016` and _Application Version_ `18` and, unlike the name suggests, do _not_ use the neutral wire.

> I also notice that the switches 'switch off' by themselves after a random period of time?

I've seen this behaviour with two `lumi.ctrl_ln2.aq1` switches, and one OSRAM Plug - Lightify (the old model).  I suspect this is caused by hardware failures.  Eventually, the OSRAM plug would switch off immediately after switching on.  The one `lumi.ctrl_ln2.aq1` that's still connected also reboots when switching off (it sends a device announcement afterwards).  It switches off both channels, irrespective of whether they're connected or whether the connected device is drawing power.  This happens very regularly, so I don't think it's caused by an irregularity in the power grid.

I had the OSRAM plug switching to a dumb fan.  I had/have the `lumi.ctrl_ln2.aq1`s switching a dumb ceiling fan.  I suspect the circuitry in the switches and plug doesn't agree with the inductive load of the electric motors.  Or maybe it's just my bad luck.  My `lumi.ctrl_neutral2`s switch two more ceiling fans, so far without any issues (knock on wood).

> If the switch (light) is toggled on and I physically turn off electricity in my home for 1 second and turn it back on again.  The switch (light) will come of for a split second before turning itself off automatically...

As far as I understand, there's two sorts of relays:
1. Those that require power to remain on and switch off as soon as power is cut;
2. Those that only require power to switch (from off to on, or from on to off) and remain on when power is cut.
I'd theorise Xiaomi uses the second sort, and that it turns the relay off when initialising its firmware on boot.  Fun fact: the innr SP 120 plug remains on when unplugging it and moving it to another power outlet.

ebaauw, 2018-12-29:
> At this point I dont know how easy or difficult it is to implement the fix because I dont have access to the system.

The fix has been included in deCONZ for quite some time, see #798 and #836.

> Maybe the kitchen light 'load' is too much - who knows?

I think the Xiaomi switches are rated for 2500W - that's an awful lot of light.  Are you using regular tungsten bulbs, or a transformer or TL tube or something?

greencharlie100, 2018-12-29:
> The fix has been included in deCONZ for quite some time, see #798 and #836.

But it's not implemented for the QBKG03LM (2018 version) as stated by @manup  a few comments above.

The QBKG03LM max load is 800W /gang. 
I have two fluorescent bulbs in the Kitchen. I dont think they are anywhere near that wattage. 

The circuit in the kitchen was previously two way but was converted to one way to allow the use of the new switch(s).  The two way system (other switch) is overcome with the addition of a wireless switch.
Maybe the electrician has wired something wrong when closing off the curcit to the old switch.

ebaauw, 2018-12-29:
I would suspect the fluorescent bulbs, or rather, the switch not agreeing with them.  They briefly require a lot of power when turning on.

greencharlie100, 2018-12-29:
> I would suspect the fluorescent bulbs, or rather, the switch not agreeing with them. They briefly require a lot of power when turning on.

I don't know, the timing is random but the switch has never gone off for at least 15 minutes after the switch (light) has been turned on and sometimes up to several hours later. I will live with it for now and post any findings at a later date should I find any.

@manup 
Can you add the required support in the next release for 'this' issue #1044 

>@manup
 Interesting so this is a end-device which tells receiverOnWhenIdle = false, which I think is wrong. We've had this in the past and need to add some code to workaround this.

manup, 2018-12-29:
The related commit is https://github.com/dresden-elektronik/deconz-rest-plugin/commit/5b1c6ca1ae4a8ea26ccef38844834ab679fd5679

Two things that we need to check
1. Does the device still send special reports?
  – The device might have gone silent as in https://github.com/dresden-elektronik/deconz-rest-plugin/issues/849#issuecomment-450490183
2. If reports are received, do we hit a `continue` branch before the `reachable` attribute is updated?

greencharlie100, 2018-12-30:
@manup Is this information you require from me?

greencharlie100, 2019-01-12:
@manup any progress with this:

> Xiaomi Aqara QBKG03LM switch not available after deconz reboot- Request Device Support #1044

manup, 2019-01-20:
Not yet, but looking forward to resolve this soon.

manup, 2019-01-24:
I've added some code to repair the `receiver on when idle` flag in node descriptor for `lumi.ctrl_neutral*`. Will be part of deCONZ 2.05.58.

greencharlie100, 2019-01-25:
> 've added some code to repair the receiver on when idle flag in node descriptor for lumi.ctrl_neutral*. Will be part of deCONZ 2.05.58.

Great @manup , thanks.

greencharlie100, 2019-02-01:
2.05.58 update installed and problem solved!

closed 2019-02-01 by greencharlie100

A-BnSultan, 2019-11-30:
wait how did you managed to pair it?
on their website, this model is not supported

and i have tried everything and i couldn't pair it

my deconz is on 2.05.71

please help me

Smanar, 2019-11-30:
Have you tried to long press the button ?

A-BnSultan, 2019-11-30:
i purchase xiaomi kit with motion, temperature, water leak, door/window sensor they all pair fine except the light switch

i long-press the switch then lights turn off and the led of the switch bling blue rapidly 
but never register, i tried as light, switch and sensor and nothing :(

Smanar, 2019-11-30:
>Pairing
Press and hold the button on the device for +- 10 seconds (until the blue light starts blinking and stops blinking), release and wait.

And wired switches are bulbs.

A-BnSultan, 2019-11-30:
wow it registered, i thought i had to release when it start blinking blue
now it shows 5 devices

  | Name                 Vendor    Product
  | On/Off light 3 | Unknown | Unknown
  | On/Off light 5 | Unknown | Unknown
  | Smart plug 1 | Unknown | Unknown
  | Smart plug 2 | Unknown | Unknown
  | On/Off light 4 | Unknown | Unknown

now i have to spend time playing with it and found what i can do 

thank you a lot sir <3