dresden-elektronik/deconz-rest-plugin #1047: Can't pair IKEA GU10

saurabh984, 2018-12-17:
I bought a couple of IKEA GU10 Dimmable Whites yesterday and can't pair either of them.

I can reset them using Hue Dimmer Switch and they both respond by fading in and out.

I'm running v2.05.51 with firmware 262f0500 ConBee on ubuntu 18.04

This is my first IKEA device on the network. Until now I've only had Hue and ubisys.

Is this a bug or do I need to do something different to pair these?

PS: I'm running the Homebridge Service and deConz in Headless mode.

tleegaard, 2018-12-17:
I've got 16 IKEA GU10's up and running - but they were quite hard to pair. First I had to reset them (on off ~5 times). Then I added them in the Phoscon app one at a time with all other Hue/Trådfri bulbs turned off.

Smanar, 2018-12-17:
You can find video on youtube, https://www.youtube.com/watch?v=mJm9YpPrGzk

closed 2018-12-19 by saurabh984