dresden-elektronik/deconz-rest-plugin #1048: v2.05.51 Xiaomi switch still not working

ademobile, 2018-12-17:


manup, 2018-12-17:
Is it visible in the REST-API?

ademobile, 2018-12-17:
No and never has done - I wish it would as this would get my partner off my back over 'That ******* switch, that does not work and for how much longer!'

Sometimes it is shown in the GUI with a link to the controller, however in this version, no link (if that helps)

ademobile, 2018-12-17:
Just also found that the lumi.ctrl_ln2.aq1 also is still not right. The switch now shows up correctly in Phoscon and there are two entries in the lights, however only one is controllable. Looking at the switch this is the left paddle.

The lumi.ctrl_ln1.aq1 also is not showing up as a switch in Phoscon, however it still shown as a lamp and works fine, including the API

ademobile, 2018-12-17:
OK, getting somewhere. I deleted it (again), and added it (again) as described above, however nothing happened. So I moved on to the 2-gang (lumi.ctrl_ln2.aq1), removed and added that. It was detected, however again only the left switch is seen by deCONZ, Phoscon and the API. 

I then went back to the lumi.remote.b186acno1 and tried again and this now worked! (I am going to be off the hook tonight!). So that one can be crossed off my list.

Just need to know if the lumi.ctrl_ln1.aq1 should show up as a switch before I 'break' it.

manup, 2018-12-17:
Not sure yet why they don't show up properly. Having the lumi.ctrl_ln1.aq1 and lumi.ctrl_ln2.aq1 here, currently testing if I can recreate the issue.

ademobile, 2018-12-17:
I was 'messing around' (yes, I know I should not do this) and the lumi.ctrl_ln1.aq1 popped up in the switch list on the Phoscon GUI. No idea what I did not make this happen, so I will leave that well alone now!

However the right switch on the lumi.ctrl_ln2.aq1 is still not being seen by deCONZ, Phoscon or API. Just to confirm, local control from the switch its self is fine.

ademobile, 2019-01-07:
Just updated to 2.05.54 and now the lumi.ctrl_ln2.aq1 is still having issues. If I click the light in the group, the switch (right side) still does not trigger:

[image]

However if you click the light assigned to it, it does work:

[image]

The left hand switch works in both group and light assigned with no issues. Have I done something wrong?

manup, 2019-01-09:
I'll do some tests with the switch. Not 100% sure if I remember correctly but I think in the worst case only one of the endpoints supports being part of a group. This would describe the cause of this issue.

ebaauw, 2019-01-09:
I find the `lumi.ctrl_neutral2`, implements the _Groups_ server cluster for both light endpoints.  It reacts to _Group_ cluster commands, but doesn't react to _On/Off_ commands sent to the group.

The `lumi.ctrl_ln2.aq1` only implements _Groups_ on the first light endpoint.  I'm not sure if it reacts to _On/Off_ commands sent to the group.

closed 2019-05-16 by stale[bot]