dresden-elektronik/deconz-rest-plugin #1049: Conbee on Synology 1513+ Firmware Not connected

CarstenKlein, 2018-12-17:
Hi, 

I am very new to zigbee and Deconz. 
I am currently running openhab 2.3 on a Synology 1513+ all works fine so far, but I want to add some xiaomi sensors, lights and switches with zigbee. So I decided to try the Conbee USB Stick.  I saw that Deconz can be installed with the marthoc/Deconz Docker on the Synology.

I think I installed that Deconz Docker correctly because I can run the Phoscon App correctly.

But in the Gateway Settings "Firmware Not connected" is shown.
In  the deconz docker Protokoll on the Synology i get the following logs that didnt sound great:
"Serial com disconnected, reason: 4"
and
"API error 1, /config, unauthorized user"

The whole log can be found here:
http://klein-carsten.de/openhab2/logs/marthoc-deconz1.html

Can somebody help me with some hints.

I think only when I get this fixed I can try to integrated the stick to openhab. 

Thanks in advance 
Carsten

CarstenKlein, 2018-12-19:
So nobody is here running Deconz in a Synology Docker?

There are not that many information about Deconz/Synology. But I saw some threads where this combination worked. So I bought the stick with the hope that I can run this.

Every little hint/link would help a lot.

Carsten

manup, 2018-12-19:
Do you have a Desktop to check if ConBee works there on – Ubuntu, Raspbian or Windows?

CarstenKlein, 2018-12-19:
On my windows PC the stick is working.

http://klein-carsten.de/openhab2/logs/Deconz_Conbee.JPG

With the Phoscon App i am also able to add my Philips Hue dimmer switch (my only zigbee device for testing for now).

If I plug the stick into the Synology, the log looks like this:
http://klein-carsten.de/openhab2/logs/Deconz_Synology_Terminal.JPG

And I have absolutely no idea what I can do here, because there are no Synology informations.

manup, 2018-12-19:
Ok so firmware is present and working. I've no experience with Docker, but it might be a permission issue. deCONZ needs to have the rights to open the /dev/ttyUSB0 device. On Ubuntu the user needs to be in the group "dialout" which has permissions to **rw** the device.

SO you may check that the Docker user has permissions to open and read/write the device.

CarstenKlein, 2018-12-19:
Thank you for the hint, i am now one step further. The dongle is now visible in the Phoscon App within the docker:

http://klein-carsten.de/openhab2/logs/Deconz_Conbee_ready.JPG
In another forum I found that I need to check "Container mit hoher Priorität ausführen" in english:
"Execute container using high privilege".
After doing that, this first part is working now.

but...

Openhab says: 
Status: OFFLINE - CONFIGURATION_ERROR Failed to create new user on hue bridge.
And the log file says:
`2018-12-19 18:53:19.088 [INFO ] [binding.hue.handler.HueBridgeHandler] - Creating new user on Hue bridge 192.168.2.11 - please press the pairing button on the bridge.
2018-12-19 18:53:19.417 [WARN ] [binding.hue.handler.HueBridgeHandler] - Failed creating new user on Hue bridge`
http://klein-carsten.de/openhab2/logs/Conbee_Synology_failed.JPG

I clicked the "Allow authentification" button in the Phoscon App but its not working. :-(

Perhaps this has something to do because i could not use the port 8080 for deconz because its already in use with openhab.
I have DECONZ_WEB_PORT: 8090 and DECONZ_WS_PORT: 8093 for deconz instead.

This all makes me so unhappy I have I no idea where I have to look now. Is it a Synology Port problem, is it just Openhab, .... I nearly spend the whole day searching and googleing. hm

Perhaps somebody has another hint for me.

manup, 2018-12-19:
I guess at this point since connection between deCONZ and Phoscon App is working, the link between  Openhab and deCONZ needs to be checked.

From the binding documentation it looks like you need to specify ip address and port for the `host` field.

https://www.openhab.org/addons/bindings/deconz/

CarstenKlein, 2018-12-19:
The problem  is, that this Binding isn't working with the Conbee Stick
(The Zigbee binding currently does **not support the Dresden Elektronik Raspbee and Conbee Zigbee dongles**)

So I need to use the Hue Binding I think.

I will google and google... 
But thanks a lot to this point

I I find a solution I will post i here

manup, 2018-12-19:
>The problem is, that this Binding isn't working with the Conbee Stick
(The Zigbee binding currently does not support the Dresden Elektronik Raspbee and Conbee Zigbee dongles)

I think the sentence refers to the ZigBee binding:

https://www.openhab.org/addons/bindings/zigbee/

manup, 2018-12-19:
Question is if the hue binding also supports configuring a port, or if it only works with port 80.

https://www.openhab.org/addons/bindings/hue/#thing-configuration

CarstenKlein, 2018-12-20:
It is very late now, but i cant give up.

In the mean time I installed on our backup Synology the latest Openhab 2.4 (now also in a docker, because 2.4 is not available as a normal Synology package at the moment).

I did it because only the 2.4 has the "Dresden Elektronik deCONZ" Binding.

And with that Binding the Conbee is finally marked as "online" in openhab now :-)

The next step is to add my Hue dimmer switch. If I want to add the switch as a "thing" there is a   "Configuration Parameters" called:

Sensor ID:

The deCONZ bridge assigns an integer number ID to each sensor.

I have now idea where I cant find this ID, but its too late now to check that. 
Will try and google that tomorrow .

There are still a lot of things in the deconz log that didnt sound great like:

`01:46:46:461 API error 1, /lights, unauthorized user`
or
`01:46:47:212 	/rest/events --> HTTP/1.1 404 Not Found`
or
`
01:46:48:063 Node 0x0017880104E52528 is known by 1 neighbors, last seen 2 s
01:46:48:395 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
`
but I have no idea what that means.

manup, 2018-12-22:
Is it working now?

CarstenKlein, 2018-12-22:
Not really. It is nearly Christmas and I have a lot of other things to do now, because I lost so much time in the past days trying to get that all working :-)

Is think the Conbee stick is connected with obenhab now. But I have no idea what the Sensor ID is:
http://klein-carsten.de/openhab2/logs/Openhab_Sensor_ID.JPG

I couldn't find any IDs in the Phoscon App:
http://klein-carsten.de/openhab2/logs/Deconz_Switch.JPG

And In the log I only have these numbers:
http://klein-carsten.de/openhab2/logs/Button_press_log.JPG

I think the 1000-4000 are button numbers obe the hue dimmer swtich.
Maybe this is the Sensor ID: 0x0017880104E52528
but I I copy & paste this number to the Sensor ID field in OH: ....
Status: UNINITIALIZED

manup, 2018-12-22:
Not sure if Openhab uses the sensor `id` or also supports sensor `uniqueid` (recommended).
You can see both when you query the REST-API via Postman or directly in the the browser.

    http://<ip gateway>[:<port>]/api/<apikey>/sensors

For example on my gateway:

    http://192.168.2.4:8090/api/2037A73F57/sensors

Based on your log output I guess the `id` of the dimmer switch is 2.

manup, 2018-12-22:
You may create one using the REST-API as described here in the docs
http://dresden-elektronik.github.io/deconz-rest-doc/getting_started

Or more simple, use the apikey from Phoscon App (only valid until logout).

* In the Phoscon App press F12 to open the developer tools;
* In the console enter `app.apikey` and press enter.

CarstenKlein, 2018-12-22:
I think I am too stupid for all that. 
I think the basics are missing. 
If I do that with the logout PhosconApp I get this:
http://klein-carsten.de/openhab2/logs/Phoscon_API_Key.JPG

manup, 2018-12-22:
>If I do that with the logout PhosconApp I get this:

You need to be logged in :)
By logging in, a apikey is generated which the Phoscon App uses.

CarstenKlein, 2018-12-22:
This is what I get when i am logged in :-)

`{"1":{"config":{"configured":true,"on":true,"sunriseoffset":30,"sunsetoffset":-30},"etag":"8beebe1ab31595ef5101ce3cfffbcb08","manufacturername":"Philips","modelid":"PHDL00","name":"Daylight","state":{"dark":true,"daylight":false,"lastupdated":"2018-12-22T17:26:52","status":230},"swversion":"1.0","type":"Daylight","uniqueid":"00:21:2e:ff:ff:03:25:81-01"},"2":{"config":{"battery":100,"group":"1473","on":true,"reachable":true},"ep":2,"etag":"73484711af61304f2b34f0b2d66d2c55","manufacturername":"Philips","mode":1,"modelid":"RWL021","name":"Dimmer Switch","state":{"buttonevent":4002,"lastupdated":"2018-12-22T10:06:10"},"swversion":"5.45.1.17846","type":"ZHASwitch","uniqueid":"00:17:88:01:04:e5:25:28-02-fc00"}}`

In Openhab I tried this as a "Sensor ID": 

- 1
- 2
- 8beebe1ab31595ef5101ce3cfffbcb08
- PHDL00
- 00:21:2e:ff:ff:03:25:81-01
- 00:21:2e:ff:ff:03:25:81
- RWL021
- 00:17:88:01:04:e5:25:28-02-fc00
- 00:17:88:01:04:e5:25:28-02
- 00:17:88:01:04:e5:25:28

and Openhab shows always:
Status: UNINITIALIZED

If I have a look at this example from here:
https://www.openhab.org/addons/bindings/deconz/

Things file:
```
Bridge deconz:deconz:homeserver [ host="192.168.1.3", apikey="ABCDEFGHIJ" ] {
    presencesensor livingroom-presence          [ id="1" ]
    temperaturesensor livingroom-temperature    [ id="2" ]
}
```

How can they write an API-Key in the Thing file, when the API-Key changes every time you login?

And I have no idea where the [ id="1" ] and  [ id="2" ] comes from.

manup, 2018-12-23:
>How can they write an API-Key in the Thing file, when the API-Key changes every time you login?

The gateway maintains many apikeys, usually every app has its own key.

It looks like the Openhab binding can aquire it's own permanent key from deCONZ.

https://www.openhab.org/addons/bindings/deconz/#thing-configuration

>Important: You need to authorize 3rd party applications in deCONZ for the bridge Thing to successfully connect. The Online/Offline status of the bridge Thing will tell you, when to enable the authorization process.

The way I read it is you need to **not** specify the apikey in the configuration. Openhab will fill that automatically with a valid key.

To allow Openhab to aquire the key you need to enable authorization for 60 seconds in the Phoscon App settings, as described here: 

https://doc.phoscon.de/app/doc.html#settings-gateway-advanced-en

Cybersmurfen, 2019-01-26:
SSH into your Synology, then:

sudo insmod /lib/modules/usbserial.ko
sudo insmod /lib/modules/ftdi_sio.ko

circuuz, 2019-01-29:
> SSH into your Synology, then:
> 
> sudo insmod /lib/modules/usbserial.ko
> sudo insmod /lib/modules/ftdi_sio.ko

@Cybersmurfen Does that work?

circuuz, 2019-01-30:
> > SSH into your Synology, then:
> > sudo insmod /lib/modules/usbserial.ko
> > sudo insmod /lib/modules/ftdi_sio.ko
> 
> @Cybersmurfen Does that work?

For anyone coming here, it does work.

imbaimbaimba, 2019-03-11:
> SSH into your Synology, then:
> 
> sudo insmod /lib/modules/usbserial.ko
> sudo insmod /lib/modules/ftdi_sio.ko

Wow, the solution listed here, is not something I could find anywhere else on the web, but that works. Maybe it would make sense to add that to relevant instructions so it is easier to find for other people? Took me hours and hours to find it here.

Elvys8, 2019-03-12:
> Thank you for the hint, i am now one step further. The dongle is now visible in the Phoscon App within the docker:
>
> http://klein-carsten.de/openhab2/logs/Deconz_Conbee_ready.JPG
> In another forum I [...]

Hi! I have a Synology DS218+. I'm trying to get my new ConBee stick working, and I get the "Firmware Not connected" message, too. How did you solve the permissions issue? 
[https://i32.photobucket.com/albums/d26/Elvys1985/Captura%20de%20pantalla%202019-03-12%20a%20las%2016.20.14_zpsepyfzac4.png](https://i32.photobucket.com/albums/d26/Elvys1985/Captura%20de%20pantalla%202019-03-12%20a%20las%2016.20.14_zpsepyfzac4.png)

Elvys8, 2019-03-13:
> SSH into your Synology, then:
> sudo insmod /lib/modules/usbserial.ko
> sudo insmod /lib/modules/ftdi_sio.ko

I've tried that, but I still get "Firmware Not connected". What is it supposed to do?

imbaimbaimba, 2019-03-13:
I couldn’t find it in any documentation anywhere, so I have no idea. But before doing this, I couldn’t see USB01 at all. After, it worked.

Elvys8, 2019-03-13:
@imbaimbaimba Could you please post some screenshots of your Docker container configuration? I think I'm missing something, but I'm too dumb to know what it is.

themfinger, 2019-03-23:
> SSH into your Synology, then:
> 
> sudo insmod /lib/modules/usbserial.ko
> sudo insmod /lib/modules/ftdi_sio.ko

How can it be absent from everywhere?
It's mandatory to do that!!

Thanks, millions!!

degofedal, 2019-04-24:
@CarstenKlein I'm having the exact same problem regarding "Firmware not connected".

I've read this thread several times and searched everywhere online for at solution, but simply cannot find one.

My Docker container is running with high privileges so Write/Read access to /dev/ttyUSB0 shouldn't be a problem should it? also I tried the commands suggested by @Cybersmurfen, also with no luck. I have attached some screenshots below this thread, everything seems fine to me.

How did you manage to get to the point where the "Firmware Not Connected" issue was [resolved](http://klein-carsten.de/openhab2/logs/Deconz_Conbee_ready.JPG)?

Any help will be greatly appreciated.

Thanks in advance
Jeppe

Phoscon App:
[image]

Deconz terminal log:
[image]

Container settings:
[image]

Cybersmurfen, 2019-04-24:
@degofedal
Did you SSH in to your synology and ran the commands I pasted? The log shows /dev/ttyUSB even if you haven't. You can not do this in the docker terminal.

If yes:

Try download deconz for windows (if you use it) and update the firmware for the Conbee that way - much easier than run it trough docker until you get it running.
Also, I'm running it in priviledged mode (Havent tried otherwise)
And for a last note, i run the docker part mount for deconz at /root/.local/share/dresden-elektronik/deCONZ

degofedal, 2019-04-24:
@Cybersmurfen Thanks for your input. I tried updating the firmware and running deconz in Windows 10. Works fine and I'm able to connect a Xiaomi device.

Just for good measure I ran your commands again and connected the ConBee stick to the NAS. Unfortunately no luck. Still the same problem with "Firmware Not Connected" when I login to Phoscon App.

I notice you say you run your container In priviledged mode. Do you mean "Execute container with high priviledges"? If so, I have this checked also.

Execute container with high priviledges:
[image]

Windows 10 Gateway:
[image]

Windows 10 Device working:
[image]

Cybersmurfen, 2019-04-24:
@degofedal - its not certain that your device runs as ttyUSB0, it can as well be ttyUSB1 or ttyUSB2 if you have another hardware connected to your port. If you use the Conbee II it will probably show up as ttyACM0 or ttyACM1

Also, I would recommend an extension cable for the Conbee when you have got it running - with the Conbee directly attached to the NAS there is almost no radio signal.

degofedal, 2019-04-24:
@Cybersmurfen Thank you so much for that tip. I have it up and running. 

Did an `ls /dev/*` and it turns out the stick was showing up at `/dev/ttyACM0`.

Didn't think to check as the documentation states...

> Pass the serial device at ttyUSB0 (i.e. a Conbee USB device) into the container for use by deCONZ (if using RaspBee, use /dev/ttyAMA0).

Nevertheless, thank you so much for your help, I really do appreciated it!

pcherkash, 2019-04-25:
[image]

RUN:

docker run -d \
    --name=deconz \
    --net=bridge \
    -p 6681:80 \
    -p 6682:443 \
    --restart=always \
    -v /etc/localtime:/etc/localtime:ro \
    -v /volume1/docker/deconz:/root/.local/share/dresden-elektronik/deCONZ \
    -e DECONZ_VNC_PASSWORD=qPeXT8Qq \
    --device=/dev/ttyACM0 \
    marthoc/deconz

That's right. But still not connected

pcherkash, 2019-04-25:
Found the cause!! Need to specify
 -e DECONZ_DEVICE=/dev/ttyACM0 \

closed 2019-08-30 by stale[bot]