dresden-elektronik/deconz-rest-plugin #1050: 2.05.51 creating disaster?

wvuyk, 2018-12-17:
I just upgraded to 2.05.51 on my raspbee and am afraid this created a disaster here. Most sensors and many lights are suddenly unknown?

[image]

No new devices added, was nicely formatted on 2.05.50....

[image]

What just happend?

manup, 2018-12-17:
Holy cow, this looks bad, do you have a backup from before this happened?

manup, 2018-12-17:
And are the devices shown correctly in Phoscon App / REST-API or are they gone there?

wvuyk, 2018-12-17:
@manup This is one of those times where the backup is too old... 3 months. Good reminder to an old IT guy what the best habits are... no backup applicable. 
All device that lost there name are no more available. I returned to 2.05.50 for now and am in the process of rescaning sensors and the lights... I cannot see a real line in this, one philips motion sensor survived this, but 4 others are gone, some xiaomi sensor, like my smoke alarm survived this issue too, bot most other xiaom devices are gone.... 

On my Conbee I went to 2.05.51 yesterday, this is an IKEA only environment. Still running fine here without issues.... Back to adding lights and sensors here... 
When finished I will crate that backup LOL and try again? Or should I stay put?

manup, 2018-12-17:
I can only imagine it is related to this cleanup commit added in 2.05.51

https://github.com/dresden-elektronik/deconz-rest-plugin/commit/db2eaa1dcc4e6d3a75040a4414485dbd2667e2ea

However this was tested for various cases and didn't cause damage. 

>When finished I will crate that backup LOL and try again? Or should I stay put?

Sorry for the inconvenience, I guess most people including myself  only do backups the next time after something gone wrong. I'll add some code so that if a new version is installed a backup of the database is created automatically. In fact for updates via the update button this is already done, but not for manual installs.

wvuyk, 2018-12-17:
It is ok... Am adding the sensors, and am crossing my fingers for when I dare to start the plugin.. Don't hope to many people will see what I am seeing :-)
I am having the feeling a lot of sensors are newly added, instead of taking back the stale entries, can this be true? Should I remove thos later?

manup, 2018-12-17:
>All device that lost there name are no more available.

>I am having the feeling a lot of sensors are newly added, instead of taking back the stale entries, can this be true? Should I remove thos later?

I'm still not sure what was lost in the database, by no more available do you mean not listed anymore or displayed as unreachable?

The mentioned commit doesn't delete any sensors or lights but double entries of some lower layer stuff. So in theory the devices should have still been visible in the Phoscon App.

wvuyk, 2018-12-17:
They have been removed somehow....
I am now working here on the lights. IN this case a GU10 from Philips... I am here pressing 0's and F5's m the object, but there is no way it refreshes its info so that it will be presented again in Phoscon. I can cotrol it through the cluster info though, and it reads all info when I press the read buttons.

[image]

Do I have to reset these bulbs?

gust1n, 2018-12-17:
FWIW something similar happened to me when upgrading from V2_05_39 to V2_05_50 last week. ALL lights, sensors and button went unreachable, even after restoring backup (which I luckily had made). I had to manually re-add everything (just opening network and issuing add command on all devices). Sorry I have no relevant logs, just wanted to chime in that it might not be only the last couple of commits that triggered it.

wvuyk, 2018-12-18:
Finally all devices are back, plugin is running. Will create a nice and save backup first and upgrade to 2.0.05.52. One is suprised how much a house can depend on its automation.... :-)

wvuyk, 2018-12-18:
a ps, my vibration sensors did not rejoin yet though... will check with the new version

manup, 2018-12-18:
And don't forget about the backup :)

wvuyk, 2018-12-18:
I am getting errors while compiling the 2.05.52 packages?

> se/rest_gateways.o rest_gateways.cpp
rest_configuration.cpp:1748:59: warning: unused parameter ‘req’ [-Wunused-parameter]
 int DeRestPluginPrivate::updateSoftware(const ApiRequest &req, ApiResponse &rsp)
                                                           ^
rest_configuration.cpp:1774:59: warning: unused parameter ‘req’ [-Wunused-parameter]
 int DeRestPluginPrivate::restartGateway(const ApiRequest &req, ApiResponse &rsp)
                                                           ^
rest_configuration.cpp:1802:55: warning: unused parameter ‘req’ [-Wunused-parameter]
 int DeRestPluginPrivate::restartApp(const ApiRequest &req, ApiResponse &rsp)
                                                       ^
rest_configuration.cpp:1828:60: warning: unused parameter ‘req’ [-Wunused-parameter]
 int DeRestPluginPrivate::shutDownGateway(const ApiRequest &req, ApiResponse &rsp)
                                                            ^
rest_configuration.cpp:1856:59: warning: unused parameter ‘req’ [-Wunused-parameter]
 int DeRestPluginPrivate::updateFirmware(const ApiRequest &req, ApiResponse &rsp)
                                                           ^
rest_configuration.cpp:1879:57: warning: unused parameter ‘req’ [-Wunused-parameter]
 int DeRestPluginPrivate::exportConfig(const ApiRequest &req, ApiResponse &rsp)
                                                         ^
rest_configuration.cpp:1902:57: warning: unused parameter ‘req’ [-Wunused-parameter]
 int DeRestPluginPrivate::importConfig(const ApiRequest &req, ApiResponse &rsp)
                                                         ^
g++ -c -pipe -Wno-

Not sure if this is supposed to happen?

wvuyk, 2018-12-18:
Looking a ot better now, vibration sensor is added again. Still searching in the house for 4 more sensore that are shown..... will close this one!

[image]

closed 2018-12-18 by wvuyk