dresden-elektronik/deconz-rest-plugin #1054: Don't succeed in updating deConz

olderbag, 2018-12-18:
I'm running deConz in a UBUNTU 16.04 VM on Proxmox.

Tried to update deConz with following output:
`user@VM-Ubuntu:~$ sudo dpkg -i deconz-dev-2.05.52.deb
(Lese Datenbank ... 185502 Dateien und Verzeichnisse sind derzeit installiert.)
Vorbereitung zum Entpacken von deconz-dev-2.05.52.deb ...
Entpacken von deconz-dev (2.05.52) über (2.05.52) ...
deconz-dev (2.05.52) wird eingerichtet ...
user@VM-Ubuntu:~$`

Bt even after rebbot of the VM still 2.05.20 is running:
[image]

Can u help me pls to analyse what's going wrong? Unfortunately i'm not the LINUX expert.

manup, 2018-12-18:
This is only the -dev package, you also need to install the deCONZ package

https://www.dresden-elektronik.de/rpi/deconz/beta/deconz-2.05.52-qt5.deb

manup, 2018-12-18:
And please make a backup first in the Phoscon App :)

olderbag, 2018-12-18:
Worked finally - after chanching the path to UBUNTU sources. :-)

Thanks!!!

closed 2018-12-18 by olderbag