dresden-elektronik/deconz-rest-plugin #1055: Power-on behavior

TimmyUll, 2018-12-18:
When updating and change power-on behavior on Hue-lights to "last uses color and brightness". Then connect them to conbee again. The Power-on behavior changes back to 100% White light regardless what previous state was.

ebaauw, 2018-12-18:
That’s the idea of a factory reset, which you need to make the light join another network.  Change the poweron settings from the deCONZ GUI.

ferdel73, 2018-12-20:
Hey ebaauw,
I dont get how to do it...
Would anyone please be so kind to explain how to do it exactly?
Where do I have to fill in what value?
Regards
Ferdel73

ebaauw, 2018-12-21:
Go to the _Cluster Info_ panel and write the _PowerOn_ attributes of the _OnOff_, _Level Control_, and _Color Control_ clusters.  See https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1014#issuecomment-445452409 and the next comment.

ferdel73, 2018-12-21:
Thank you for the quick answer! I will try during Chrismas holidays. Just last night I got deCONZ running over x11vnc, before I only used deconz.service headless (deconz-gui.service wont start...). But I ran into some problems regarding different data / databases under phosconn with gui and headless. Regards Ferdel

bytat, 2019-02-08:
I've tried to set PowerOn OnOff to both "On" and "Off" ("Previous" not working) but it doesn't seem to matter. The light still comes on at 100%, not remembering last state when turning off electricity. Am I misunderstanding what's supposed to happen?

ebaauw, 2019-02-08:
What light type and what light firmware version are you using?  It works for me for all types of Hue lights at the latest (Dec 2018) firmware.

bytat, 2019-02-08:
it's a hue ambiance white e27 bulb. Model: LTW001 SW: 5.127.1.26420. Just updated to the latest fw.

It should be mentioned, I didn't seem to get it to work as expected with the hue gateway either. I was able to chose the power on behavior in the app but it seemed to reset both light levels and color when turning off the electricity. I didn't spend much time trying before changing to conbee, though, since I wasn't that interested in the hue gateway more than updating to the latest firmware.

ebaauw, 2019-02-08:
That is the firmware that supports _PowerOn OnOff_ (and the dimming and colour variants).

Note that it might take a couple of seconds before the light has written the setting to non-volatile memory.  You should be able to read back the setting in the deCONZ GUI.

bytat, 2019-02-09:
Ok. To rule out anything I might be doing wrong, would you mind telling me exactly what variable, and to what, I would need to set to get a light to go back to the previous state of light level and color when turning the electricity back on again?  All I could gather when looking at the linked threads was setting power onoff to "off" since it was already set to on.
(on another note, now when trying to set power onoff back to "on" again I get "failed to write")

ebaauw, 2019-02-09:
For previous settings, you need to write 0xff or 0xffff to the following attributes:
- _PowerOn OnOff_ in the _On/Off_ cluster.  Unfortunately, the deCONZ GUI won’t let you, see https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1014#issuecomment-445452409.
- _PowerOn Level_ in the _Level Control_ cluster;
- _PowerOn Color Temperature_ in the _Color Control_ cluster;
- _PowerOn X_ and _PowerOn Y_ in the _Color Control_ cluster.  You need to edit general.xml, comment out the _Current X_ and _Current Y_ attributes, and uncomment the _PowerOn_ variants.  They use the same attribute IDs, but the latter are manufacturer specific.  Unfortunately, the deCONZ GUI cannot handle both sets st the same time, see https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1014#issuecomment-445453767.  Make sure to restart deCONZ after editing general.xml (in /usr/share/deCONZ/zcl).

bytat, 2019-02-12:
I'm sorry for the late reply. I've managed to set the default value of PowerOn OnOff and I've uncommented the PowerOn variants. I haven't gotten around to fix/understand the rest yet but I'll be back with more questions when/if they'll arise. Thanks for the help!

ferdel73, 2019-02-25:
Hello again,
after repairing my raspberry installation -'which took quite some time -'the deCONZ GUI is working now. But as described it fails to write the “Previous“ value to Poweron onoff. How can i change this value without the software? It is the latest version, bulb fw is at december version.
Do i have to use newer github Files?
Will there be a patched release soon?
Sorry for being pesky (AND late) about this, but i appreciate your knowledge!

ebaauw, 2019-02-25:
This is an issue with the deCONZ GUI, not with the open source REST API plugin.

If you're comfortable reading and writing ZCL payloads, you might try https://github.com/ma-ca/deconz-cli-plugin.

closed 2019-07-02 by stale[bot]

frankhommers, 2019-11-30:
Hey @ebaauw, I have read all your work on the Power On stuff.
I just upgraded all my bulbs to the latest firmware since I have never done that. Now I want to make use of the "Previous" mode. So the lights come on in the same state/color/level as they went out. 
I still doesn't work through the deconz GUI.

Can you help me do this? What steps do I need to follow?

I have edited general.xml to circumvent the enum problem by adding this:

```
<value name="Unused2" value="0x02"/>
<value name="Unused3" value="0x03"/>
<!--etc-->
<value name="Unused253" value="0xfd"/>
<value name="Unused254" value="0xfe"/>
```
That seems to work for the on/off state, but not for the color and level... Please help me.

ebaauw, 2019-11-30:
See [above](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1055#issuecomment-462084698).

For Hue lights to power on at previous settings, you need to set the following attributes:
1. _OnOff_ cluster 0x0006, _PowerOn OnOff_ attribute 0x4003 to value _Previous_ (0xff).
The issue with this is, indeed, that the deCONZ GUI doesn't support `enum` values with gaps.  Specifying all 256 possible values in general.xml might very well work around it; I've never tried this myself, though;
2. _Level Control_ cluster 0x0008, _PowerOn Level_ attribute 0x4000 to value 255 (0xff).
The deCONZ GUI supports this without any issues.  Note that the light needs to remain powered for some time after writing the attribute, presumably to copy the setting to non-volatile memory.  Best read back the setting after writing it, to make sure it caught on.
3. _Color Control_ cluster 0x0300, _PowerOn Color Temperature_ attribute 0x4010 to value 65536 (0xffff).
The deCONZ GUI supports this without any issues.  This is needed only for color temperature and extended color lights (supporting `ct`).
4. _Color Control_ cluster 0x0300, manufacturer-specific _PowerOn Current X_ attribute 0x0003 to value 0x65536 (0xffff), and manufacturer-specific _PowerOn Current Y_ attribute 0x0004 to value 0x65536 (0xffff).
The deCONZ GUI can support this, but not while also supporting the standard _Current X_ and _Current Y_ attributes.  You need to comment out these in `general.xml`:
https://github.com/dresden-elektronik/deconz-rest-plugin/blob/14c07293647d78385ee0b4dea61a8fdd04e270d7/general.xml#L1565-L1566
and then uncomment the manufacturer-specific attributes:
https://github.com/dresden-elektronik/deconz-rest-plugin/blob/14c07293647d78385ee0b4dea61a8fdd04e270d7/general.xml#L1610-L1613
This is needed only for (extended) color lights (supporting `xy`).

frankhommers, 2019-11-30:
@ebaauw Awesome! I had a hard time with step 4. That works now!

Maybe for others that need this same help correct 0x65535 into 65535 :-)

AudiTTR, 2020-01-21:
Thx @ebaauw for the time you've put into this, for providing the steps above.  Also thx to @frankhommers for the clever workaround to circumvent the enum8 issue wrt non-continuous values.

I've been able to successfully setup my deCONZ GUI so that I can now configure my Hue bulbs so they're default to the 'Previous' power on state.  SWEET!

However, I've got nearly 100 bulbs to do, and doing those 4 steps in the deCONZ GUI for each and every bulb is going to very tedious.  Not to mention mistake-prone.  I also want to be able to quickly re-apply the 'Previous' power on state if I have to remove groups of my bulbs from my Raspbee/deCONZ/Phoscon bridge, link them to the original Hue bridge, do a FW upg, and then re-link them back to my Raspbee bridge.

Therefore, is there a way these 4 steps could be automated/scripted via bash or python so I do a large number of bulbs quickly and without making any mistakes?

ebaauw, 2020-01-21:
Lacking (pending?) support for this by the REST API plugin, you would need to use the [deconz-cli-plugin](https://github.com/ma-ca/deconz-cli-plugin) to send the attribute write commands.

zladukas, 2020-01-24:
@ebaauw, 
could You please help me make correct deconz-cli ZCL command?

I'm trying to set PhilipsHUE to "previous" 0xFF setting with no success.

I was using based on searching in gitHub: 

  zclattr <shortaddr> <ep> <cluster> <command>
  zclattr 0x5CCC 11 0x0006 02400320FF

ebaauw, 2020-01-24:
I like to keep my sniffer running when composing cli commands, to make sure that it indeed sends the ZigBee message I want.

You need to specify the attribute ID in the payload as big endian, so `0340` for 0x4003.  The type of the attribute is enum8, so `30`.  Not sure if the endpoint needs to be in decimal or hex.

So try:
```
zclattr 0x5CCC 11 0x0006 02034030FF
```
or
```
zclattr 0x5CCC 0x0B 0x0006 02034030FF
```

zladukas, 2020-01-24:
@ebaauw Many thanks, first one is working!!!
I was very near, but was using wrong data type.

Bruskey, 2020-05-02:
After going through all this, im still confused as to how to make this change. I want to set my Hue light which is not paired to a Hue bridge anymore, but directly to Deconz to adjust the power-on behavior to "previous state". Based on all of the above and any limitations or issues that might exist, what is the best way to do that? I am a noob in regards to Deconz and the GUI so might need a little bit of extended info if anyone is able to share their method to do this.

Thanks
Andrew

mgrn0, 2020-05-02:
@frankhommers would you mind sharing the lines you extended the general.xml as a gist/snippet?

Bruskey, 2020-05-04:
Are you able to provide step by step instructions around how to make this change if possible?

mgrn0, 2020-05-04:
> Are you able to provide step by step instructions around how to make this change if possible?

will do.

Bruskey, 2020-05-10:
Have you had a chance to list the steps needed to do this?

mgrn0, 2020-05-19:
sorry for the delay @Bruskey , for me the following steps were successful: 

- update the bulbs to latest firmware (I did so through a Hue bridge, removed the bulbs from the bridge and added them to my Raspbee 2 based system). (I only used regular bulbs, so no color temperatur nor hue adjustable ones.)
- add the missing values / lines (https://gist.github.com/mgrn0/8f52a05601a5fdc7d6e35e80c5cdc088) in /usr/share/deCONZ/zcl/general.xml , restart deCONZ
- select bulb which you want to set on/off behaviour for; click second grey round dot on the right to open dropdown (1)
- click "OnOff cluster 0x0006" (2)
- scroll down on the left and read 'Cluster info" attributes (3)
- double click "PowerOn OnOff" attribute (4)
[image]

- set attribute to "Previous" on modal / popup (it is the very last value on the bottom after all the "unused" entries we added in the xml above) (5)
[image]

- do similar with the "Level Control" Cluster (6) and "PowerOn Level" Attribute (7), set to value to "255"
- read value to make sure it was really written to non-volatile memory (light has to be on for some seconds)
[image]

- if you are using a bulb with temperature or hue control you might want to follow the steps 3 and 4 mentioned at https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1055#issuecomment-559993258

Hope this helps :)

Webunity, 2020-05-19:
@mgrn0 if you want to have your bulbs to 'off' state after a power cut (e.g. my home automation system will decide if they will be turned on) what steps should i then take, can you also elaborate on that in such detail?

mgrn0, 2020-05-20:
@Webunity I was only after the "last / previous" setting, but from what I understand you would chose "off" in step (5) instead of "previous", which is at the very top of the dropdown menu.
[image]

Bruskey, 2020-06-02:
@mgrn0 Apologies for sounding like a noob, but i have installed Deconz through Home Assistant as an addon on the same server Home Assistant is running on. When you say the below, how do i actually access the file system to make those changes to the xml file?

add the missing values / lines (https://gist.github.com/mgrn0/8f52a05601a5fdc7d6e35e80c5cdc088) in /usr/share/deCONZ/zcl/general.xml , restart deCONZ

mgrn0, 2020-06-04:
@Bruskey hm, I am running deCONZ on a Raspbee 2, so I don't know where in your case the files are located; if you go to "Edit > Preferences" (deCONZ) you will be able to see the path to the XML in your case, however you have to figure out how to edit the file / how to 'get' there, sorry (maybe somebody else can help?)

Gerben321, 2020-06-07:
I tried to follow the guide for my Hue GU10 bulbs, but when I turn them off and on again they're just cold white? Is the color part for X and Y correct? Can't set that anymore because it's gone commented out in the XML like in the guide.

ebaauw, 2020-06-07:
There a bug in the GUI, that it doesn't understand the same attribute IDs for standard and manufacturer-specific attributes.  If you comment out the standard _Current X_ and _Current Y_ attributes in `general.xml` and uncomment the PowerOn variants, you should be able to set these from the GUI.

Gerben321, 2020-06-07:
I've done that. I think I have to try it again some other time. It does seem to remember my brightness value though. But my Windows 10 seems to crash or logout when I am trying it too much, rather not mess around too much with it because I have to restart my VM's every time.

miljume, 2020-06-14:
I also try to find the `general.xml` file but with no luck. I am using deCONZ in Hass.IO as a Docker plugin. In Docker it says that the volume should be at /usr/share/hassio/addons/data/core_deconz but there is no such directory

miguelcaravantes, 2020-06-29:
why is the deconz GUI not fixed yet?

miguelcaravantes, 2020-06-29:
> See [above](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1055#issuecomment-462084698).
> 
> For Hue lights to power on at previous settings, you need to set the following attributes:
> 
> 1. _OnOff_ cluster 0x0006, _PowerOn OnOff_ attribute 0x4003 to value _Previous_ (0xff).
>    The issue with this is, indeed, that the deCONZ GUI doesn't support `enum` values with gaps.  Specifying all 256 possible values in general.xml might very well work around it; I've never tried this myself, though;
> 2. _Level Control_ cluster 0x0008, _PowerOn Level_ attribute 0x4000 to value 255 (0xff).
>    The deCONZ GUI supports this without any issues.  Note that the light needs to remain powered for some time after writing the attribute, presumably to copy the setting to non-volatile memory.  Best read back the setting after writing it, to make sure it caught on.
> 3. _Color Control_ cluster 0x0300, _PowerOn Color Temperature_ attribute 0x4010 to value 65536 (0xffff).
>    The deCONZ GUI supports this without any issues.  This is needed only for color temperature and extended color lights (supporting `ct`).
> 4. _Color Control_ cluster 0x0300, manufacturer-specific _PowerOn Current X_ attribute 0x0003 to value 65536 (0xffff), and manufacturer-specific _PowerOn Current Y_ attribute 0x0004 to value 65536 (0xffff).
>    The deCONZ GUI can support this, but not while also supporting the standard _Current X_ and _Current Y_ attributes.  You need to comment out these in `general.xml`:
>    https://github.com/dresden-elektronik/deconz-rest-plugin/blob/14c07293647d78385ee0b4dea61a8fdd04e270d7/general.xml#L1565-L1566
>    
>    and then uncomment the manufacturer-specific attributes:
>    https://github.com/dresden-elektronik/deconz-rest-plugin/blob/14c07293647d78385ee0b4dea61a8fdd04e270d7/general.xml#L1610-L1613
>    
>    This is needed only for (extended) color lights (supporting `xy`).

it seems not work with the last philips hue firmware i set the values but it doesn't work
[image]

ludipq, 2020-07-10:
When I try to write the value `Previous` to my Hue bulbs through the GUI, it gives me the error `writing failed`.

However, writing the value `off` works fine.

Why can't I set the `Previous` value?

mgrn0, 2020-07-14:
@ludipq did you add the value entries in the xml file as described above?

TheAlphaLaw, 2020-07-16:
It's unfortunate this isn't already corrected for Hue bulbs. The above method does not seem to work well for my color lights. Hoping we see some effort put into this from Deconz.

ludipq, 2020-07-18:
> @ludipq did you add the value entries in the xml file as described above?

No, I thought I didn't have to because the option was there in the GUI.

Where do I find `general.xml` when running on Docker?
I have mapped the following folder and this is the only content of it.

`/volume1/docker/deconz:/root/.local/share/dresden-elektronik/deCONZ`

[image]

ludipq, 2020-11-02:
> No, I thought I didn't have to because the option was there in the GUI.
>
> Where do I find `general.xml` when running on Docker?
> I have mapped the following folder and this is the only content of it. [...]

Anyone?

ludipq, 2020-11-21:
I just updated both my ConBee and deCONZ to the latest versions, and suddenly I am able to write the "previous" value to the bulbs, without editing the xml file!

gvdhoven, 2020-11-24:
@ludipq can you elaborate on the steps that you have taken?

Mimiix, 2020-11-24:
@gvdhoven On/off cluster in deCONZ. Then attribute 0x4003 :)

miljume, 2020-11-25:
I still can't write to the attribute, I receive the same error as before `writing failed`. Running deCONZ v2.05.88 and updated the FW to 0x26390500

bytat, 2020-11-25:
I managed to write to à couple of them. I think I need to upgrade the firmware on the others.

TheAlphaLaw, 2020-12-28:
> @gvdhoven On/off cluster in deCONZ. Then attribute 0x4003 :)

Writing previous works for me, but it doesn't function as intended.

theCroh, 2021-01-24:
Has anyone found a way to do this with Hass.IO deCONZ addon?

Mimiix, 2021-01-25:
@theCroh That works exactly the same :)

TheAlphaLaw, 2021-01-25:
> Has anyone found a way to do this with Hass.IO deCONZ addon?

I can't get the above to work in either. This should be built in by now...

theCroh, 2021-01-26:
> @theCroh That works exactly the same :)

Okay. How do I do this: "-- after editing general.xml (in /usr/share/deCONZ/zcl)." when there is no deCONZ folder in /use/share/ folder of my Hass.IO system? I have no idea, where the general.xml is but I'm sure, it's not in the folder mentioned in the guide.

Mimiix, 2021-01-26:
You should ask in the HA git/helpchannels then, as i have no clue where it stores it. @theCroh 

Either way. The HA Addon does provide a deCONZ web interface and you can change attribute in the on/off cluster (0x4003) afaik.

theCroh, 2021-01-28:
I got the settings enabled by following these instructions: https://community.home-assistant.io/t/where-to-find-general-xml-in-deconz/206739/5

Basically, you first need to use command "docker cp addon_core_deconz:/usr/share/deCONZ/zcl/general.xml ." (without the quotes but including the dot) in Terminal in Home Assistant to copy the settings file from the docker container. Then you need to edit it and **while restarting deCONZ** (when it's booting up), use command "docker cp general.xml addon_core_deconz:/usr/share/deCONZ/zcl/" to copy the file back inside the container. I was having a hard time getting the timing right so I just restarted deCONZ in one window and repeated the command for a while in another window. That way I could apply the settings.

I also tried to edit the file inside the container using Portainer addon but found no way to make it work since deCONZ needs to be restarted in order to apply the settings and restarting seems to reset the container.

theCroh, 2021-01-28:
I'm still unable to get this to work with at least Philips Hue Being light.

"OnOff cluster 0x0006, PowerOn OnOff attribute 0x4003 to value Previous (0xff)."
- I'm able to set this from the UI by just selecting the option from the dropdown. It also sticks.

"Level Control cluster 0x0008, PowerOn Level attribute 0x4000 to value 255 (0xff)."
- I'm able to set this as well from the UI by saving the value 255. It also sticks.

Color Control cluster 0x0300, PowerOn Color Temperature attribute 0x4010 to value 65536 (0xffff).
- This I can't do. Trying to save value 65536 seems to work as expected but reading the value afterwards shows that it saved as "0"

SHxKM, 2021-01-29:
@theCroh - same here but with ZHA (using Conbee) which to me indicates these values may have been updated since @ebaauw latest post.

theCroh, 2021-01-31:
> @theCroh - same here but with ZHA (using Conbee) which to me indicates these values may have been updated since @ebaauw latest post.
> 
> Edit: but `65535` works! I only needed to change:
> 
> OnOff cluster 0x0006, PowerOn OnOff attribute 0x4003 → `0xff`
> Level Control cluster 0x0008, PowerOn Level attribute 0x4000 → `255`
> Color Control cluster 0x0300, PowerOn Color Temperature → `65535`
> 
> I didn't need to change the X & Y values (I couldn't persist them anyway).

I can set the values as you say but the lamp still turns on with the default warm bright light. 

My OnOff cluster looks like this:
[image]

Level Control cluster like this:
[image]

And Color Control cluster like this:
[image]

Am I missing something obvious?

SHxKM, 2021-01-31:
@theCroh - Sometimes it takes a little time to commit its state, maybe you were doing the same as I was an immediately disconnecting it to see if it works? :)

ebaauw, 2021-01-31:
> I can set the values as you say but the lamp still turns on with the default warm bright light.

Some lights carry the attributes, but don't honour them.  What type of light is this?  What do the _PowerOn_ attributes report after power cycling the light?

What's the color mode after power on?  If it's `xy`, you might need to set _PowerOn X_ and _PowerOn Y_.  Afaik, only Hue lights support these.  To set these from the deCONZ GUI, you need to comment out the regular _Current X_ and _Current Y_ attributes in `general.xml`:
https://github.com/dresden-elektronik/deconz-rest-plugin/blob/f46e3321715f755a057c19992e51ce9cf6f01e35/general.xml#L2138-L2139
Next, uncomment the _PowerOn_ versions:
https://github.com/dresden-elektronik/deconz-rest-plugin/blob/f46e3321715f755a057c19992e51ce9cf6f01e35/general.xml#L2207-L2210
Make sure to restart deCONZ for the changes to take effect.

Note that the _PowerOn_ versions use the Philips _Manufacturer Code_ 0x100b.

theCroh, 2021-01-31:
> @theCroh - Sometimes it takes a light a little time to commit its state, maybe you were doing the same as I was an immediately disconnecting it to see if it works? :)

This is probably the issue. User stupidity error also known as ID-10T error in customer support lingo. I read from somewhere that it takes about seven seconds for the values to stick and my tests may very well have been:
1. Turn the light on ✔️ 
2. Adjust it somehow ✔️ 
3. Instantly turn the light off ❌ 
4. Turn the light on
5. Swear
6. Tinker something and repeat from 1

> Some lights carry the attributes, but don't honour them. What type of light is this? What do the _PowerOn_ attributes report after power cycling the light?
> What's the color mode after power on? If [...]

It's Philips Hue Being (this one https://www.philips-hue.com/en-gb/p/hue-white-ambiance-being-ceiling-light/3261031P6). The values seem to stick after reboot and color mode is "Color temperature" so the problem is probably sitting in front of the keyboard as suggested by @SHxKM above 😄 

Sorry about the bother and thank you for the help.

[9 more replies]