dresden-elektronik/deconz-rest-plugin #1056: Different time zones in log and REST responses

chbndrhnns, 2018-12-18:
I use the [dockerized version](https://github.com/marthoc/docker-deconz) of the rest plugin and set the environment variable `TZ=Europe/Berlin`.
While this setting seems to be respected when looking at the timestamps of the log entries, the API responds with `{"timezone": "Etc/GMT-1"}` when querying it at `host:port/api/KEY/config`.

duffbeer2000, 2019-04-17:
I'm running the docker container of marthoc too and I'm having the same Problem. The Rest-Api uses a different timezone as the host and Phoscon. Will there be a fix for that behavior? Or how can I tell the rest-api that it should use the timezone of the host?
[](https://github.com/marthoc/docker-deconz/issues/73)

manup, 2019-04-17:
From the container docs I think it's following parameter:

    -e TZ=America/Toronto

https://github.com/marthoc/docker-deconz

duffbeer2000, 2019-04-18:
I already set the timezone with -e TZ=Europe/Berlin, this didn't fix the issue. After that I mounted /etc/localetime to the container and it didn't fixe the issue. The time of the host is correct, inside the container the time is correct, phoscon shows the correct time too. The only thing which gives the wrong time is the rest api

manup, 2019-04-18:
Keep in mind that most parts of the REST-API use UTC time, not the local time.

chbndrhnns, 2019-04-18:
So here is again what I observed:
- I have `TZ=Europe/Berlin` in my Docker config
- When I look at the config, I see

```
{
"swversion": "2.5.60",
  "timeformat": "24h",
  "timezone": "Etc/GMT-2".
}
```

I would expect timezone to be `Etc/GMT+2` instead of `Etc/GMT-2`.

Wildbill-Z, 2019-04-18:
I'm using deconz without Docker directliy with a Raspi3 and Raspbian and see same "problem". Deconz/Phoscon is showing the correct time, but the api gives timestamps with 2 hours difference as shown above. Also it's rather sa cosmeticsl problem, is there anything I can do to fix this?

ebaauw, 2019-04-18:
All times reported by the API are in UTC, with the obvious exception of `/config/localtime`.

chbndrhnns, 2019-04-19:
Why not making it explicit then in the API responses that it's UTC?

ebaauw, 2019-04-19:
Do you mean report `state.lastupdated` as `2019-04-19T05:22:37Z` instead of `2019-04-19T05:22:37`?  That's actually not a bad idea, but might give compatibility issues with existing applications that don't expect the timezone.  It would also deviate from the Hue API.

closed 2019-08-24 by stale[bot]

llamafilm, 2021-08-10:
I have the same issue.  Using the dockerized version, I set `TZ=America/Los_Angeles`.  That had no effect so I also mounted the `- /etc/localtime:/etc/localtime:ro`

The web UI still shows _Timezone: America/New_York_.  If I change it there, it always reverts back to New York after restarting the container.