dresden-elektronik/deconz-rest-plugin #1059: Lightify a19 tunable white showing unexpected fields in rest api

wvuyk, 2018-12-18:
A user of my plugin is using this lightifyl bulb. It should support CT values, but it is not presented in the REST API results. It shows as a xy only bulb?
Sending a CT command through the REST API addresses the bulb correctly and the CT is set on the bulb. Main issue, the CT value is not returned through the api. Is this because it is OSRAM (....) or just a miss for some reason?

[image]

manup, 2018-12-18:
Some screenshots of the endpoints and clusters would be helpful.
Also in the color control cluster the `color capabilities` attribute.

MattL0, 2018-12-18:
I manup,

I am sorry I am running deconz on a headless machine :s So i can’t have access to those values I think?

And i removed thoses two osram bulbs and replaced them with hue bulbs.

But I can readd then for testing purpose

closed 2018-12-18 by wvuyk