dresden-elektronik/deconz-rest-plugin #1060: Deconz 2.05.52 lost Vimar/Hue Tap switches

rafuz, 2018-12-18:
Installing the last version I have completely lost my switches again (as in 2.05.50)
I only see them as groups:

[image]

I am using the Vimar switches that I linked in the issue #1020 that should be seen as Hue Tap.
I use them through the rest interface.

manup, 2018-12-19:
Damn the group device membership check was not complete.

Fixed already in the online version [http://dresden-elektronik.de/pwa](http://dresden-elektronik.de/pwa?19-12) and will be part of 2.05.53.

**Note:** This is only shown for former added switches, newly added switches won't show up on the main page.

ZigBee Green Power (ZGP) devices like Hue Tap, Vimar and Illumra don't control a own group directly, the group was added in the past by deCONZ with some default rules.

This was before the Switch Editor existed.

With recent versions (.51 and .52):
* No group and default rules will be created
* Instead the switch can control one or more normal light groups
* Also a light group can be controlled by multiple switches

We publish the related documentation very soon, hope it helps.

[image]

With the Switch Editor custom actions can be assigned to the buttons.

rafuz, 2018-12-19:
Fine, I will check as soon Marthoc (https://github.com/marthoc/docker-deconz) bumps the version.

So to get rid of the groups I have to rejoin the switches?

manup, 2018-12-19:
> So to get rid of the groups I have to rejoin the switches?

Or just delete the groups, but only do this if they aren't used or have lights assigned.

rafuz, 2018-12-22:
I have checked the online version and I see all the vimar switches as philips tap switches. This means that when I update my local PWA it will be the same?

manup, 2018-12-22:
Yes, basically the local version is just a copy which is updated with each release.

I've seen that the switches support more features like dimming. Looking forward to support these with a own implementation in early 2019. However the `buttonevents` and `modelid` might be different from the Hue tap.

rafuz, 2018-12-22:
To be honest I am using them only capturing their buttonevents in the REST interface for activating lights in Home Assistant. In deconz their buttonevents do not match Philips Tap of Friends of Hue switches so they need a custom integration. 
Anyway if you will implement them properly it will be very nice, they also cost less than their Friend of Hue counterparts.

manup, 2018-12-22:
>In deconz their buttonevents do not match Philips Tap of Friends of Hue switches so they need a custom integration.

Yes I guess so, do you know the button events for the Friends of Hue switches used by the Hue bridge? The Philips documentation hasn't been updated yet.

Digging deeper into ZGP land, I've just sniffed the ZigBee traffic of the Hue bridge when a Tap is paired. Since the Hue bulbs (and likely other ZigBee 3.0 devices) support ZigBee Green Power, the Hue bridge does setup them so they repeat the Hue Tap messages through the mesh.

Currently deCONZ doesn't support this, so the ZGP switches need to be in signal range of the gateway. I'm planning to implement the ZGP commissioning via mesh soon, so the switches can be placed anywhere near the mesh.

rafuz, 2018-12-22:
> Yes I guess so, do you know the button events for the Friends of Hue switches used by the Hue bridge? The Philips documentation hasn't been updated yet.

No, unfortunately I don't know the buttonevents for Friends of Hue (Vimar 03906). I can only find the buttonevents for the generic ones I have (Vimar 03905)

Currently I have to put Deconz in the center of my house to avoid to loose button presses from these wall switches. I have some smart plugs that could route that events in the mesh so it is a much needed feature.

manup, 2018-12-22:
>I can only find the buttonevents for the generic ones I have (Vimar 03905)

Seems to use the PTM-215ZE-DA module.

https://www.zigbee.org/zigbee-products-2/#zigbeecertifiedproducts/productdetails3/58a567c06b774f4349e1fe42/

rafuz, 2018-12-22:
yes, it is exactly this

rafuz, 2019-01-08:
Just update to the latest phoscon 2.05.54 and I have lost again all my Vimar switches again :(
Last version working is still 2.05.49. The strange thing is that the online version of PWA seems to work.

[deconz.log](https://github.com/dresden-elektronik/deconz-rest-plugin/files/2738253/deconz.log)

[image]

[image]

[image]

manup, 2019-01-09:
:/ I don't understand why this happens, can you please send me a backup to mpi@dresden-elektronik.de 
I'll try to load the config here and look why they don't show up.

rafuz, 2019-01-09:
I will send you the current backup (2.05.49) this evening.

manup, 2019-01-14:
Hi, yes I've checked the file the switches are there and all have a assigned group. So from this side it looks good. The Phoscon App should already check exactly this to show the switches on the main page.

I wonder why they are missing on the *Menu > Devices > Switches* page. Here I need to do some more tests with your backup file. But it should be fixed in 2.05.55 if I can find anything.

rafuz, 2019-01-14:
Not only they are missing in the Menu > Devices > Switches page but I also do not get any output from them in the REST interface. If I can't get output from REST when I operate the switches I can't use them to fire events in HomeAssistant.

manup, 2019-01-14:
Found the cause, the commit 280cceacbf5f9d2c2bfbaa4527db1dfee063e2f8 prevents ZGPSwitch resources from loading.

rafuz, 2019-01-14:
Great to hear :-)
Will it be fixed in 2.05.55?

manup, 2019-01-14:
Yes :)

rafuz, 2019-01-14:
Good, when I will report back when I will be able to test it.

rafuz, 2019-01-15:
I can confirm it is fixed in 2.05.55, this can be closed.
I hope for future support for Vimar ZGPSwitches :-)

closed 2019-01-15 by rafuz

manup, 2019-01-16:
>I can confirm it is fixed in 2.05.55, this can be closed.
I hope for future support for Vimar ZGPSwitches :-)

Cool, yes the Friends of Hue switches will be fully supported. :)