dresden-elektronik/deconz-rest-plugin #1062: Samsung SmartThings outletv4 power plug

apmon, 2018-12-19:
Hello,
Is it possible to get the power metering for the Samsung SmartThings power plug (US) exported through the rest API?

Attached are the screenshots from deCONZ:

[image]

[image]

[image]

[image]

clementx2006, 2018-12-19:
seems Duplicate of #819 ?

apmon, 2018-12-19:
Manufacturer Name and model number seem different, which is why I opened a separate ticket. The electrical measurement cluster also seems to have some differences.

clementx2006, 2019-04-18:
em - not sure if this one is ever going to be resolved...

manup, 2019-04-19:
Our near future plan is to order and support most of the LEDVANCE, Samsung / Smart Things devices which are popular in the US.

closed 2019-08-24 by stale[bot]

clementx2006, 2019-11-10:
Can we reopen this? Samsung released a new plug this year (2019). Would really love to see this one working (for power monitoring)...

smartypantsuk, 2019-11-21:
I'll second this. Samsung have released a Smartthing Smart Plug with power metering (UK) too.
Reasonably well priced and good quality. Can we get support? I think the UK model variant is GP-WOU019BBDWG