dresden-elektronik/deconz-rest-plugin #1066: Problems after firmeware-Update

frettijule, 2018-12-23:
I use a RaspberyPi with Raspbee.
After updating the Rasbee-shield to firmware 2.05.48 i lost all my connections to the zigbee-devices!!??

Some sensors, switches and  bulbs like Aqara, Osram, Tradfri i can reset with knob on device  or reset-sequenz. So i can connect i again to raspbee. 

But i also have some Hue Bulbs, but no Hue Bridge.

How can i reset them oder connect them to the raspbee-shield?

manup, 2018-12-23:
Which  deCONZ and firmware versions were installed before?

Please create a backup file in the Phoscon App gateway settings and send it to mpi@dresden-elektronik.de. I can check the network parameters, likely some parameter like channel or network id has changed. Changes are recorded in the database, so these can be restored (currently manually).

Hue bulbs can only be reset via Touchlink, either in the old Webapp or more simple with a Hue dimmer switch.

frettijule, 2018-12-23:
I think the last firmware before update was 2.05.44.
I just send the backup-file per email.

frettijule, 2018-12-23:
Thanks, it works.

closed 2018-12-23 by frettijule