dresden-elektronik/deconz-rest-plugin #1067: Can't connect FHEM Module

brown78, 2018-12-24:
Hello,
I have the Raspbee running on a Raspi 3 on Jessie. It's connected in FHEM and works very well.
No I wanted to upgrade to the 3B+ and Stretch. I can install and run deCONZ, also the Raspbee is working and all lights are found. But I can't get the FHEM Module connected. I open the Bee for connection, define the Module in FHEM and get a key. If I use this key in a browser and go to:
ip.to.the.raspi/api/<apikey>
I get error unauthorized user.
With the Jessie/FHEM setup, I can use the key to get the config.
I already mask getty.service, disabled serial console, bluetooth and any other thing I found regarding this issue.
Additionally I resetted the bee and flashed latest firmware using GCFFlasher. Although I resetted the bee, the lights were still there. Also if I open the PhosconApp there is a icon for the Gateway on the new raspi and for the old one even if the old one is not online (I'm using the same Raspbee for the old and the new one). So I guess there might be some configuration files which are not resetted.
Which files can I delete or modify to make it like a clean fresh install of deconz, I would not like to setup the raspi again, because there is FHEM and Owncloud and others already running. I already ordered a second Raspbee, but I think this might not help, if there are some config files on the raspi which maybe are causing my issue.

manup, 2018-12-24:
I guess the problem is more suptle. Looks connection is working, but lhe FHEM module needs to get an *apikey* to use the REST-API.

To allow FHEM acquiring an apikey, you need to enable authorization in the gateways settings.

https://doc.phoscon.de/app/doc.html#settings-gateway-advanced-en

I don't know what is needed to setup the connection on the FHEM side, it might work automatically. If not please refer to the install documentation of the related module.

brown78, 2018-12-24:
I have done all the steps required. After "Authenticate App" is active, in FHEM you simply define the module and you get an api-key. 
The api-key is shown in the device readings, but if I try to use this key to get the config via a browser with the above mentioned method, I get "unautorized user".
If I use the api-key from the old and running FHEM-Raspi to get the config - I do get the config.
So acquiring the api-key is not the issue - the api-key is either invalid or there is some other issue on stretch.
Do you know which folders and files I need to remove after uninstalling deconz to start over new?

manup, 2018-12-24:
The apikey is stored in the sqlite database file `zll.db`. If it works with the former installation but not with the new, it means FHEM uses a apikey which doesn't exist in the new deCONZ database.

I guess there are two ways to fix this:
* Copy the old zll.db file to your new Pi (deCONZ must be closed first)
* Or, remove/clear the "old" apikey  in FHEM and generate a new one

brown78, 2018-12-24:
To be clear, I get a new api-key if I define the module on the new raspi and this does not work for some reason. I can try with the zll.db from the old raspi. Or maybe on the new raspi delete/rename the zll.db to generate a complete new one?

brown78, 2018-12-25:
It works now, for some reason I had to set the attribute httpUtils=1 in FHEM, which wasn't needed on my old installation....
Merry Christmas!

closed 2018-12-25 by brown78