dresden-elektronik/deconz-rest-plugin #1068: 2.05.52 disaster, help please

donnib, 2018-12-26:
So i had 2.05.42 and i 0x262B0500 and i upgraded to deconz 2.05.52 and build the rest api (i did not upgrade the firmware) now nothing works. It exact same case as #1050 and i DID make a backup before i upgraded. Now i just want to go back to 2.05.42 so everything works in the house again.

How can i do it ? I did downgrade to 2.05.42 and i rebuild the rest api from 2.05.42 but nothing is recognized , nothing in REST api, no sensors, deconz shows all devices as one pile and are not shown organized as i used to have them. I tried restore but it won't work from PWA but it says it succeedes from the old UI (i did the backup from the old ui). 

Please help. Nothing works.

@manup any ideas ?

manup, 2018-12-26:
Hi donni can you please send me the backup from the old version as well as one created with the current installed version to mpi@dresden-elektronik.de I'll have a look, guess some network parameter has changed. Please don't reset anything in the meanwhile.

donnib, 2018-12-26:
[raspbee_gateway_config_2018-12-26.dat.zip](https://github.com/dresden-elektronik/deconz-rest-plugin/files/2710763/raspbee_gateway_config_2018-12-26.dat.zip)

manup, 2018-12-26:
OK thanks I'll have a look as soon as I get back in 2 hours, currently not at home.

donnib, 2018-12-26:
Anything i can do ? Everything i can do in the mean time, nothing works...

donnib, 2018-12-26:
I know how to open the db in a sqlite editor, what should i look for ?

donnib, 2018-12-26:
I see that the raspbee config is just the config file and not the database backup, i though the old ui Save configuration saved the db as well. I do have the db backed up so i have one here : 
[zll~20181226-181437.db.zip](https://github.com/dresden-elektronik/deconz-rest-plugin/files/2710789/zll.20181226-181437.db.zip)
I just don't understand where the information is saved other than the db if i haven't touched anything else ??

donnib, 2018-12-26:
[Archive.zip](https://github.com/dresden-elektronik/deconz-rest-plugin/files/2710792/Archive.zip)
Here is a copy of the deconz folder as it is now when it's not working.

donnib, 2018-12-26:
Ok i succeded to get back to an older backup (not the one i took just before). I had syncthing to backup /home/pi/.local/share/dresden-elektronik/deCONZ and not /home/pi/.local/share/data/dresden-elektronik/deCONZ so i had something but what is the difference between these two, i have replaced the data inside /home/pi/.local/share/data/dresden-elektronik/deCONZ

manup, 2018-12-26:
The differences in the path comes from older versions of Qt. Since a year or so *new installed* deCONZ versions will always use `~/.local/share/data/dresden-elektronik/deCONZ` as path to store database and other files. But only if no older path exists.

deCONZ does check for other locations, which were created by older deCONZ versions.
The following order applies:
```
~/.local/share/data/dresden-elektronik/deCONZ
~/.local/share/dresden-elektronik/deCONZ
~/.local/share/deCONZ
```

Not sure why your setup has two folders. There should be only the one which was created first. I'll do some tests to double check.

donnib, 2019-01-02:
@manup do you know the reason all this happened and when can i exepct to upgrade and not run into problems ?

manup, 2019-01-02:
In the last days I've fixed several issues in regard of backup creation/restore and updates. These fixes will be part of deCONZ 2.05.53.

But the multiple configuration directories in your setup must be cleaned up manually. Therefore it would be good if you keep only the one directory which contains the latest files (I assume `.local/share/data/dresden-elektronik/deCONZ`) and move or backup the other directorie(s) to some different location. Note deCONZ must be closed while doing that.

wvuyk, 2019-01-04:
@manup  Backup send. Let me know if I should just go for the upgrade or wait for you to investigate?

wvuyk, 2019-01-04:
Updated to 2.05.54, will let it run for a while now

closed 2019-05-11 by stale[bot]