dresden-elektronik/deconz-rest-plugin #1071: ConBee Configuration Broken After Backup/Restore - hot to reset?

caroso-de, 2018-12-27:
Hi,

I just tried to move my Configuration from my Laptop to a Raspberry Pi using the Backup/Restore functions of the Phoscon App. As a first result, all lights were gone and all sensors reported as "not seen" (or something like that). Starting Deconz back on my Laptop, the network settings looked strange (i.e. security was off, no NWK Ext PAN ID set).

I tried to reset the Network settings to the defaults from the user manual (Security High/TC Link Key, default TC link key). But after joining the Network I still only see the mockup Daylight sensor and in the Network settings the TC Link Key is gone and the channel reset to 11 (was 15).  

Since I'm just testing with a few devices, my primary question ist: how do I reset the ConBee Configuration to factory defaults so I can start over? I've browsed through similar issues, but nothing helped so far.

Nevertheless the failure to recover from backup bothers me quite a bit. What could I've possibly done wrong?

DeConz  2.05.20
Firmware: 0x261F0500

thanks a lot!

[image]

manup, 2018-12-27:
>Since I'm just testing with a few devices, my primary question ist: how do I reset the ConBee Configuration to factory defaults so I can start over? I've browsed through similar issues, but nothing helped so far.

Is the ConBee connected with deCONZ on the Raspberry Pi? This is indicated when a blue node is shown. From the settings it looks like the connection isn't established or something is really wrong.

Since the low level network parameters are stored directly on the device, they should show up unchanged when plugging the ConBee back in the laptop. They should have been also available on the Raspberry Pi.

If you send the backup to mpi@dresden-elektronik.de I can have a look later tomorrow.
The backup should contain all parameters in included database file.

manup, 2019-01-06:
The specific problem with backup restore is fixed in 2.05.54.

https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/V2_05_54

Closing the issue for now

closed 2019-01-06 by manup